arendst/Tasmota #8298: MQTT "Friendly Name" not provided since v8.2.0.5

richard-scott, 2020-04-28:
### PROBLEM DESCRIPTION

When looking in Home Assistant's list of auto-discovered MQTT devices the Device is named as the Template, and not the Friendly Name.

[image]

In the picture above you can see here, I have a "Boys Globe", and a "Grid Monitor House" device.
The Boys Globe is running v8.2.0, and the incorrectly named one is running v8.2.0.5

[image]

This is a picture of both devices running v8.2.0

### REQUESTED INFORMATION

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Device used (e.g., Sonoff Basic): Wemos D1 Mini
- [x] Tasmota binary firmware version number used: 8.2.0.5
  - [x] Pre-compiled
- [x] Flashing tools used: Tasmotizer

### TO REPRODUCE
Install v8.2.0 and set a nice friendly MQTT name.
Look for this name in Home Assistant.
Upgrade to v8.2.0.5 and when the device restarts, the name gets changed to the Template, and not the Friendly Name.

I seem to remember that v8.2.0.4 was working fine too.

### EXPECTED BEHAVIOUR

The device name should be the expected one.

effelle, 2020-04-28:
By design on latest devs, see https://github.com/arendst/Tasmota/pull/8242 .

`Friendlyname1` is the name assigned to the first relay and can cause some troubles when you have more than a relay assigned on device.  
See[ this issue](https://github.com/home-assistant/core/issues/34432) on HA to better understanding why I changed it.

Two solutions: 
- Create a template with `Template 255` under Tasmota and change the name of it to one you prefer and then activate it on WebUI or using `module 0` command in console;
- Use rename function under device info in Home Assistant to change the name of the device, it will also ask if you want to change the root name for the sensors too.

richard-scott, 2020-04-28:
I did try creating my own Template but found that it will use that template name for both the Device and the Model name in HA. I have ended up renaming the device in Home Assistant and that works for me. 

It's just a pain that if you add multiple devices in one go, you have to go into HA and remember to rename them before adding the next one otherwise you have no idea which device is which.

effelle, 2020-04-28:
Noted, thanks for your feedback. 
For now is the only solution because use Friendlyname1 was not correct in the first place and that is an old story with a lot of discussion behind. 
Perhaps in the future I'll add a configurable device name.

richard-scott, 2020-04-28:
Maybe you could get it to send the hostname as the friendly name like ESPHome does.

richard-scott, 2020-05-16:
Wow. Upgrading from 8.2.0.4 to 8.3.0 is now impossible to keep your HA setup running nicely.

If you have SetOption19 1 set to enable Auto Discovery all your devices rename themselves to "Generic" (the Template I use), and then HA creates new sensors/switches etc as generic_switch.

Then, you can't rename anything, as they all share the same names.
My only upgrade path is:

1. SetOption19 0
2. Upgrade Device one at a time.
3. Rename device in HA.
4. SetOption19 1

I think a warning should go on the release notes as upgrading to the latest release is now a really bad end user experience  ;-(

richard-scott, 2020-05-16:
Ok, i have to say the "idea" behind renaming all your devices in HA is a great one... but the Upgrade requires that you rename all your devices to the SAME name before upgrading. That way HA knows to ignore the MQTT Sent names, and keeps the current ones.

Rock on Tasmota!! :-)

effelle, 2020-05-16:
Richard,
sorry if it gave you some extra work but sadly sometimes we must take some uncomfortable steps to archive a better experience.
If you look on first page of [docs](https://tasmota.github.io/docs/) the change of the name for Discovery is shown as breaking change, I have also added a note on Discovery docs advising the name of the device is now the name of the module/template.
Since is an important change in the structure, the best way to update all the device at the same time is to broadcast a `SetOption19 0`, setup all your devices with a new template name you like and then broadcast a `setoption19 1`. 
Alternatively, and that's my personal suggestion, delete the device under HA, disable the discovery on device, update its configuration and then enable the discovery again one by one.

richard-scott, 2020-05-16:
Actually, I've attempted adding a new device to a new setup and it still goes badly.
I'm using a Wemos D1 Mini, connected to a PZEM0016 via a R485/TTL board.

When I first type 'SetOption19 1' the single power monitor is added to HA, However, when I restart the Wemos D1, multiple sensors are added to HA.

For example this:

sensor.grid_monitor_house_energy_reactivepower

Gets changed to this:

sensor.grid_monitor_house_energy_reactivepower_2

It seems that ALL of my original sensors have been renamed to _2, and some new _1 ones have been created. 

I've even got some crazy ones such as this:

sensor.grid_monitor_house_energy_current_2_2

effelle, 2020-05-16:
I'm using it since I introduced it and I don't have this issue with HA 109.0 (didn't upgrade to latest yet).
Remove completely the integration from HA (not the single device) to start over with a fresh database. 
HA appends _1, _2 when the id of the device was already previously discovered.

richard-scott, 2020-05-17:
Normally I'd say yes, but my MQTT server says otherwise:

[image]

richard-scott, 2020-05-17:
Oh, and this is with a deleted, and re-installed MQTT server. So everything is fresh.

richard-scott, 2020-05-17:
Ah, this seems to be due to the Template Name being truncated to 14chars on a 8.2.0 -> 8.3.0 upgrade?

Before Upgrade at 8.2.0 the Template I created was called "Grid Monitor House" Module, and after when 8,3.0 booted it was just ""Grid Monitor H" Module,

richard-scott, 2020-05-17:
Now I don't see any new data ;-(

[image]

richard-scott, 2020-05-17:
This is what I get in 8.2.0:

[image]

effelle, 2020-05-17:
Let me check with the new HA release. Hold on.

effelle, 2020-05-17:
I can reproduce the issue with long names. 
I'll fix it ASAP.

PR #8470 (2020-05-17): HA Discovery: Oops! New DeviceName
## Description:
Disocovery now uses the (new) `DeviceName` instead of Module/Template name.
`DeviceName` defaults to `FriendlyName1` if not set and can be changed from console or from Configuration--> Configure Other --> Device Name. 

**Related issue (if applicable):** fixes #8298 

## Checklist:
  - [X] The pull request is done against the latest dev branch
  - [X] Only relevant files were touched
  - [X] Only one feature/fix was added per PR.
  - [X] The code change is tested and works on core ESP8266 V.2.7.1
  - [X] The code change is tested and works on  [... cut]
closed 2020-05-17 by arendst