arendst/Tasmota #8300: Fallback Topic not unique

pgScorpio, 2020-04-28:
From the manual:

> By default the Fallback Topic is DVES_XXXXXX_fb where xxxxxx is derived from the last 6 charactes of the device's MAC address (excluding :)

This means that we only use the second half of the MAC address. However the first 3 bytes of the MAC address are (or at least should be) vendor specific, and you would expect that all 'Espressif' devices have the same first 3 bytes in their MAC, but I discovered this is **not** the case (see picture, ). Espressive has multiple MAC ranges  (at least 2, see links.), so you **can** have two devices with the same Fallback Topic !

[image]

[https://hwaddress.com/oui-iab/C4-4F-33/](url)
[https://hwaddress.com/oui-iab/D8-F1-5B/](url)

So why don't we simply use the  **full** MAC address like MAC_XXXXXXXXXXXX as a fallback topic ?
since this should always be unique.

meingraham, 2020-04-28:
What is the probability that _you_ will have **_the_** two devices with the same least significant three bytes of these two Espressif MAC ranges? It's smaller than 1/(2*256^3) (~3e-8) because you have to also account for the population of people buying Espressif based devices and you being the one that gets the "duplicates". OK, it's not zero, but is it high enough for concern?

pgScorpio, 2020-04-28:
Hi meingraham, thanks for the quick reaction.

First of all a slight correction:

The chance is (number of tasmota devices on the network)  *  (number of 'Espressif' MAC ranges') / (2^24), (in my case about 4e-6 or 1 on 250000.)

Secondly: I **did** run into this problem (I wish I was that lucky in the lottery!) while trying to add a new sonoff on my network. And it took me hours to find out out what was going on (a esp12 on the network with the same lower 3 bytes in the mac !). 

Thirthly: With the exponentially growing number of Espressif based devices (I expect that there are already more than 2 Espressif MAC ranges.) the chances will only grow too 

Finally: Even if the chances are very low, it's so easy to prevent it, so why don't change it !

meingraham, 2020-04-28:
I never was much good with statistics :wink:

I still think you need to take into account the population. But, yes, if only you were as lucky as you appear to be unlucky :wink:

If you want this issue to be looked more critically, you need to open a "feature request" and fully complete the template so that the issue isn't automatically closed by the bot.

pgScorpio, 2020-04-28:
Yes I know you're right, but still I think it should be an easy fix that one day could save someone else some big problems too.

And I did think I fully completed the template, but no idea why it still got closed right away )-:)
just because I didn't fill in the non-applicable 'Tasmota binary firmware version number' ??

meingraham, 2020-04-28:
Adding the three bytes (six characters) may have other implications in how the fb topic is treated elsewhere and impact that code... but yes, could be minimal.

Even though it's superfluous, providing some of the information (e.g., `Status 0` which has lots of pertinent information) could be useful... and the bot is expecting that.

I'll try to re-open. But if I can't then please create a new issue with all the info you can... even if it might be N/A.

pgScorpio, 2020-04-28:
Thanks for reopening.
If purely the lenght of the fb topic would be an issue it could be shortened to just M_XXXXXXXXXXXX (same length)
but I guess that would not be the case.

pgScorpio, 2020-04-28:
FYI:

I just found out that Espressif already has 34 MAC ranges !
See https://hwaddress.com/company/espressif-inc/

stefanbode, 2020-05-13:
What other projects are doing is to use the ID from the splash and the ID from the mac as we do. In this case, it should be unique. I still had no problems and I have quite a lot ESP here. Class-D does not work anymore...

pgScorpio, 2020-05-14:
> In this case, it should be unique. 

No it isn't ! It should be if the manufacturer has only one mac range, espressif has already 34 mac ranges in use (and still growing)..

> I still had no problems and I have quite a lot ESP here

So do I, and I ran into this problem ! That's why I reported this issue.

> Class-D does not work anymore...

No idea what you mean with that...

ascillato2, 2020-05-26:
Closing this issue as it has been added by Theo. 

Thanks.

closed 2020-05-26 by ascillato2