arendst/Tasmota #8303: Power outage simulation causes Sonoff to lose it config

jonnimoura, 2020-04-29:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

I simulated a power outage, turning off and on the power supply of the Sonoff several times and on the third, fourth time, the Sonoff came back to the previus state, the same state after I've recorded using Tasmotizer lastest Tasmota bin file, version 8.2.

So, I've been asking myself because it was not possible to find similar issue here, if it's happening only with me, what I mean, is not such a kind of difficult to replicate the same issue...

It happens repeatily, it's jsut a matter of do what I described above.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): _____ Sonoff
- [x] Tasmota binary firmware version number used: _____ 8.2
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: _____ Tasmotizer
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Just turn off and turn on the power supply from the Sonoff several times quickly, trying to reproduce a power outage.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

Sonoff did not lose all your configuration

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Einstein67, 2020-04-29:
You can disable the "Fast Power Cycle Device Recovery" with Setoption65 1

https://tasmota.github.io/docs/Device-Recovery/#fast-power-cycle-device-recovery

ascillato2, 2020-04-29:
Closing this issue as it is **duplicated**. Sorry.

Please, read the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-04-29 by ascillato2

jonnimoura, 2020-04-29:
> You can disable the "Fast Power Cycle Device Recovery" with Setoption65 1
> 
> https://tasmota.github.io/docs/Device-Recovery/#fast-power-cycle-device-recovery

Thank you Einstein67, I'll try it. Sorry, but even reading the documentation was not possible for me to find the answer.