arendst/Tasmota #8305: Proposal: Precisely synced group messages

lmcd, 2020-04-29:
**Have you looked for this feature in other issues and in the docs?**  

Yes

**Is your feature request related to a problem? Please describe.**  

I’m using the new group feature to have a handful of lights all respond at once to dimming events. For my use case, it helps that the lights are precisely synced, but in practice, even under ideal network conditions there’s often one or two lights lagging behind around 10-15% of the time.

**Describe the solution you'd like**  

Here’s my rough idea for a solution:

When other devices are discovered in a group, we ping them continuously until we receive a response below a certain latency threshold (say 2ms). In the payload of the ping will be a ms-accurate uptime of the device. We’ll maintain a log of everyone else’s uptime relative to ours. Now even though every device’s uptime may be different, we’ll know exactly by how much at millisecond precision.

Next we can add a new command, such as `DevGroupSendSync` that will UDP broadcast to everyone in the group to execute a command at a specific timestamp 50ms from now. The recipient devices can all use information they have about the sender to resolve this timestamp accurately and execute at exactly the right time, regardless of network conditions. The 50ms is arbitrary and could be customisable. If the network is so unpredictable that 100ms ping times are common, then this value could be increased.

**Describe alternatives you've considered**  
NTP synchronisation. But I think this could be easier, more accurate and use the existing UDP polling code.

**Additional context**  
I’m new to Tasmota, and am still familiarising myself with its features and the code.

pcdiem, 2020-04-30:
The low level ICMP handling does not seem to support the timestamp message type and I don't see any way to include the local clock time in the payload of a low level ICMP reply.

I took an initial crack at adding logic to handle this right into the device groups module. It send timestamp request message to new members and calculates the round-trip time and clock offset estimate. It only adds 328 bytes to the code but, unfortunately, the round trip times as calculated by the high level device groups code via the UDP socket is never less than 47 or so milliseconds It doesn't seem like that's going give us "precisely synced group messages"

lmcd, 2020-04-30:
Thanks for looking at this so quick!

Is the 47ms wall you're running into a consequence of using UDP? Is there some underlying latency to UDP that I'm not aware of yet?

I guess an insanely hacky approach would be to encode the timestamp in the MAC address, but I don't think routers will like that :)

ascillato, 2020-04-30:
The ping time depends on how heavily is being used your wifi Network and how many services are running inside your router. So this time is changing. For example a normal common router have a fluctuating time between 20 to 80 ms ping response on normal usage (with less than 20 wifi devices) but if its ISP disconnects, while reconnecting to internet, the ping time to wifi devices goes from 200 ms to 1500 ms. On heavy usage (more than 20 devices or more than 1 device streaming video like a TV or a phone) this time goes from 40 to 200 ms.

So, a precisely sync over wifi UDP can't be warranted because you measure the response time but for the next sample it may be totally different.

lmcd, 2020-04-30:
I'm talking about measuring the response time only once, and only when it drops below a reasonable threshold. The device will then store that time and reuse it indefinitely to offset against its own time.

lmcd, 2020-04-30:
Also, I've been using a dedicated network (with no internet connection) to test my Tasmota setup. A router - with line of sight to the lights by a few meters, and 4 lights. Every 5th or 6th ping jumps from 4-5ms to 80-90ms. Some packets are occasionally lost too. On this dedicated network under ideal conditions, the group is often out of sync in a way that's noticeable to the naked eye.

pcdiem, 2020-04-30:
I can ping my Tasmota devices from my server with around 2ms rtt. I believe the high rtt I'm seeing is due to the higher level of the code. When I ping from my server, it's getting ICMP replies directly from the low-level network stack on the Tasmota device. For the device groups code, the request has to pass all the way up the stack and be queued. It sits there until the next loop pass when device groups checks to see if there are any packets available. If so, it dequeues the packet and processes it.

pcdiem, 2020-04-30:
BTW, on a full implementation, you can get the time (in ns) that the packet arrive at the low network stack level. There doesn't seem to a way to do this with the Arduino stack.

lmcd, 2020-04-30:
If the high-level UDP roundtrip is 20x that of ICMP, could that mean there's a bottleneck/architectural issue in the main loop of the app?

pcdiem, 2020-04-30:
I would think the right way to address this would be to have rules be able to be triggered based on a wall time including milliseconds and let the existing NTP logic handle keeping the clock accurate.

ascillato, 2020-04-30:
>I would think the right way to address this would be to have rules be able to be triggered based on a wall time including milliseconds and let the existing NTP logic handle keeping the clock accurate.

I agree because seems to be a lot easier to implement and the sync should be warranted by the NTP Sync. What some audio systems does to sync, is a pre-fetch. So they save internally what they have to play like for example 1 second of audio and then, they start playing at the same NTP time. While playing, they are receiving the next second of audio they will play. With that approach using NTP, they should start at the same time IMHO.

lmcd, 2020-04-30:
Do we know the margin of error with NTP? How accurate is it in practice, with the Tasmota stack? If different devices in a group have clocks out by 100ms, then it's not all that different from the current approach that often incurs similar delays over UDP.

pcdiem, 2020-04-30:
With all the IoT devices syncing to the same NTP server, I'd think the devices would stay in sync within a few ms of each other even if they're 100ms or more out of sync with real world time. In any case, NTP is going to be able to do as good a job as anything else. Also, if you only sync the clocks once, they will eventually drift.

closed 2020-05-30 by stale[bot]