arendst/Tasmota #8306: Fix HA Discovery regression to use Friendly Name again.

potto007, 2020-04-29:
### PROBLEM DESCRIPTION
The HA Discovery commit (3e47b6fe19a512181145da211e965ee0118e2814) introduced a regression to the HomeAssistant discovery interactions, resulting in the device name = model name. Upgrading to any firmware dated _after_ `20200421-120032-1065381-tasmota.bin` results in the device name being changed to the model name of that device within HomeAssistant.

### REQUESTED INFORMATION
- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): ZooZee SA101 & ZooZee SA102
- [x] Tasmota binary firmware version number used: 8.2.0.5; development HEAD
  - [x] Pre-compiled
  - [x] Self-compiled
    - [x] IDE / Compiler used: PlatformIO VSCode
- [x] Flashing tools used: OTA
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
21:27:16 MQT: stat/extra1/RESULT = {"NAME":"ZooZee","GPIO":[57,255,56,255,21,133,0,0,131,17,132,255,255],"FLAG":0,"BASE":18}
21:27:16 MQT: stat/extra1/RESULT = {"Module":{"0":"ZooZee"}}
21:27:16 MQT: stat/extra1/RESULT = {"GPIO0":{"57":"Led2i"},"GPIO1":{"0":"None"},"GPIO2":{"56":"Led1i"},"GPIO3":{"0":"None"},"GPIO4":{"21":"Relay1"},"GPIO5":{"133":"HLW8012 CF"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"131":"HLWBL SELi"},"GPIO13":{"17":"Button1"},"GPIO14":{"132":"HLWBL CF1"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```
21:27:56 MQT: stat/extra1/STATUS = {"Status":{"Module":0,"FriendlyName":["Extra 1"],"Topic":"extra1","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
21:27:56 MQT: stat/extra1/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/tasmota.bin.gz	","RestartReason":"Software/System restart","Uptime":"0T00:01:28","StartupUTC":"2020-04-29T20:26:28","Sleep":50,"CfgHolder":4617,"BootCount":19,"BCResetTime":"2020-04-09T19:39:28","SaveCount":125,"SaveAddress":"F9000"}}
21:27:56 MQT: stat/extra1/STATUS2 = {"StatusFWR":{"Version":"8.2.0.5(2e725b7-tasmota)","BuildDateTime":"2020-04-29T21:00:14","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"379/699"}}
21:27:56 MQT: stat/extra1/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Precise",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00028009","2805C8000100060000005A00000000000000","00000200","00000000"]}}
21:27:56 MQT: stat/extra1/STATUS4 = {"StatusMEM":{"ProgramSize":585,"Free":416,"Heap":24,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144068","FlashMode":3,"Features":["00000809","8FDAE797","043683A0","000000CD","010013C0","C000F981","00000024"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
21:27:56 MQT: stat/extra1/STATUS5 = {"StatusNET":{"Hostname":"extra1-1639","IPAddress":"10.4.0.238","Gateway":"10.4.0.1","Subnetmask":"255.255.255.0","DNSServer":"10.4.0.2","Mac":"80:7D:3A:7B:C6:67","Webserver":2,"WifiConfig":2,"WifiPower":17.0}}
21:27:56 MQT: stat/extra1/STATUS6 = {"StatusMQT":{"MqttHost":"homeassistant.ad.ottoops.com","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_7BC667","MqttUser":"mqtt","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
21:27:56 MQT: stat/extra1/STATUS7 = {"StatusTIM":{"UTC":"2020-04-29T20:27:56","Local":"2020-04-29T21:27:56","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"05:32","Sunset":"20:02"}}
21:27:56 MQT: stat/extra1/STATUS9 = {"StatusPTH":{"PowerDelta":0,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
21:27:56 MQT: stat/extra1/STATUS10 = {"StatusSNS":{"Time":"2020-04-29T21:27:56","ENERGY":{"TotalStartTime":"2020-04-09T19:21:36","Total":0.022,"Yesterday":0.000,"Today":0.006,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}}
21:27:56 MQT: stat/extra1/STATUS11 = {"StatusSTS":{"Time":"2020-04-29T21:27:56","Uptime":"0T00:01:28","UptimeSec":88,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Precise","BSSId":"78:8A:20:DD:17:D5","Channel":11,"RSSI":80,"Signal":-60,"LinkCount":1,"Downtime":"0T00:00:04"}}}
```
- [x] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
21:29:18 CMD: SetOption19 1
21:29:18 SRC: WebConsole from 10.4.0.209
21:29:18 CMD: Group 0, Index 19, Command "SETOPTION", Data "1"
21:29:18 MQT: stat/extra1/RESULT = {"SetOption19":"ON"}
21:29:19 MQT: homeassistant/sensor/7BC667_ENERGY_Total/config = {"name":"Extra 1 ENERGY Total","stat_t":"tele/extra1/SENSOR","avty_t":"tele/extra1/LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"7BC667_ENERGY_Total","dev":{"ids":["7BC667"]},"unit_of_meas":"kWh","dev_cla":"power","frc_upd":true,"val_tpl":"{{value_json['ENERGY']['Total']}}"} (retained)
21:29:19 MQT: homeassistant/sensor/7BC667_ENERGY_Yesterday/config = {"name":"Extra 1 ENERGY Yesterday","stat_t":"tele/extra1/SENSOR","avty_t":"tele/extra1/LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"7BC667_ENERGY_Yesterday","dev":{"ids":["7BC667"]},"unit_of_meas":"kWh","dev_cla":"power","frc_upd":true,"val_tpl":"{{value_json['ENERGY']['Yesterday']}}"} (retained)
21:29:19 MQT: homeassistant/sensor/7BC667_ENERGY_Today/config = {"name":"Extra 1 ENERGY Today","stat_t":"tele/extra1/SENSOR","avty_t":"tele/extra1/LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"7BC667_ENERGY_Today","dev":{"ids":["7BC667"]},"unit_of_meas":"kWh","dev_cla":"power","frc_upd":true,"val_tpl":"{{value_json['ENERGY']['Today']}}"} (retained)
21:29:19 MQT: homeassistant/sensor/7BC667_ENERGY_Power/config = {"name":"Extra 1 ENERGY Power","stat_t":"tele/extra1/SENSOR","avty_t":"tele/extra1/LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"7BC667_ENERGY_Power","dev":{"ids":["7BC667"]},"unit_of_meas":"W","dev_cla":"power","frc_upd":true,"val_tpl":"{{value_json['ENERGY']['Power']}}"} (retained)
21:29:20 MQT: homeassistant/sensor/7BC667_ENERGY_ApparentPower/config = {"name":"Extra 1 ENERGY ApparentPower","stat_t":"tele/extra1/SENSOR","avty_t":"tele/extra1/LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"7BC667_ENERGY_ApparentPower","dev":{"ids":["7BC667"]},"unit_of_meas":"VA","dev_cla":"power","frc_upd":true,"val_tpl":"{{value_json['ENERGY']['ApparentPower']}}"} (retained)
21:29:20 MQT: homeassistant/sensor/7BC667_ENERGY_ReactivePower/config = {"name":"Extra 1 ENERGY ReactivePower","stat_t":"tele/extra1/SENSOR","avty_t":"tele/extra1/LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"7BC667_ENERGY_ReactivePower","dev":{"ids":["7BC667"]},"unit_of_meas":"VAr","dev_cla":"power","frc_upd":true,"val_tpl":"{{value_json['ENERGY']['ReactivePower']}}"} (retained)
21:29:20 MQT: homeassistant/sensor/7BC667_ENERGY_Factor/config = {"name":"Extra 1 ENERGY Factor","stat_t":"tele/extra1/SENSOR","avty_t":"tele/extra1/LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"7BC667_ENERGY_Factor","dev":{"ids":["7BC667"]},"unit_of_meas":"Cos φ","ic":"mdi:alpha-f-circle-outline","frc_upd":true,"val_tpl":"{{value_json['ENERGY']['Factor']}}"} (retained)
21:29:20 MQT: homeassistant/sensor/7BC667_ENERGY_Voltage/config = {"name":"Extra 1 ENERGY Voltage","stat_t":"tele/extra1/SENSOR","avty_t":"tele/extra1/LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"7BC667_ENERGY_Voltage","dev":{"ids":["7BC667"]},"unit_of_meas":"V","ic":"mdi:alpha-v-circle-outline","frc_upd":true,"val_tpl":"{{value_json['ENERGY']['Voltage']}}"} (retained)
[... cut ...]
The Tasmota device is listed in HA Device list using its Model Name; which is the Template Name, when in the Module config. Incidentally, changing this value _does_ correct the name seen in HA, but it also changes the type of device to be the same as that name. Ie: instead of "Sonoff Basic", the model would be "Test 1".

### SCREENSHOTS
[image]
[image]
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

blakadder, 2020-04-29:
This is the designed behaviour for next stable release.

closed 2020-04-29 by blakadder

effelle, 2020-04-29:
And will be extended on sensors name too as soon as I'll finish the new light management.

potto007, 2020-04-29:
I guess I'll ask about something like this in chat next time, first. Lesson learned.

effelle, 2020-04-29:
no problem at all, I thank you for the effort and the time you spent for it.

Jibun-no-Kage, 2022-07-01:
How do you query for the friendly name using MQTT only?  is there a cmnd subscriber on TASMOTA that will return INFO1,2, etc. on demand?

I have been watching my MQTT broker and reviewing the message logging and other than initial power up of TASMOTA firmware INFO1,2, etc. never happens?  No way to trigger it, thus no easy way to get the friendly name?

barbudor, 2022-07-01:
What about sending the 'FriendlyName' command to Tasmota and get the result?

sfromis, 2022-07-01:
In addition to the directly applicable command, the "friendly names" are also included in the output of the `Status` command, and in the `tasmota/discovery/<mac>/config` message published to MQTT when connecting - not in INFO1/2/3.