arendst/Tasmota #8309: Toggling up/down Buttons in Shutter mode

herrlado, 2020-04-30:
### PROBLEM DESCRIPTION
When enabled shutter mode with the option 80 the SwitchMode 0 does not any more work for the Buttons. With ShuttonButton there is not option to stop moving the shutter when pressed again. Generally I did not found the possibility how to stop the shutter when moving, without firing ShutterClose event,

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _Sonof Dual R2 rev 1.4
- [ ] Tasmota binary firmware version number used: ___8.3__
  - [ x] Pre-compiled
  - [ x] Self-compiled
    - [ms code with platformio ] IDE / Compiler used: _____
- [ ] Flashing tools used: __platformio___
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
02:09:19 MQT: rollowaschkueche/stat/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
02:09:19 MQT: rollowaschkueche/stat/RESULT = {"Module":{"39":"Sonoff Dual R2"}}
02:09:19 MQT: rollowaschkueche/stat/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"11":"Switch3"},"GPIO4":{"0":"None"},"GPIO5":{"22":"Relay2"},"GPIO9":{"0":"None"},"GPIO10":{"12":"Switch4"},"GPIO12":{"21":"Relay1"},"GPIO13":{"56":"Led1i"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
02:09:56 MQT: rollowaschkueche/stat/STATUS = {"Status":{"Module":39,"FriendlyName":["rollowaschkuechehoch","rollowaschkuecherunter"],"Topic":"rollowaschkueche","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"aussenlicht","SwitchMode":[0,0,1,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
02:09:56 MQT: rollowaschkueche/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"1T00:57:28","StartupUTC":"2020-04-28T23:12:28","Sleep":50,"CfgHolder":4619,"BootCount":13,"BCResetTime":"2020-04-27T12:09:20","SaveCount":168,"SaveAddress":"F4000"}}
02:09:56 MQT: rollowaschkueche/stat/STATUS2 = {"StatusFWR":{"Version":"8.2.0(tasmota)","BuildDateTime":"2020-04-27T10:38:27","Boot":7,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8285","CR":"437/699"}}
02:09:56 MQT: rollowaschkueche/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"genlap.home","LogPort":514,"SSId":["malenik","genlap"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["000AC009","2805C8000100060000005A00000000000000","40000209","00000000"]}}
02:09:56 MQT: rollowaschkueche/stat/STATUS4 = {"StatusMEM":{"ProgramSize":577,"Free":424,"Heap":22,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","000000CD","010013C0","C000F981","00000004"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
02:09:56 MQT: rollowaschkueche/stat/STATUS5 = {"StatusNET":{"Hostname":"rollowaschkueche","IPAddress":"192.168.1.28","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"DC:4F:22:xx:xx:xx","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
02:09:56 MQT: rollowaschkueche/stat/STATUS6 = {"StatusMQT":{"MqttHost":"xxxx.home","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_D85922","MqttUser":"xxxx","MqttCount":2,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
02:09:56 MQT: rollowaschkueche/stat/STATUS7 = {"StatusTIM":{"UTC":"2020-04-30T00:09:56","Local":"2020-04-30T02:09:56","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":99,"Sunrise":"05:53","Sunset":"20:41"}}
02:09:56 MQT: rollowaschkueche/stat/STATUS10 = {"StatusSNS":{"Time":"2020-04-30T02:09:56","Switch3":"ON","Switch4":"ON","Shutter1":{"Position":43,"Direction":0,"Target":43}}}
02:09:56 MQT: rollowaschkueche/stat/STATUS11 = {"StatusSTS":{"Time":"2020-04-30T02:09:56","Uptime":"1T00:57:28","UptimeSec":89848,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":2,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"xxxxx","BSSId":"B4:FB:E4:xx:xx:xx","Channel":11,"RSSI":74,"Signal":-63,"LinkCount":2,"Downtime":"0T00:00:14"}}}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Enable Shutters. Toggle does not work for the Webbuttons.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2020-05-06:
@stefanbode Any idea how to address this?

malbinola, 2020-05-06:
I'm using Tasmota 8.2 with shutter driver and all seems working as expected.
From the Web interface, after starting an open or close action, i just need to press any button (Up or Down) to immediately stop the movement.
Maybe depends on how ShutterButtons are configured?

This is my setup:
`ShutterButton1 1 100 50 - - - - - - -`
`ShutterButton1 2 0 50 - - - - - - -`

herrlado, 2020-05-06:
@malbinola hmm ok, i will check again. But do you have configured to have up down arrows or default on / off?

LotharF, 2020-05-08:
I have almost the same problem. If I install the V8.2 on my D1mini "Shutter testboard" it takes a lot of time to toggle the buttons. On my Sonoff T1 it is working a little bit fastern with V8.2 and the same settings I use on my test board.. 
But even here the button does not react immediately when you press it again.

to-scho, 2020-05-08:
Ein Tasmota Rollo in der Waschküche, was es nicht alles gibt ;-)

For clarification, web buttons to control a shutter are independent from ShutterButton configuration.

Is the web interface responsive while shutter is moving? You may e.g. enter console etc. 

Can you please provide a console log while the problem occurs? Please type in "weblog 4" before to enable more debug output. You can revert using "weblog 2" afterwards. Please try to give an indication when you clicked the web button during the log.  Maybe you can use two browser windows, one for clicking, one to observe the console.

malbinola, 2020-05-10:
> @malbinola hmm ok, i will check again. But do you have configured to have up down arrows or default on / off?

I have Up/Down arrows but I have not set any particular configuration, they appears after enabling shutter mode with `SetOption80 1`

stefanbode, 2020-05-13:
Yes, the up/down button in the web interface where introduced some time ago instead of the relais button. With these buttons the behavior is now completely different. Before a press again on relay1 stops a moving shutter. now the buttons send shutteropen or shutterclose to the driver. I have seen that somebody added some nice commands like stopopen and stopclose. I think, if we reassign the the up/down webbuttons with these functions your wanted behavior should be there. I will check and provide a test on my branch. If it is ok, we can commit it to the master

stefanbode, 2020-05-13:
With 8.2 the webserver.ino already has the new command implemented to stopopen/stopclose

```
#ifdef USE_SHUTTER
      int32_t ShutterWebButton;
      if (ShutterWebButton = IsShutterWebButton(device)) {
        snprintf_P(svalue, sizeof(svalue), PSTR("ShutterPosition%d %s"), abs(ShutterWebButton), (ShutterWebButton>0) ? PSTR(D_CMND_SHUTTER_STOPOPEN) : PSTR(D_CMND_SHUTTER_STOPCLOSE));
        ExecuteWebCommand(svalue, SRC_WEBGUI);
      } else {
```

Therefore my assumption is that if the shutter is running and you hit the up or down again it will just stop as you requested.

stefanbode, 2020-05-13:
I double checked with the current version and it exactly behave as you requested. an additional click on the up/down button during the movement stops the shutter. There is NO definition required of shutterbuttons. Just OOTB standard configuration

herrlado, 2020-05-13:
Ok, seems my fault. Sorry for this issue. I have just rechecked this and it works as expected. Maybe the problem was with my mobile browser or so, so yes, it works as expected.

meingraham, 2020-05-13:
Question answered. Closing issue.

closed 2020-05-13 by meingraham