arendst/Tasmota #8310: BME280 (also BMP280, BME680) option set sample rate

GrendelWWU, 2020-04-30:
**Have you looked for this feature in other issues and in the docs?**  
Yes, but I didn't find an option.

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
As far, as I know, the connected I2C sensors like BME280 (in my case) are triggered every second. (The LED on my ESP01 (==>SCLK) is flashing every second.) So, the reported temperatures are inaccurate. (mentioned in Bosch data sheet!) I'd like to change the frequency, the sensor is read. For my purposes, every 30 seconds will be very fine.

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._
I'd like to have a SETOPTION to change the interval for reading I2C sensors. Let's give it a value from 1 to 300 in seconds.

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._
I don't know any...

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2020-04-30:
This is asked and answered before. 

As rules are checked every several seconds for sensor changes to act on there is no way not to read the sensor data that often.

GrendelWWU, 2020-04-30:
At the moment, I don't need rules. In my special application, I'm using ESP01 just as climate sensors. These sensors are asked by a ArduiTouch/ESP system with a modified color-weather-station (original by Daniel Eichhorn). I'm sending a HTTP-Request (STATUS 8 afair) and interpreting the JSON return values. This is running pretty cool, but the sensor temperature values aren't really accurate. I really like your Tasmota software. You're doing a great job. I don't have intentions to invent the wheel a second time. ;-)
I attached a picture of my device, so you can see, what the hell I'm doing...
Yes, and the next step will be switching Tasmota actuators with this device, sending a Power Toggle or something like that isn't really difficult.

[image]

DD8HL, 2020-04-30:
Hi,
It might be quick and dirty, but maybe you could change the i2c adress of the rule that reads the sensor for 29 cycles and only change the i2c adress to the correct one for the cylce 30 ? I dont know much about programming so that may be nonsense.
If not you may go the hardware route and add a transistor to break one of the i2c lines to the bme280

Jason2866, 2020-04-30:
Both is not possible. If you cut a i2c line the sensor is after connecting in undefined state
and needs to be initialized. Initializing is done ONE time at startup.
Anyway i have many BME280 and BMP280 and the do work fine with Tasmota.
Real life testing. There is NO heating caused from Tasmota code. The change you want is snake oil.

sfromis, 2020-04-30:
The data sheet specifically suggest 1 sample / second for weather monitoring. Do you have a specific reference to backup the thing about "the reported temperatures are inaccurate. (mentioned in Bosch data sheet!)".

The closest I saw was a reminder of potential heating from PCB, sensor element and internal ambient temperature, but not a warning about sampling rate.

ascillato2, 2020-05-04:
Hi,

Any news on this?

GrendelWWU, 2020-05-05:
Hello, at the moment I don't have the time to make some tests with higher and lower sample-rate. I will do later this month. I will play with increasing/decreasing the rate and I'll see, what will happen.

ascillato2, 2020-05-06:
> I will play with increasing/decreasing the rate and I'll see, what will happen.

Great. Please, tell us your results. Thanks.

> I don't have the time to make some tests with higher and lower sample-rate. I will do later this month.

Putting this issue on hold until then.

closed 2020-05-06 by ascillato2