arendst/Tasmota #8312: Get configuration at runtime from external JSON response

ortegafernando, 2020-04-30:
**Have you looked for this feature in other issues and in the docs?**  
Yes

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
I will be great the possibility of getting some setttings from a JSON response of a web page (in each power up of the device, I mean, at runtime)

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._
You can configure a webpage from which TASMOTA will get a JSON response with some configurations. This way, you can change your MQTT broker (for example) in a centralized way.

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._
DDNS is not a solution for your MQTT broker, sometimes, we also change the port.

**Additional context**  
_Add any other context or screenshots about the feature request here._
I am designing a home system for a house without broadband connection. I will use a mobile phone.
I am interested in configuring the MQTT server of my SONOFF's and other Tasmota's devices in runtime, that is, reading the MQTT broker configuration from a web page or service (AWS's lambda probably)
You will need to configure the web page in your tasmota configuration at first.
NOTE: DDNS is not useful in this case since the MQTT broker will be on a mobile (as Nodered, installed in Termux), and I will use the free version of ngrok. As Ngrok changes everytime I reboot the phone, I must serach from my Tasmota's the real Ngrok tunnel address and port.

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2020-04-30:
https://github.com/arendst/Tasmota/issues/7135

closed 2020-04-30 by ascillato2

ortegafernando, 2020-05-01:
HI, I think I have not explain well my request.

I have already read the information that you sent me. All of this stuff needs to be in the same net as your tasmotas.

They also need (MQTT commands) to know and be connected with an MQTT broker with which you could send commands.

My problem / my request is to configure INITIALLY the MQTT configuration. After, as you said, you have option to configure tasmota device in "remote mode".

That is why I am asking for an option to read, at least, MQTT configuration from a URL. 

In my case, that configuration changes a lot, and I am not in the same net as my tasmotas, so I need a URL from which my tasmota could download the MQTT broker configuration everytime they wake up or every X hours (just in case MQTT broker configuration has changed).

I think all of your links don't cover this problem, do they?

I hope you could reopen this request. It would be perfect to check that URL every time TASMOTA wakes up and every X hours. With only MQTT configuration, we could use afterwards MQTT commands, nodered , .... to send all other settings to our tasmotas, but MQTT broker config is a must.

Thanks.

Jason2866, 2020-05-01:
You can configure mqtt wifi (and other) settings when you flash your device with Tasmotizer
There will be no change made in Tasmota.

ortegafernando, 2020-05-01:
Hi, I am not speaking about changing them when flashing device, I am speaking about getting the MQTT configuration from a known webpage.  I know that this is a request and can be develop or not, but what I want know is it must be understood or it must be well explained.
I think is a good option from some cases.

meingraham, 2020-05-01:
`http://<IP>/dl?` will initiate a download of the dmp (binary). You can then run decode-config against that file at any time afterwards to convert it to JSON.

meingraham, 2020-05-01:
If you want non-interactive - `wget <IP>/dl` - saves the configuration to a file named `dl`

Jason2866, 2020-05-01:
Broker config change is no problem. You dont need to input fixed IP addresses.
Why not using for mqtt a entry in Tasmota device like `mqtt.lan` ?
And why should broker setup change? 
Tbh i dont get your use case.

ortegafernando, 2020-05-02:
HI, thanks a lot for your respones. What I propose is NOT to have the config backed up, I want a way to configure my tasmota far away from them in an automatic way. Al least, MQTT, as after connecting to MQTT they could be configured.
Imagine that my MQTT broker is installed in TERMUX in a mobile PHONE (that is why I can get normal broanband connection in that location). I want to use a personal mqtt broker. 
I need to connect to that Broker several local nodes (easy) and other nodes in other location.

How those nodes know the address and port of the MQTT broker if every time the phone starts or resets the NGROK application change both IP and port?

In those nodes (not local ones) could get the information of the MQTT broker from a webpage (my personal web page which address doesn't change) I will have the problem solved.
When my mobile restart or detect that NGROK has changed IP or PORT, the phone will publish new values in my webpage, so I could check all config information from my webpage, and "those nodes" could get the MQTT settings automatically.

What we need is an option in MQTT configuration of our tasmota to get the config from an url instead of taking ip and port.

If you don't like it, ok, but I want to explain it well.

Thanks.

meingraham, 2020-05-02:
Currently Tasmota does not poll to discover MQTT brokers on its network... thus your feature request. If Tasmota were to implement this feature, how would it determine which broker to choose? Some environments have more than one MQTT broker active. OK, in your case you may only have one broker at any time. But how would this feature work if enabled by the user and the user has multiple brokers? This might wreak havoc. And trust me, users will make inadvertent errors.

The only workaround I can think of in your case is to have a script that sends the Tasmota MQTT setup commands to each of the local network Tasmota devices. Where that script would run; the mobile phone which is running the MQTT broker as well? Assuming that the Tasmota devices are connected to the local Wi-Fi network (even if they are not connected to the local MQTT broker), you'll either have to know the IP address or the hostname **_of each Tasmota device_** on which you want to reset the MQTT broker information. If it is connected to Wi-Fi and you can use HTTP to send the commands to each device to set up MQTT (once you have the MQTT host IP or name and the connection credentials).

https://tasmota.github.io/docs/Commands/#with-web-requests