arendst/Tasmota #8316: TasmotaSlave - problem.

TrDA-hab, 2020-05-01:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
"TasmotaSlave" - does not work
I did a lot of experiments. I lost a lot of my free time.
I tested:
  - 8.2.0.5
  - 7.1.2.6

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x] Device used (e.g., Sonoff Basic): _____
- [ x] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ x] Flashing tools used:PlatformioIDE
- [ x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
21:02:20 CMD: Backlog Template; Module; GPIO 255
21:02:20 RSL: stat/tasmota/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
21:02:20 RSL: stat/tasmota/RESULT = {"Module":{"18":"Generic"}}
21:02:20 RSL: stat/tasmota/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"185":"Slave RX"},"GPIO2":{"186":"Slave RST"},"GPIO3":{"184":"Slave TX"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [x ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
21:03:32 CMD: Status 0
21:03:32 RSL: stat/tasmota/STATUS = {"Status":{"Module":18,"FriendlyName":["Tasmota"],"Topic":"tasmota","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
21:03:32 RSL: stat/tasmota/STATUS1 = {"StatusPRM":{"Baudrate":57600,"GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Power On","Uptime":"0T00:31:03","StartupUTC":"2020-05-01T19:32:29","Sleep":50,"CfgHolder":4617,"BootCount":38,"SaveCount":54,"SaveAddress":"F5000"}}
21:03:32 RSL: stat/tasmota/STATUS2 = {"StatusFWR":{"Version":"7.1.2.6(tasmota)","BuildDateTime":"2020-05-01T19:17:21","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"276/1151"}}
21:03:32 RSL: stat/tasmota/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Cisco-3G",""],"TelePeriod":60,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
21:03:32 RSL: stat/tasmota/STATUS4 = {"StatusMEM":{"ProgramSize":587,"Free":160,"Heap":27,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640EF","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","00007881"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,29,31","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
21:03:32 RSL: stat/tasmota/STATUS5 = {"StatusNET":{"Hostname":"tasmota-1326","IPAddress":"192.168.1.6","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"EC:FA:BC:6E:25:2E","Webserver":2,"WifiConfig":4}}
21:03:32 RSL: stat/tasmota/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_6E252E","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
21:03:32 RSL: stat/tasmota/STATUS7 = {"StatusTIM":{"UTC":"Fri May  1 20:03:32 2020","Local":"Fri May  1 21:03:32 2020","StartDST":"Sun Mar 29 02:00:00 2020","EndDST":"Sun Oct 25 03:00:00 2020","Timezone":"+01:00","Sunrise":"05:28","Sunset":"20:05"}}
21:03:32 RSL: stat/tasmota/STATUS10 = {"StatusSNS":{"Time":"2020-05-01T21:03:32"}}
21:03:32 RSL: stat/tasmota/STATUS11 = {"StatusSTS":{"Time":"2020-05-01T21:03:32","Uptime":"0T00:31:03","UptimeSec":1863,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"Cisco-3G","BSSId":"58:6D:8F:83:AC:94","Channel":11,"RSSI":82,"Signal":-59,"LinkCount":1,"Downtime":"0T00:00:06"}}}
21:03:36 RSL: tele/tasmota/STATE = {"Time":"2020-05-01T21:03:36","Uptime":"0T00:31:07","UptimeSec":1867,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"Cisco-3G","BSSId":"58:6D:8F:83:AC:94","Channel":11,"RSSI":74,"Signal":-63,"LinkCount":1,"Downtime":"0T00:00:06"}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
https://github.com/andrethomas/TasmotaSlave/issues/3

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
[image]
[image]
[image]
[image]
### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arovak, 2020-05-01:
For my project I have been sniffing the traffic between the slave and tasmota. This is what I'm seeing.

Tasmota 8.2.0.1 and Tasmota 8.1.0.5 working fine, I see this traffic send to the slave when doing 
 slavesend command and status 8 command to get the slave tele.
```
 tasmota 8.2.0.1
baudrate 38400
ü[05][1A][00][00]ýþp[0].b[0].bg_color=#ff0000ÿü[02][00][00][00]ýü[02][00][00][00]ý

tasmota 8.1.0.5
baudrate 115200
ü[05][1A][00][00]ýþp[0].b[0].bg_color=#ff0000ÿü[02][00][00][00]ýü[02][00][00][00]ý
```
On Tasmota 8.2.0.6 data send to the slave is different. Same commands send as above.
```
Tasmota 8.2.0.6  

baudrate 38400
þ[05][1A][00][00]ýþp[0].b[0].bg_color=#ff0000ÿþ[02][00][00][00]ýþ[02][00][00][00]ý

baudrate 115200
ÿ[05][1A][00][00]ýþp[0].b[0].bg_color=#ff0000ÿÿ[02][00][00][00]ýÿ[02][00][00][00]ý
ÿ[05][1A][00][00]ýþp[0].b[0].bg_¬±½Éõ#ff0000ÿÿ[02][00][00][00]ýÿ[02][00][00][00]ý
```

Data send out from the slave however does come through to tasmota, although somewhat garbled.
I have the slave do publish command.
```23:04:16 MQT: hasp/state/page = 1
23:04:19 MQT: hasp/state/json = {"p[1].b[1].event":"DOWN"}
23:04:19 MQT: hasp/state/idle = OFF
23:04:19 MQT: hasp/state/json = {"p[1].b[1].ev�nt":"SHORT"}
23:04:19 MQT: hasp/state/json = {"p[1].b[1].event":"UP"}
23:05:22 MQT: hasp/state/idle = SHORT
```
If I can try anything to test let me know, I'm continuing to work with one of my older build.

arendst, 2020-05-02:
So to make sure, v7.1.2.6 functions OK, 8.2.0.5 doesn't function correctly?

TrDA-hab, 2020-05-02:
No.
Both 7.1.2.6 and 8.2.0.5 do NOT work correctly.

TrDA-hab, 2020-05-02:
I can replace the Arduino Pro Mini with the Arduino Nano and repeat the test. If you are interested.

arendst, 2020-05-02:
Ok.

I have no Nano connected and never used it but I noticed with your current setup you use Software Serial. If you swap TX/RX it should switch to Hardware serial which, I suspect would work fine.

[image]

 Again, I have no nano attached but I see this on the serial connected console with hardware serial:

```
0c 08 21 90 84 0a 4a 52 4a 8f 21 21 a5 48 0f bd  ..!„.JRJ!!¥H.½
0a 31 e0 38 a8 d6 88 84 a5 53 88 88 3b 4a 48 24  .1à8¨Öˆ„¥Sˆˆ;JH$
ca 84 0b cc 9f 62 8c 98 9c 21 80 4e 11 e6 48 4e  Ê„.ÌŸbŒ˜œ!€N.æHN
11 f4 0a 6a 88 c4 af 0d 4a 87 4a 4a 24 bd 0a e2  .ô.jˆÄ¯.J‡JJ$½.â
d6 6b 4a a8 9d 31 b0 a8 0c c2 ed 84 8c f8        ÖkJ¨1°¨.Âí„Œø  
fc 01 00 00 00 fd                                ü....ý          
fc 01 00 00 00 fd                                ü....ý          
fc 01 00 00 00 fd                                ü....ý          
fc 01 00 00 00 fd                                ü....ý          
fc 01 00 00 00 fd                                ü....ý          
fc 01 00 00 00 fd                                ü....ý      
```
I don't see the ``fc 01 00 00 00 fd`` so I suspect it does better work than software serial.

Could you give it a try?

TrDA-hab, 2020-05-02:
I can try it. 
What version of Tasmota should I use?

arendst, 2020-05-02:
Try whatever you have active, they all supported hardware serial. I suggest the latest though.

TrDA-hab, 2020-05-02:
7.1.2.6

00:00:00 CFG: Loaded from flash at F8, Count 59
00:00:00 QPC: Flag 0E
00:00:00 SRC: Restart
00:00:00 Project tasmota Tasmota Version 7.1.2.6(tasmota)-STAGE
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Connecting to AP1 Cisco-3G in mode 11N as tasmota-1326...
00:00:01 WIF: Checking connection...
00:00:01 WIF: Attempting connection...
00:00:01 CMD: TrDA3
00:00:01 SRC: Serial
00:00:01 CMD: Group 0, Index 3, Command "TRDA", Data ""
00:00:01 RSL: stat/tasmota/RESULT = {"Command":"Unknown"}
00:00:02 WIF: Checking connection...
00:00:02 WIF: Attempting connection...
00:00:02 CMD: TrDA3
00:00:02 SRC: Serial
00:00:02 CMD: Group 0, Index 3, Command "TRDA", Data ""
00:00:02 RSL: stat/tasmota/RESULT = {"Command":"Unknown"}
00:00:04 WIF: Checking connection...
00:00:04 WIF: Attempting connection...
00:00:04 CMD: TrDA3
00:00:04 SRC: Serial
00:00:04 CMD: Group 0, Index 3, Command "TRDA", Data ""
00:00:04 RSL: stat/tasmota/RESULT = {"Command":"Unknown"}
00:00:04 CMD: TrDA3
00:00:04 SRC: Serial
00:00:04 CMD: Group 0, Index 3, Command "TRDA", Data ""
00:00:04 RSL: stat/tasmota/RESULT = {"Command":"Unknown"}
00:00:05 WIF: Checking connection...
00:00:05 WIF: Connected
00:00:05 HTP: Web server active on tasmota-1326 with IP address 192.168.1.6
00:00:05 CMD: TrDA3
00:00:05 SRC: Serial
00:00:05 CMD: Group 0, Index 3, Command "TRDA", Data ""
00:00:05 RSL: stat/tasmota/RESULT = {"Command":"Unknown"}
11:56:46 CMD: TrDA3
11:56:46 SRC: Serial
11:56:46 CMD: Group 0, Index 3, Command "TRDA", Data ""
11:56:46 RSL: stat/tasmota/RESULT = {"Command":"Unknown"}
11:56:46 NTP: Drift 0, (UTC) Sat May  2 10:56:46 2020, (DST) Sun Mar 29 02:00:00 2020, (STD) Sun Oct 25 03:00:00 2020
11:56:46 QPC: Reset
11:56:47 CMD: TrDA3
11:56:47 SRC: Serial
11:56:47 CMD: Group 0, Index 3, Command "TRDA", Data ""
11:56:47 RSL: stat/tasmota/RESULT = {"Command":"Unknown"}
11:56:47 HTP: Main Menu
11:56:48 CMD: TrDA3
11:56:48 SRC: Serial
11:56:48 CMD: Group 0, Index 3, Command "TRDA", Data ""
11:56:48 RSL: stat/tasmota/RESULT = {"Command":"Unknown"}
11:56:48 APP: Boot Count 42
11:56:48 CFG: Saved to flash at F7, Count 60, Bytes 4096
11:56:48 HTP: Console
11:56:49 CMD: TrDA3
11:56:49 SRC: Serial
11:56:49 CMD: Group 0, Index 3, Command "TRDA", Data ""
11:56:49 RSL: stat/tasmota/RESULT = {"Command":"Unknown"}
11:56:49 SNS: Hardware Serial
11:56:49 Tasmota Slave Enabled
11:56:50 Tasmota Slave Version 1095004756 not supported!
11:57:05 WIF: Checking connection...
11:57:05 WIF: Connected

TrDA-hab, 2020-05-02:
8.2.0.5

00:00:00 CFG: Loaded from flash at FB, Count 9
00:00:00 Project tasmota Tasmota Version 8.2.0.5(tasmota)-STAGE
00:00:00 WIF: Connecting to AP1 Cisco-3G Channel 11 BSSId 58:6D:8F:83:AC:94 in mode 11N as tasmota_6E252E-1326...
00:00:01 WIF: Connected
00:00:01 HTP: Web server active on tasmota_6E252E-1326 with IP address 192.168.1.6
14:05:13 RSL: tele/tasmota_6E252E/INFO1 = {"Module":"Generic","Version":"8.2.0.5(tasmota)","FallbackTopic":"cmnd/DVES_6E252E_fb/","GroupTopic":"cmnd/tasmotas/"}
14:05:13 RSL: tele/tasmota_6E252E/INFO2 = {"WebServerMode":"Admin","Hostname":"tasmota_6E252E-1326","IPAddress":"192.168.1.6"}
14:05:13 RSL: tele/tasmota_6E252E/INFO3 = {"RestartReason":"Power On"}
14:05:17 RSL: tele/tasmota_6E252E/STATE = {"Time":"2020-05-02T14:05:17","Uptime":"0T00:00:09","UptimeSec":9,"Heap":30,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"Cisco-3G","BSSId":"58:6D:8F:83:AC:94","Channel":11,"RSSI":78,"Signal":-61,"LinkCount":1,"Downtime":"0T00:00:03"}}
14:05:19 SNS: Hardware Serial
14:05:19 Tasmota Slave Enabled
14:05:20 Tasmota Slave Version 1095004756 not supported!

TrDA-hab, 2020-05-02:
[image]

TrDA-hab, 2020-05-02:
Now I will try 8.2.0.6 (I downloaded it now, it is clean).

TrDA-hab, 2020-05-02:
8.2.0.6

00:00:00 CFG: Loaded from flash at FB, Count 9
00:00:00 Project tasmota Tasmota Version 8.2.0.6(tasmota)-2_7_0
00:00:00 WIF: Connecting to AP1 Cisco-3G Channel 11 BSSId 58:6D:8F:83:AC:94 in mode 11N as tasmota_6E252E-1326...
00:00:01 WIF: Connected
00:00:01 HTP: Web server active on tasmota_6E252E-1326 with IP address 192.168.1.6
00:00:03 RSL: tele/tasmota_6E252E/INFO1 = {"Module":"Generic","Version":"8.2.0.6(tasmota)","FallbackTopic":"cmnd/DVES_6E252E_fb/","GroupTopic":"cmnd/tasmotas/"}
00:00:03 RSL: tele/tasmota_6E252E/INFO2 = {"WebServerMode":"Admin","Hostname":"tasmota_6E252E-1326","IPAddress":"192.168.1.6"}
00:00:03 RSL: tele/tasmota_6E252E/INFO3 = {"RestartReason":"Power On"}
00:00:07 RSL: tele/tasmota_6E252E/STATE = {"Time":"1970-01-01T00:00:07","Uptime":"0T00:00:09","UptimeSec":9,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":23,"MqttCount":0,"Wifi":{"AP":1,"SSId":"Cisco-3G","BSSId":"58:6D:8F:83:AC:94","Channel":11,"RSSI":84,"Signal":-58,"LinkCount":1,"Downtime":"0T00:00:03"}}
00:00:09 SNS: Hardware Serial
00:00:09 Tasmota Slave Enabled
00:00:10 Tasmota Slave Version 1095004756 not supported!

TrDA-hab, 2020-05-02:
7.1.2

00:00:00 CFG: Loaded from flash at F5, Count 7
00:00:00 Project tasmota Tasmota Version 7.1.2(tasmota)-STAGE
00:00:00 WIF: Connecting to AP1 Cisco-3G in mode 11N as tasmota-1326...
00:00:05 WIF: Connected
00:00:05 HTP: Web server active on tasmota-1326 with IP address 192.168.1.6
12:29:24 SNS: Hardware Serial
12:29:24 Tasmota Slave Enabled
12:29:25 Tasmota Slave Version 1095004756 not supported!

TrDA-hab, 2020-05-02:
Can I still help?

arendst, 2020-05-02:
Ever tried to swap the wires too (humbly asking)

TrDA-hab, 2020-05-02:
Do I need to swap Tx and Rx wires? It is easy. Now I will do it.

TrDA-hab, 2020-05-02:
00:00:00 CFG: Loaded from flash at F7, Count 13
00:00:00 Project tasmota Tasmota Version 7.1.2(tasmota)-STAGE
00:00:00 WIF: Connecting to AP1 Cisco-3G in mode 11N as tasmota-1326...
00:00:05 WIF: Connected
00:00:05 HTP: Web server active on tasmota-1326 with IP address 192.168.1.6
12:42:43 SNS: Hardware Serial
12:42:43 Tasmota Slave Enabled
12:42:43 Tasmota Slave Version 248 not supported!

arendst, 2020-05-02:
At least the version has changed :-(

Anyway. I will hookup a nano too and dive in the wonderful world of Tasmota Slave.

NB. I understand the nano needs at least the TasmotaSlave library too to be able to respond with the correct version being ``20191129`` as that is the string to be tested. You have that installed on the Nano?

TrDA-hab, 2020-05-02:
00:00:00 CFG: Loaded from flash at F5, Count 7
00:00:00 Project tasmota Tasmota Version 8.2.0.6(tasmota)-2_7_0
00:00:00 WIF: Connecting to AP1 Cisco-3G Channel 11 BSSId 58:6D:8F:83:AC:94 in mode 11N as tasmota_6E252E-1326...
00:00:01 WIF: Connected
00:00:01 HTP: Web server active on tasmota_6E252E-1326 with IP address 192.168.1.6
12:47:47 RSL: tele/tasmota_6E252E/INFO1 = {"Module":"Generic","Version":"8.2.0.6(tasmota)","FallbackTopic":"cmnd/DVES_6E252E_fb/","GroupTopic":"cmnd/tasmotas/"}
12:47:47 RSL: tele/tasmota_6E252E/INFO2 = {"WebServerMode":"Admin","Hostname":"tasmota_6E252E-1326","IPAddress":"192.168.1.6"}
12:47:47 RSL: tele/tasmota_6E252E/INFO3 = {"RestartReason":"Software/System restart"}
12:47:51 RSL: tele/tasmota_6E252E/STATE = {"Time":"2020-05-02T12:47:51","Uptime":"0T00:00:09","UptimeSec":9,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":31,"MqttCount":0,"Wifi":{"AP":1,"SSId":"Cisco-3G","BSSId":"58:6D:8F:83:AC:94","Channel":11,"RSSI":98,"Signal":-51,"LinkCount":1,"Downtime":"0T00:00:03"}}
12:47:53 SNS: Hardware Serial
12:47:53 Tasmota Slave Enabled
12:48:54 CMD: SlaveSend ON
12:48:54 RSL: stat/tasmota_6E252E/RESULT = {"SlaveSend":"Done"}
12:50:11 LOG: SerialLog 0, WebLog 2, MqttLog 0, SysLog 0, LogHost , LogPort 514, TelePeriod 60
12:50:11 RSL: tele/tasmota_6E252E/STATE = {"Time":"2020-05-02T12:50:11","Uptime":"0T00:02:29","UptimeSec":149,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"Cisco-3G","BSSId":"58:6D:8F:83:AC:94","Channel":11,"RSSI":90,"Signal":-55,"LinkCount":1,"Downtime":"0T00:00:03"}}
12:50:11 Tasmota Slave Version 29 not supported!

TrDA-hab, 2020-05-02:
Now I am using Arduino Pro Mini (3.3V 8MHZ Atmega328P).
If necessary, I can install NANO, it will take some time.

arendst, 2020-05-02:
No. Just leave it for now.

I expect the last 29 you received means the comms is fine but just the characters don't make it. So the hardware is connected OK. You might want to try a lower baudrate but the Hardware Serial should have been able to handle this without error.

TrDA-hab, 2020-05-02:
What speed options should I test?

arendst, 2020-05-02:
28800 and 9600

TrDA-hab, 2020-05-02:
#define USE_TASMOTA_SLAVE
#define USE_TASMOTA_SLAVE_FLASH_SPEED 57600
#define USE_TASMOTA_SLAVE_SERIAL_SPEED 28800

00:00:00 CFG: Loaded from flash at F7, Count 13
00:00:00 Project tasmota Tasmota Version 8.2.0.6(tasmota)-2_7_0
00:00:00 WIF: Connecting to AP1 Cisco-3G Channel 11 BSSId 58:6D:8F:83:AC:94 in mode 11N as tasmota_6E252E-1326...
00:00:01 WIF: Connected
00:00:01 HTP: Web server active on tasmota_6E252E-1326 with IP address 192.168.1.6
13:01:40 RSL: tele/tasmota_6E252E/INFO1 = {"Module":"Generic","Version":"8.2.0.6(tasmota)","FallbackTopic":"cmnd/DVES_6E252E_fb/","GroupTopic":"cmnd/tasmotas/"}
13:01:40 RSL: tele/tasmota_6E252E/INFO2 = {"WebServerMode":"Admin","Hostname":"tasmota_6E252E-1326","IPAddress":"192.168.1.6"}
13:01:40 RSL: tele/tasmota_6E252E/INFO3 = {"RestartReason":"Power On"}
13:01:44 RSL: tele/tasmota_6E252E/STATE = {"Time":"2020-05-02T13:01:44","Uptime":"0T00:00:09","UptimeSec":9,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"Cisco-3G","BSSId":"58:6D:8F:83:AC:94","Channel":11,"RSSI":80,"Signal":-60,"LinkCount":1,"Downtime":"0T00:00:03"}}
13:01:46 SNS: Hardware Serial
13:01:46 Tasmota Slave Enabled
13:02:34 CMD: SlaveSend ON
13:02:34 RSL: stat/tasmota_6E252E/RESULT = {"SlaveSend":"Done"}
13:02:44 RSL: tele/tasmota_6E252E/STATE = {"Time":"2020-05-02T13:02:44","Uptime":"0T00:01:09","UptimeSec":69,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":24,"MqttCount":0,"Wifi":{"AP":1,"SSId":"Cisco-3G","BSSId":"58:6D:8F:83:AC:94","Channel":11,"RSSI":80,"Signal":-60,"LinkCount":1,"Downtime":"0T00:00:03"}}
13:02:44 Tasmota Slave Version 29 not supported!

arendst, 2020-05-02:
It looks as if there is someting basics wrong other than comms reliability. 

As said, I'm trying to regain my Arduino environment and flash a nano mini.

TrDA-hab, 2020-05-02:
Should I synchronously change the speed in the firmware of the slave? Or just in the ESP8266?

arendst, 2020-05-02:
I would think both.

TrDA-hab, 2020-05-02:
Serial.begin(9600);

#define USE_TASMOTA_SLAVE 
  #define USE_TASMOTA_SLAVE_FLASH_SPEED 57600 
  #define USE_TASMOTA_SLAVE_SERIAL_SPEED 9600

00:00:00 CFG: Loaded from flash at F5, Count 7
00:00:00 Project tasmota Tasmota Version 8.2.0.6(tasmota)-2_7_0
00:00:00 WIF: Connecting to AP1 Cisco-3G Channel 11 BSSId 58:6D:8F:83:AC:94 in mode 11N as tasmota_6E252E-1326...
00:00:01 WIF: Connected
00:00:01 HTP: Web server active on tasmota_6E252E-1326 with IP address 192.168.1.6
13:11:43 RSL: tele/tasmota_6E252E/INFO1 = {"Module":"Generic","Version":"8.2.0.6(tasmota)","FallbackTopic":"cmnd/DVES_6E252E_fb/","GroupTopic":"cmnd/tasmotas/"}
13:11:43 RSL: tele/tasmota_6E252E/INFO2 = {"WebServerMode":"Admin","Hostname":"tasmota_6E252E-1326","IPAddress":"192.168.1.6"}
13:11:43 RSL: tele/tasmota_6E252E/INFO3 = {"RestartReason":"Software/System restart"}
13:11:47 RSL: tele/tasmota_6E252E/STATE = {"Time":"2020-05-02T13:11:47","Uptime":"0T00:00:09","UptimeSec":9,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"Cisco-3G","BSSId":"58:6D:8F:83:AC:94","Channel":11,"RSSI":86,"Signal":-57,"LinkCount":1,"Downtime":"0T00:00:03"}}
13:11:49 SNS: Hardware Serial
13:11:49 Tasmota Slave Enabled
13:12:26 CMD: SlaveSend ON
13:12:26 RSL: stat/tasmota_6E252E/RESULT = {"SlaveSend":"Done"}
13:14:11 CMD: SlaveSend ON
13:14:11 RSL: stat/tasmota_6E252E/RESULT = {"SlaveSend":"Done"}
13:14:33 LOG: SerialLog 0, WebLog 2, MqttLog 0, SysLog 0, LogHost , LogPort 514, TelePeriod 60
13:14:34 RSL: tele/tasmota_6E252E/STATE = {"Time":"2020-05-02T13:14:34","Uptime":"0T00:02:56","UptimeSec":176,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":22,"MqttCount":0,"Wifi":{"AP":1,"SSId":"Cisco-3G","BSSId":"58:6D:8F:83:AC:94","Channel":11,"RSSI":86,"Signal":-57,"LinkCount":1,"Downtime":"0T00:00:03"}}
13:14:34 Tasmota Slave Version 29 not supported!

TrDA-hab, 2020-05-02:
Serial.begin(28800);

#define USE_TASMOTA_SLAVE
#define USE_TASMOTA_SLAVE_FLASH_SPEED 57600
#define USE_TASMOTA_SLAVE_SERIAL_SPEED 28800

00:00:00 CFG: Loaded from flash at F5, Count 7
00:00:00 Project tasmota Tasmota Version 8.2.0.6(tasmota)-2_7_0
00:00:00 WIF: Connecting to AP1 Cisco-3G Channel 11 BSSId 58:6D:8F:83:AC:94 in mode 11N as tasmota_6E252E-1326...
00:00:03 WIF: Connected
00:00:03 HTP: Web server active on tasmota_6E252E-1326 with IP address 192.168.1.6
13:22:34 RSL: tele/tasmota_6E252E/INFO1 = {"Module":"Generic","Version":"8.2.0.6(tasmota)","FallbackTopic":"cmnd/DVES_6E252E_fb/","GroupTopic":"cmnd/tasmotas/"}
13:22:34 RSL: tele/tasmota_6E252E/INFO2 = {"WebServerMode":"Admin","Hostname":"tasmota_6E252E-1326","IPAddress":"192.168.1.6"}
13:22:34 RSL: tele/tasmota_6E252E/INFO3 = {"RestartReason":"Software/System restart"}
13:22:38 RSL: tele/tasmota_6E252E/STATE = {"Time":"2020-05-02T13:22:38","Uptime":"0T00:00:11","UptimeSec":11,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"Cisco-3G","BSSId":"58:6D:8F:83:AC:94","Channel":11,"RSSI":76,"Signal":-62,"LinkCount":1,"Downtime":"0T00:00:05"}}
13:22:38 SNS: Hardware Serial
13:22:38 Tasmota Slave Enabled
13:22:38 Tasmota Slave Version 29 not supported!
13:22:58 LOG: SerialLog 0, WebLog 2, MqttLog 0, SysLog 0, LogHost , LogPort 514, TelePeriod 30

TrDA-hab, 2020-05-02:
Can I still help?

arendst, 2020-05-02:
No need. I have hooked it up and see the same error as you do.

I will investigate the communication between the two. Will take some time.

TrDA-hab, 2020-05-02:
I understand, I will wait.

I get this error even if I do not physically connect the slave:
13:29:37 SNS: Hardware Serial
13:29:37 Tasmota Slave Enabled
13:30:05 Tasmota Slave Version 29 not supported!

arendst, 2020-05-02:
I just loaded the tasmota-slave blink.ino on a, what I know relearned Arduino Pro Mini 3V3, and see the led blink when data is transfered between the two devices. SO comms is there just the protocol is wrong.

Investigating...

arovak, 2020-05-02:
I can confirm that switching to hardware serial works for me. 
```
14:30:05 SNS: Hardware Serial
14:30:05 Tasmota Slave Enabled
14:30:05 Tasmota Slave Version 20191129
14:31:05 MQT: stat/tasmotaslave/STATUS8 = {"StatusSNS":{"Time":"2020-05-02T14:31:05","Switch1":"OFF","TasmotaSlave":{"status":"available","version":"0.1.0","uptime":815,"espCanUpdate":"false","page":0,"numPages":4,"tftDriver":"ILI9341","tftWidth":240,"tftHeight":320}}}
```
Also tele message and commands send from the slave come through OK now.

[image]

I am using STM32F4 board as a slave. and am using baudrate 38400.

arendst, 2020-05-02:
With external power and a config like this 

[image]

I'm able to run a stable blink on the Arduino Pro mini 3V3.

Using hardware serial like 

[image]

It's even more stable:
```
00:00:05 HTP: Web server active on wemos8 with IP address 192.168.2.172
17:28:36 NTP: UTC 2020-05-02T15:28:35, DST 2020-03-29T02:00:00, STD 2020-10-25T03:00:00
17:28:36 QPC: Reset
17:28:36 HTP: Main Menu
17:28:37 MQT: Attempting connection...
17:28:37 MQT: Connected
17:28:37 MQT: tele/wemos8/LWT = Online (retained)
17:28:37 MQT: cmnd/wemos8/POWER = 
17:28:37 MQT: Subscribe to cmnd/wemos8/#
17:28:37 MQT: Subscribe to cmnd/tasmotas/#
17:28:37 MQT: Subscribe to cmnd/DVES_17E067_fb/#
17:28:37 MQT: tele/wemos8/INFO1 = {"Module":"Generic","Version":"8.2.0.6(theo)","FallbackTopic":"cmnd/DVES_17E067_fb/","GroupTopic":"cmnd/tasmotas/"}
17:28:37 MQT: tele/wemos8/INFO2 = {"WebServerMode":"Admin","Hostname":"wemos8","IPAddress":"192.168.2.172"}
17:28:37 MQT: tele/wemos8/INFO3 = {"RestartReason":"Software/System restart"}
17:28:37 MQT: stat/wemos8/RESULT = {"POWER":"OFF"}
17:28:37 MQT: stat/wemos8/POWER = OFF
17:28:38 APP: Boot Count 67
17:28:38 CFG: Saved to flash at F8, Count 118, Bytes 4096
17:28:39 SNS: Hardware Serial
17:28:39 Tasmota Slave Enabled
17:28:41 MQT: tele/wemos8/STATE = {"Time":"2020-05-02T17:28:41","Uptime":"0T00:00:13","UptimeSec":13,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":25,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":2,"SSId":"indebuurt_IoT","BSSId":"18:E8:29:CA:17:C1","Channel":11,"RSSI":100,"Signal":-37,"LinkCount":1,"Downtime":"0T00:00:07"}}
17:28:41 DMP: 99 17 34 01 02 00 00 00
17:28:41 Tasmota Slave Version 20191129
```
The extra DMP: line is now shown with ``weblog 4``

TrDA-hab, 2020-05-02:
I'm testing it now. It does not take much time.

TrDA-hab, 2020-05-02:
I downloaded the firmware again. I am not getting an error now. But Blink is not working.

You have a version:
Version":"8.2.0.6(theo)"

I have a version:
Version":"**8.2.0.6(tasmota)*

TrDA-hab, 2020-05-02:
00:00:00 CFG: Loaded from flash at FA, Count 18
00:00:00 QPC: Flag 0E
00:00:00 CFG: CR 327/699
00:00:00 SRC: Restart
00:00:00 Project tasmota Tasmota Version 8.2.0.6(tasmota)-2_7_0
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Connecting to AP1 Cisco-3G Channel 11 BSSId 58:6D:8F:83:AC:94 in mode 11N as tasmota_6E252E-1326...
00:00:01 WIF: Checking connection...
00:00:01 WIF: Connected
00:00:01 HTP: Web server active on tasmota_6E252E-1326 with IP address 192.168.1.6
17:40:30 NTP: UTC 2020-05-02T16:40:29, DST 2020-03-29T02:00:00, STD 2020-10-25T03:00:00
17:40:30 RSL: tele/tasmota_6E252E/INFO1 = {"Module":"Generic","Version":"8.2.0.6(tasmota)","FallbackTopic":"cmnd/DVES_6E252E_fb/","GroupTopic":"cmnd/tasmotas/"}
17:40:30 RSL: tele/tasmota_6E252E/INFO2 = {"WebServerMode":"Admin","Hostname":"tasmota_6E252E-1326","IPAddress":"192.168.1.6"}
17:40:30 RSL: tele/tasmota_6E252E/INFO3 = {"RestartReason":"Software/System restart"}
17:40:30 RSL: stat/tasmota_6E252E/RESULT = {"POWER":"OFF"}
17:40:30 RSL: stat/tasmota_6E252E/POWER = OFF
17:40:33 QPC: Reset
17:40:33 HTP: Main Menu
17:40:34 RSL: tele/tasmota_6E252E/STATE = {"Time":"2020-05-02T17:40:34","Uptime":"0T00:00:09","UptimeSec":9,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Cisco-3G","BSSId":"58:6D:8F:83:AC:94","Channel":11,"RSSI":88,"Signal":-56,"LinkCount":1,"Downtime":"0T00:00:03"}}
17:40:35 APP: Boot Count 10
17:40:35 CFG: Saved to flash at F9, Count 19, Bytes 4096
17:40:36 SNS: Hardware Serial
17:40:36 Tasmota Slave Enabled
17:40:42 HTP: Console
17:40:48 WIF: Checking connection...
17:41:04 RSL: tele/tasmota_6E252E/STATE = {"Time":"2020-05-02T17:41:04","Uptime":"0T00:00:39","UptimeSec":39,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":33,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Cisco-3G","BSSId":"58:6D:8F:83:AC:94","Channel":11,"RSSI":82,"Signal":-59,"LinkCount":1,"Downtime":"0T00:00:03"}}
17:41:08 WIF: Checking connection...
17:41:28 WIF: Checking connection...
17:41:34 RSL: tele/tasmota_6E252E/STATE = {"Time":"2020-05-02T17:41:34","Uptime":"0T00:01:09","UptimeSec":69,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":27,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Cisco-3G","BSSId":"58:6D:8F:83:AC:94","Channel":11,"RSSI":84,"Signal":-58,"LinkCount":1,"Downtime":"0T00:00:03"}}
17:41:48 WIF: Checking connection...
17:42:04 RSL: tele/tasmota_6E252E/STATE = {"Time":"2020-05-02T17:42:04","Uptime":"0T00:01:39","UptimeSec":99,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":28,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Cisco-3G","BSSId":"58:6D:8F:83:AC:94","Channel":11,"RSSI":78,"Signal":-61,"LinkCount":1,"Downtime":"0T00:00:03"}}
17:42:08 WIF: Checking connection...
17:42:18 HTP: Main Menu
17:42:20 HTP: Information
17:42:28 WIF: Checking connection...
17:42:34 RSL: tele/tasmota_6E252E/STATE = {"Time":"2020-05-02T17:42:34","Uptime":"0T00:02:09","UptimeSec":129,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":26,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Cisco-3G","BSSId":"58:6D:8F:83:AC:94","Channel":11,"RSSI":84,"Signal":-58,"LinkCount":1,"Downtime":"0T00:00:03"}}
17:42:42 HTP: Main Menu
17:42:45 HTP: Console

TrDA-hab, 2020-05-02:
For the experiment, I again changed the P and T cable.

TrDA-hab, 2020-05-02:
00:00:00 CFG: Loaded from flash at F9, Count 19
00:00:00 QPC: Flag 0E
00:00:00 CFG: CR 327/699
00:00:00 SRC: Restart
00:00:00 Project tasmota Tasmota Version 8.2.0.6(tasmota)-2_7_0
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Connecting to AP1 Cisco-3G Channel 11 BSSId 58:6D:8F:83:AC:94 in mode 11N as tasmota_6E252E-1326...
00:00:01 WIF: Checking connection...
00:00:01 WIF: Connected
00:00:01 HTP: Web server active on tasmota_6E252E-1326 with IP address 192.168.1.6
17:50:58 NTP: UTC 2020-05-02T16:50:57, DST 2020-03-29T02:00:00, STD 2020-10-25T03:00:00
17:50:58 RSL: tele/tasmota_6E252E/INFO1 = {"Module":"Generic","Version":"8.2.0.6(tasmota)","FallbackTopic":"cmnd/DVES_6E252E_fb/","GroupTopic":"cmnd/tasmotas/"}
17:50:58 RSL: tele/tasmota_6E252E/INFO2 = {"WebServerMode":"Admin","Hostname":"tasmota_6E252E-1326","IPAddress":"192.168.1.6"}
17:50:58 RSL: tele/tasmota_6E252E/INFO3 = {"RestartReason":"Power On"}
17:50:58 RSL: stat/tasmota_6E252E/RESULT = {"POWER":"OFF"}
17:50:58 RSL: stat/tasmota_6E252E/POWER = OFF
17:51:01 QPC: Reset
17:51:02 RSL: tele/tasmota_6E252E/STATE = {"Time":"2020-05-02T17:51:02","Uptime":"0T00:00:09","UptimeSec":9,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Cisco-3G","BSSId":"58:6D:8F:83:AC:94","Channel":11,"RSSI":80,"Signal":-60,"LinkCount":1,"Downtime":"0T00:00:03"}}
17:51:03 APP: Boot Count 11
17:51:03 CFG: Saved to flash at F8, Count 20, Bytes 4096
17:51:04 SNS: Hardware Serial
17:51:04 Tasmota Slave Enabled
17:51:05 Tasmota Slave Version 1095004756 not supported!
17:51:07 DMP: C9 17 34 01 02 00 00 00
17:51:08 DMP: C9 17 34 01 02 00 00 00
17:51:09 DMP: C9 17 34 01 02 00 00 00
17:51:10 DMP: C9 17 34 01 02 00 00 00
17:51:10 HTP: Configuration
17:51:11 DMP: C9 17 34 01 02 00 00 00
17:51:12 DMP: C9 17 34 01 02 00 00 00
17:51:13 DMP: C9 17 34 01 02 00 00 00
17:51:13 HTP: Main Menu
17:51:14 DMP: C9 17 34 01 02 00 00 00
17:51:15 DMP: C9 17 34 01 02 00 00 00
17:51:16 HTP: Console
17:51:16 DMP: C9 17 34 01 02 00 00 00
17:51:16 WIF: Checking connection...
17:51:17 DMP: C9 17 34 01 02 00 00 00
17:51:18 DMP: C9 17 34 01 02 00 00 00
17:51:19 DMP: C9 17 34 01 02 00 00 00
17:51:20 DMP: C9 17 34 01 02 00 00 00
17:51:21 DMP: C9 17 34 01 02 00 00 00
17:51:22 DMP: C9 17 34 01 02 00 00 00

TrDA-hab, 2020-05-02:
The software series also does not give an error. Blink doesn't work either.

TrDA-hab, 2020-05-02:
Hardware Serial
Rx-Tx
Tx-Rx
External power

00:00:00 CFG: Loaded from flash at F8, Count 28
00:00:00 QPC: Flag 0E
00:00:00 CFG: CR 327/699
00:00:00 SRC: Restart
00:00:00 Project tasmota Tasmota Version 8.2.0.6(tasmota)-2_7_0
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Connecting to AP1 Cisco-3G Channel 11 BSSId 58:6D:8F:83:AC:94 in mode 11N as tasmota_6E252E-1326...
00:00:01 WIF: Checking connection...
00:00:01 WIF: Connected
00:00:01 HTP: Web server active on tasmota_6E252E-1326 with IP address 192.168.1.6
18:31:04 NTP: UTC 2020-05-02T17:31:03, DST 2020-03-29T02:00:00, STD 2020-10-25T03:00:00
18:31:04 RSL: tele/tasmota_6E252E/INFO1 = {"Module":"Generic","Version":"8.2.0.6(tasmota)","FallbackTopic":"cmnd/DVES_6E252E_fb/","GroupTopic":"cmnd/tasmotas/"}
18:31:04 RSL: tele/tasmota_6E252E/INFO2 = {"WebServerMode":"Admin","Hostname":"tasmota_6E252E-1326","IPAddress":"192.168.1.6"}
18:31:04 RSL: tele/tasmota_6E252E/INFO3 = {"RestartReason":"Software/System restart"}
18:31:04 RSL: stat/tasmota_6E252E/RESULT = {"POWER":"ON"}
18:31:04 RSL: stat/tasmota_6E252E/POWER = ON
18:31:07 QPC: Reset
18:31:07 HTP: Main Menu
18:31:08 RSL: tele/tasmota_6E252E/STATE = {"Time":"2020-05-02T18:31:08","Uptime":"0T00:00:09","UptimeSec":9,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":24,"MqttCount":0,"POWER":"ON","Wifi":{"AP":1,"SSId":"Cisco-3G","BSSId":"58:6D:8F:83:AC:94","Channel":11,"RSSI":90,"Signal":-55,"LinkCount":1,"Downtime":"0T00:00:03"}}
18:31:09 APP: Boot Count 14
18:31:09 CFG: Saved to flash at F7, Count 29, Bytes 4096
18:31:10 SNS: Hardware Serial
18:31:10 Tasmota Slave Enabled
18:31:12 DMP: C9 17 34 01 02 00 00 00
18:31:12 Tasmota Slave Version 20191177 not supported!
18:31:13 DMP: C9 17 34 01 02 00 00 00
18:31:14 DMP: C9 17 34 01 02 00 00 00
18:31:15 DMP: C9 17 34 01 02 00 00 00

TrDA-hab, 2020-05-02:
Hardware Serial
Rx-Rx
Tx-Tx
57600-57600
External power
Blink.ino

00:00:00 CFG: Loaded from flash at F6, Count 14
00:00:00 QPC: Flag 0E
00:00:00 CFG: CR 327/699
00:00:00 SRC: Restart
00:00:00 Project tasmota Tasmota Version 8.2.0.6(tasmota)-2_7_0
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Connecting to AP1 Cisco-3G Channel 11 BSSId 58:6D:8F:83:AC:94 in mode 11N as tasmota_6E252E-1326...
00:00:01 WIF: Checking connection...
00:00:01 WIF: Connected
00:00:01 HTP: Web server active on tasmota_6E252E-1326 with IP address 192.168.1.6
18:50:58 NTP: UTC 2020-05-02T17:50:57, DST 2020-03-29T02:00:00, STD 2020-10-25T03:00:00
18:50:58 RSL: tele/tasmota_6E252E/INFO1 = {"Module":"Generic","Version":"8.2.0.6(tasmota)","FallbackTopic":"cmnd/DVES_6E252E_fb/","GroupTopic":"cmnd/tasmotas/"}
18:50:58 RSL: tele/tasmota_6E252E/INFO2 = {"WebServerMode":"Admin","Hostname":"tasmota_6E252E-1326","IPAddress":"192.168.1.6"}
18:50:58 RSL: tele/tasmota_6E252E/INFO3 = {"RestartReason":"External System"}
18:51:01 QPC: Reset
18:51:02 RSL: tele/tasmota_6E252E/STATE = {"Time":"2020-05-02T18:51:02","Uptime":"0T00:00:09","UptimeSec":9,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"Cisco-3G","BSSId":"58:6D:8F:83:AC:94","Channel":11,"RSSI":82,"Signal":-59,"LinkCount":1,"Downtime":"0T00:00:03"}}
18:51:03 APP: Boot Count 7
18:51:03 CFG: Saved to flash at F5, Count 15, Bytes 4096
18:51:04 SNS: Hardware Serial
18:51:04 Tasmota Slave Enabled
18:51:16 WIF: Checking connection...
18:51:36 WIF: Checking connection...
18:51:56 WIF: Checking connection...

TrDA-hab, 2020-05-02:
Software Serial
Rx-D1
Tx-D2
57600-57600
External power
Blink.ino

00:00:00 CFG: Loaded from flash at FB, Count 17
00:00:00 QPC: Flag 0E
00:00:00 CFG: CR 327/699
00:00:00 SRC: Restart
00:00:00 Project tasmota Tasmota Version 8.2.0.6(tasmota)-2_7_0
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Connecting to AP1 Cisco-3G Channel 11 BSSId 58:6D:8F:83:AC:94 in mode 11N as tasmota_6E252E-1326...
00:00:01 WIF: Checking connection...
00:00:01 WIF: Connected
00:00:01 HTP: Web server active on tasmota_6E252E-1326 with IP address 192.168.1.6
18:58:15 NTP: UTC 2020-05-02T17:58:14, DST 2020-03-29T02:00:00, STD 2020-10-25T03:00:00
18:58:15 RSL: tele/tasmota_6E252E/INFO1 = {"Module":"Generic","Version":"8.2.0.6(tasmota)","FallbackTopic":"cmnd/DVES_6E252E_fb/","GroupTopic":"cmnd/tasmotas/"}
18:58:15 RSL: tele/tasmota_6E252E/INFO2 = {"WebServerMode":"Admin","Hostname":"tasmota_6E252E-1326","IPAddress":"192.168.1.6"}
18:58:15 RSL: tele/tasmota_6E252E/INFO3 = {"RestartReason":"Software/System restart"}
18:58:18 QPC: Reset
18:58:18 HTP: Main Menu
18:58:19 RSL: tele/tasmota_6E252E/STATE = {"Time":"2020-05-02T18:58:19","Uptime":"0T00:00:09","UptimeSec":9,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":22,"MqttCount":0,"Wifi":{"AP":1,"SSId":"Cisco-3G","BSSId":"58:6D:8F:83:AC:94","Channel":11,"RSSI":78,"Signal":-61,"LinkCount":1,"Downtime":"0T00:00:03"}}
18:58:20 APP: Boot Count 9
18:58:20 CFG: Saved to flash at FA, Count 18, Bytes 4096
18:58:21 Tasmota Slave Enabled
18:58:21 HTP: Console
18:58:33 WIF: Checking connection...
18:58:53 WIF: Checking connection...
18:59:13 WIF: Checking connection...

TrDA-hab, 2020-05-02:
Software Serial
Rx-D2
Tx-D1
57600-57600
External power
Blink.ino

00:00:00 CFG: Loaded from flash at F9, Count 27
00:00:00 QPC: Flag 0E
00:00:00 CFG: CR 327/699
00:00:00 SRC: Restart
00:00:00 Project tasmota Tasmota Version 8.2.0.6(tasmota)-2_7_0
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Connecting to AP1 Cisco-3G Channel 11 BSSId 58:6D:8F:83:AC:94 in mode 11N as tasmota_6E252E-1326...
00:00:01 WIF: Checking connection...
00:00:01 WIF: Connected
00:00:01 HTP: Web server active on tasmota_6E252E-1326 with IP address 192.168.1.6
19:07:49 NTP: UTC 2020-05-02T18:07:48, DST 2020-03-29T02:00:00, STD 2020-10-25T03:00:00
19:07:49 RSL: tele/tasmota_6E252E/INFO1 = {"Module":"Generic","Version":"8.2.0.6(tasmota)","FallbackTopic":"cmnd/DVES_6E252E_fb/","GroupTopic":"cmnd/tasmotas/"}
19:07:49 RSL: tele/tasmota_6E252E/INFO2 = {"WebServerMode":"Admin","Hostname":"tasmota_6E252E-1326","IPAddress":"192.168.1.6"}
19:07:49 RSL: tele/tasmota_6E252E/INFO3 = {"RestartReason":"Software/System restart"}
19:07:52 QPC: Reset
19:07:52 HTP: Main Menu
19:07:53 RSL: tele/tasmota_6E252E/STATE = {"Time":"2020-05-02T19:07:53","Uptime":"0T00:00:09","UptimeSec":9,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"Cisco-3G","BSSId":"58:6D:8F:83:AC:94","Channel":11,"RSSI":66,"Signal":-67,"LinkCount":1,"Downtime":"0T00:00:03"}}
19:07:54 APP: Boot Count 17
19:07:54 CFG: Saved to flash at F8, Count 28, Bytes 4096
19:07:55 Tasmota Slave Enabled
19:07:56 HTP: Console
19:08:07 WIF: Checking connection...

TrDA-hab, 2020-05-02:
I'm out of ideas. Nothing works.

arovak, 2020-05-02:
Did you try at 9600 for tasmota and the slave? 
At that speed software and hardware serial work for me.

When I increase the speed to 57600 hardware serial keeps working but software serial does not.

With software serial communication from tasmota to the slave is broken, communication from the slave to tasmota still works. Which is why, in my case, tasmota still sees the slave on start up.
Software serial 9600, tasmota sees the slave and communication works 2-ways.
```
20:16:09 Tasmota Slave Enabled
20:16:09 DMP: 99 17 34 01 09 00 00 00
20:16:09 Tasmota Slave Version 20191129
```
Software serial 57600, tasmota sees the slave but can only receive slave communication. The slave does not see the commands tasmota sends.
```
20:20:32 Tasmota Slave Enabled
20:20:32 DMP: 99 17 34 01 09 00 00 00
20:20:32 Tasmota Slave Version 20191129
```

The only thing I see is that the character which indicates the start of the command is different when using software serial and some higher speed than 9600.

TrDA-hab, 2020-05-03:
Hardware Serial - work !!!
Blink.ino - work !!!

Rx-Tx
Tx-Rx
57600-57600
External power
weblog 4

13:22:00 SNS: Hardware Serial
13:22:00 Tasmota Slave Enabled
13:22:02 DMP: 99 17 34 01 02 00 00 00
13:22:02 Tasmota Slave Version 20191129

TrDA-hab, 2020-05-03:
Software Serial .... test

TrDA-hab, 2020-05-03:
Software Serial - work !!!
Blink.ino - work !!!

Rx-D1
Tx-D2
28800-28800
External power
weblog 4

13:44:38 Tasmota Slave Enabled
13:44:40 DMP: 99 17 34 01 02 00 00 00
13:44:40 Tasmota Slave Version 20191129

TrDA-hab, 2020-05-03:
I completely deleted all the Arduino IDE libraries. And installed it again. 
Now everything works fine.

closed 2020-05-03 by TrDA-hab