arendst/Tasmota #8321: Feature Request: Ability to rename switches and relays in Tasmota Firmware

austwhite, 2020-05-02:
**Have you looked for this feature in other issues and in the docs?**  
I was unable to find an answer via Google Search and documentation.
**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
Currently, when using Auto-Discovery, the Tasmota firmware sends the switch names as switch.tasmota, switch.tasmota1 etc.   This is not overly helpful in finding which switch is which in Home Assistant.   I know you can rename entities in Home Assistant, but I would like a way to name them in Tasmota firmware on the ESP device so it sends the correct name via discovery.
IE: I want to be able to name the switch "Kitchen Socket"  in Tasmota so it is sent automatically and detected by Home Assistant as "Kitchen Socket" instead if "Tasmota".
**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._
I have considered naming in Home Assistant, but this is sometimes trial and error, especially if multiple devices are setup at once as you then have to go to each "Tasmota" switch and test it to see what it corresponds to.
**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-05-02:
Home assistant have the option to rename those entities using home assistant UI. There is no need to add any other option in Tasmota

austwhite, 2020-05-02:
I realise what you are saying.  Not being rude, but if I have 6 devices setup and they are just names Tasmota1 through to Tasmota6, it is quite frustrating to have to turn each one on and off to find out what the device is.
That is why I asked if it is possible to consider adding it to Tasmota.

Jason2866, 2020-05-02:
Thx for your suggestion. Since the ESP82XX has very limited resources this will probably not be implemented. As you mentioned this is a one time unconvinient thing and there is a solution for

austwhite, 2020-05-02:
@Jason2866 Thank you for the reply.  Not a solution, but a workaround would be the better way to describe it.  
If it is not possible then I guess this can be closed.

closed 2020-05-03 by austwhite