arendst/Tasmota #8323: SSID is not encoded inside of html attribute.

nopjmp, 2020-05-03:
### PROBLEM DESCRIPTION
The SSID is not encoded correctly if a single quote (or potentially a double quote if that is used eventually) inside of the value.

```
<input id='s1' placeholder='' value='Example's Wifi'>
```

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Feit BR30/BPA300 Bulbs
- [x] Tasmota binary firmware version number used: 8.2.0
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: tuya-convert
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
04:45:44 MQT: stat/hallway2/RESULT = {"NAME":"Feit BR30/RGBW","GPIO":[0,0,0,0,141,140,0,0,38,142,37,0,0],"FLAG":0,"BASE":18}
04:45:45 MQT: stat/hallway2/RESULT = {"Module":{"0":"Feit BR30/RGBW"}}
04:45:45 MQT: stat/hallway2/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"141":"SM16716 DAT"},"GPIO5":{"140":"SM16716 CLK"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"38":"PWM2"},"GPIO13":{"142":"SM16716 PWR"},"GPIO14":{"37":"PWM1"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
04:46:13 MQT: stat/hallway2/STATUS = {"Status":{"Module":0,"FriendlyName":["Hallway Light 2"],"Topic":"hallway2","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
Use any SSID that includes a single quote or another potential HTML attribute violating symbol.

### EXPECTED BEHAVIOUR
All SSIDs regardless of characters should show up.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2020-05-13:
According to docs, your example is not valid. The SSID<x> command has this sentence:
**Do not use special characters or white spaces in the SSID**

nopjmp, 2020-05-14:
SSIDs can have special characters in them. This SSID already existed and still works on my devices, it's only a display bug.

sfromis, 2020-05-14:
The fact that your SSID may work in some situations, and is not validated by Tasmota, does not change the fact that it is not valid according to current docs. In case of the rules changing to allow special characters as valid, both the code and the docs would have to be changed.

majherek, 2020-05-15:
For me, a similar problem concerns the password for the WIFI network.

I use firmware compiled by me with hardcoded password and SSID in all my devices (due to using TLS). 

But from time to time for testing purposes I uploaded official firmware and the password entered through the SERIAL connection, or WIFI Manager worked. I have the impression that since the transition to Arduino Core 2.7.1 it stopped working (at the time when Jason2866 on Discord pointed out to me that I'm using the wrong Arduino Core for development version 8.2.0.6).

Such a life ... generally I do not bother, its. OK. For testing purposes I will connect devices to another WIFI network.

arendst, 2020-05-23:
The issue is caused by a HTML input value. It cannot be escaped so a workaround was made where the quote or apostrophe (``'``) is now supported but a quotation mark (``"``) is not.

So I'll close this as designed.

closed 2020-05-23 by arendst