arendst/Tasmota #8325: Shelly 2.5 and Shutter

Sawmoke, 2020-05-03:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Shutter calibration is not working on shelly 2.5 on extreme values (fully open or fully closed) and using sliding bar.

### REQUESTED INFORMATION
Is missing something in the procedure of calibration? Also should motor increase absorbed current at the extremes?

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Shelly 2.5
- [x] Tasmota binary firmware version number used: 8.2
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: 
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Shutter open overestimate time, which is correctly set. Also, moving sliding bar screws up calibration.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Shutter open close the relay at the right position.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
Hi, I got some troubles figuring out how to correctly calibrate the shutter. I followed the related tutorial (Shelly 2.5/ and Blinds and Shutters. I already tried to get support on chat but without success.

I performed setOption 80 1 and set the ShutterMode 0 as described in the doc.

It works fine as far as just pressing up or down in the web GUI. Then I performed calibration, and it is fine giving commands via console to open at certain percentage (not extremes), it's very accurate.
However, if I use the sliding bar this screws up the calibration. The same holds true if I perform a full closed or full open (even though correctly set during calibration).

Is it the whole system only based on time? Also, I don't know how it is supposed to be, but the motor of my rollershutter does not increase that much in absorbed current when reached the full open, but just stops itself regardless the relay state.
Is there anything I am missing?

**(Please, remember to close the issue when the problem has been addressed)**

kugelkopf123, 2020-05-03:
As far as i know its timebased. 
Can't reproduce that behavior. It works fine in my installation with the Shelly2.5. Perhaps it could be that there is a Motordelay? See [Wiki](https://tasmota.github.io/docs/Blinds-and-Shutters/#motor-movement-delays).
Your Motor has a fixed Start and End that is setuped at the motor itselfs. So thats the reason that the Motor Stops at the very top and the very bottom. This Rule in the Shelly2.5 Wiki is just for, if this points dont match. Some kind of a kind of last resort.

Sawmoke, 2020-05-03:
Thanks for your point. So if the motor is ok with its own values for stop on top and bottom I should not assign any rule, right? Does that mean that when fully open I can someway put a block on the relay? Because what is _dephasing_ the whole thing is that when fully open the relay remain ON for a while. 
About MotorDelay I obtained 100% accuracy on those short mid-positions. Instead, querying fully open or fully closed it goes wrong for some seconds.

kugelkopf123, 2020-05-03:
Then that is exactly where the problem lies! Your calibration for shutteropenduration and shuttercloseduration must be correct. Raise the shutter completely and switch setoption80 Off. 
Check by reopening the shutter again (switching on the corresponding relay) if the shutter is completely raised. Switch the relay off again afterwards. Now stop the time until your roller shutter is completely closed. You don't have to wait until the motor turns off when closing the shutter. You can stop the time and IMPORTANT the motor already when you reach the point where you consider the shutter to be completely closed. It is important that the engine does not continue driving after the time has stopped. Otherwise the next time will not fit.
So this is now your time for closing. Regardless of the end stop set on the motor.
Now you stop the time until the highest point, which is probably also the point where the motor switches off. So you now have the two times you can use for shutteropenduration and shuttercloseduration as soon as you have setoption80 enabled again. Write down the values somewhere.

Now you should check if the times fit. So just enter shutterclose or shutteropen in the console. 
If the times are correct, the relay should switch off at the same point where you have set the time before. When you open the shutter again, you should hear the relay click at the same time when the top point you selected is reached. If this is correct, you can continue with the rest of the calibration. 
If the values did not fit, you can still fine tune the values manually. The times can be entered as decimal numbers. Use a point as comma.
You must be aware that if you are at the top point and you then extend the shutter closed duration, you must also extend the shutter openduration a bit. Because if the shutter continues to close... it also takes more time to get to the top. So these two values are interdependent. If you try something, you will surely get a feeling for it.

Good luck!

closed 2020-05-06 by ascillato2