arendst/Tasmota #8328: Tasmota OTA update v8.2.0.6 = problems on Wemos D1 Mini & ESP-12F modules (???)

adharyanto, 2020-05-03:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

Updating (Over The Air) from Tasmota v8.2.0.4 to v8.2.0.6 worked fine on ESP-01 module and all Sonoff Modules (Sonoff Basic, Dual, POW R2), but failed on ESP-12F module and Wemos D1 mini module.
OTA updating procedure = (minimal update first, then full update)...
Updating links:
http://thehackbox.org/tasmota/tasmota-minimal.bin
http://thehackbox.org/tasmota/tasmota.bin

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

I there anybody experiences the same?

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2020-05-03:
Seems that the previous memory map you had on those devices was not the standard 1MB of Tasmota. 

There is nothing different in those hardwares, so there is nothing to do from the software side. Seems that your solution for those failed updated devices is to erase all flash and then flash the precompiled bins. 

If you need further assistant, please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

pweniger89, 2020-05-03:
I have a similar problem.
Updating to 8.2.0.6 totally broke my Shelly 2.5 (Unfortunately installed behind switch). 
I had to reflash them all. No wifi, no reaction to button presses.

arendst, 2020-05-04:
There was an issue with large OTA files being corrupted after the upload took place. It was fixed yesterday.

As a workaround execute command ``SetOption65 1`` before the OTA upload to disable Quick Power Cycle detection and the corruption of the uploaded file.

Sorry.

Jedithomas, 2020-05-04:
I have lost 4x Gosund Sp111 and 1x Obi Socket...

adharyanto, 2020-05-04:
I have tried four ways/sequences:

- MINIMAL update - FULL update = does not work, (WiFi lost connection), (this one without setoption65 1)
- setoption65 1 - MINIMAL update - FULL update = does not work, (WiFi lost connection)
- setoption65 1 - MINIMAL update - setoption65 1 - FULL update = does not work, (WiFi lost connection)
- MINIMAL update - setoption65 1 - FULL update = does not work, (WiFi lost connection)

NONE of the above sequences work on my WeMos D1 Mini...
Updating from v8.2.0.4 to v8.2.0.6 only works upto MINIMAL update, when tried to do FULL update then it does not work = (WiFi lost connection)

arendst, 2020-05-04:
What version of minimal do you use? It will need to be a version after
12:00 from yesterday

adharyanto, 2020-05-04:
Just recently......less than one hour ago....
These are the links I used:
http://thehackbox.org/tasmota/tasmota-minimal.bin
http://thehackbox.org/tasmota/tasmota.bin

I tried on my WeMos D1 Mini, always flashback to v8.2.0.4 first (the one that is working), and then tried to update to v8.2.0.6....

arendst, 2020-05-04:
Just tried on a wemos:

```
13:03:40 CMD: otaurl http://thehackbox.org/tasmota/tasmota.bin
13:03:40 SRC: Serial
13:03:40 CMD: Group 0, Index 1, Command "OTAURL", Data "http://thehackbox.org/tasmota/tasmota.bin"
13:03:40 CFG: CR 323/699
13:03:40 MQT: stat/wemos4/RESULT = {"OtaUrl":"http://thehackbox.org/tasmota/tasmota.bin"}
13:03:41 CFG: Saved to flash at F9, Count 17, Bytes 4096
13:03:45 CMD: status 0
13:03:45 SRC: Serial
13:03:45 CMD: Group 0, Index 1, Command "STATUS", Data "0"
13:03:45 MQT: stat/wemos4/STATUS = {"Status":{"Module":18,"FriendlyName":["Wemos4","Tasmota2","Tasmota3","Tasmota4",""],"Topic":"wemos4","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
13:03:45 MQT: stat/wemos4/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:58","StartupUTC":"2020-05-04T11:02:47","Sleep":50,"CfgHolder":4617,"BootCount":5,"BCResetTime":"2020-04-20T17:11:44","SaveCount":17,"SaveAddress":"F9000"}}
13:03:45 MQT: stat/wemos4/STATUS2 = {"StatusFWR":{"Version":"8.2.0.4(1af24dd-tasmota)","BuildDateTime":"2020-04-22T17:00:13","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"323/699"}}
13:03:45 MQT: stat/wemos4/STATUS3 = {"StatusLOG":{"SerialLog":3,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"domus1","LogPort":514,"SSId":["indebuurt","indebuurt_IoT"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","008000C0","00000000"]}}
13:03:45 MQT: stat/wemos4/STATUS4 = {"StatusMEM":{"ProgramSize":584,"Free":416,"Heap":25,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640E0","FlashMode":3,"Features":["00000809","8FDAE797","043683A0","000000CD","010013C0","C000F981","00000024"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
13:03:45 MQT: stat/wemos4/STATUS5 = {"StatusNET":{"Hostname":"wemos4","IPAddress":"192.168.2.223","Gateway":"192.168.2.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.27","Mac":"5C:CF:7F:83:BB:10","Webserver":2,"WifiConfig":5,"WifiPower":17.0}}
13:03:45 MQT: stat/wemos4/STATUS6 = {"StatusMQT":{"MqttHost":"domus1","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_83BB10","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
13:03:45 MQT: stat/wemos4/STATUS7 = {"StatusTIM":{"UTC":"2020-05-04T11:03:45","Local":"2020-05-04T13:03:45","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":99,"Sunrise":"05:50","Sunset":"21:06"}}
13:03:45 MQT: stat/wemos4/STATUS10 = {"StatusSNS":{"Time":"2020-05-04T13:03:45"}}
13:03:45 MQT: stat/wemos4/STATUS11 = {"StatusSTS":{"Time":"2020-05-04T13:03:45","Uptime":"0T00:00:58","UptimeSec":58,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":22,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","POWER5":"OFF","Wifi":{"AP":2,"SSId":"indebuurt_IoT","BSSId":"18:E8:29:CA:17:C1","Channel":11,"RSSI":100,"Signal":-38,"LinkCount":1,"Downtime":"0T00:00:07"}}}
13:03:54 CMD: upload 1
13:03:54 SRC: Serial
13:03:54 CMD: Group 0, Index 1, Command "UPLOAD", Data "1"
13:03:54 MQT: stat/wemos4/RESULT = {"Upload":"Version 8.2.0.4 from http://thehackbox.org/tasmota/tasmota.bin"}
13:03:55 CFG: Saved to flash at FB, Count 18, Bytes 4096
13:03:57 HTP: Web server stopped
13:03:57 UPL: http://thehackbox.org/tasmota/tasmota.bin
13:03:57 HTP: Web server active on wemos4 with IP address 192.168.2.223
13:03:59 HTP: Web server stopped
13:03:59 UPL: http://thehackbox.org/tasmota/tasmota-minimal.bin
13:04:10 HTP: Web server active on wemos4 with IP address 192.168.2.223
13:04:10 MQT: Attempting connection...
13:04:10 MQT: Connected
13:04:10 MQT: tele/wemos4/LWT = Online (retained)
13:04:10 MQT: cmnd/wemos4/POWER = 
13:04:10 MQT: Subscribe to cmnd/wemos4/#
13:04:10 MQT: Subscribe to cmnd/tasmotas/#
13:04:10 MQT: Subscribe to cmnd/DVES_83BB10_fb/#
13:04:12 OTA: Version 0x08020006, Compatible 0x07010006
13:04:12 MQT: stat/wemos4/UPGRADE = {"Upgrade":"Successful. Restarting"}
13:04:13 APP: Restarting

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3456, room 16 
tail 0
chksum 0x84
csum 0x84
v372a3ec2
@cp:0
ld
00:00:00 CFG: Loaded from flash at FB, Count 18
00:00:00 SRC: Restart
00:00:00 Project tasmota Wemos4 Version 8.2.0.6(a53c935-minimal)-2_7_0
00:00:00 WARNING This version does not support persistent settings
00:00:00 WIF: Attempting connection...
00:00:01 WIF: Network (re)scan started...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Attempting connection...
00:00:03 WIF: Attempting connection...
00:00:04 WIF: Network 0, AP2, SSId indebuurt_IoT, Channel 11, BSSId 18:E8:29:CA:17:C1, RSSI -32, Encryption 1
00:00:04 WIF: Connecting to AP2 indebuurt_IoT Channel 11 BSSId 18:E8:29:CA:17:C1 in mode 11N as wemos4...
00:00:04 WIF: Attempting connection...
00:00:05 WIF: Connected
00:00:05 HTP: Web server active on wemos4 with IP address 192.168.2.223
13:04:29 NTP: UTC 2020-05-04T11:04:28, DST 2020-03-29T02:00:00, STD 2020-10-25T03:00:00
13:04:30 MQT: Attempting connection...
13:04:30 MQT: Connected
[... 73 lines cut ...]
13:05:14 MQT: stat/wemos4/POWER3 = OFF
13:05:14 MQT: stat/wemos4/RESULT = {"POWER4":"OFF"}
[... cut ...]
13:05:22 MQT: stat/wemos4/STATUS11 = {"StatusSTS":{"Time":"2020-05-04T13:05:22","Uptime":"0T00:00:18","UptimeSec":18,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","POWER5":"OFF","Wifi":{"AP":2,"SSId":"indebuurt_IoT","BSSId":"18:E8:29:CA:17:C1","Channel":11,"RSSI":100,"Signal":-38,"LinkCount":1,"Downtime":"0T00:00:08"}}}
```
Also from 8.2.0.4 to latest 8.2.0.6 without issues.

Pls verify ``status 4`` for correct ``ProgramFlashSize`` being 1024

adharyanto, 2020-05-04:
18:17:54 CMD: status 4
18:17:54 MQT: StepperM1/stat/STATUS4 = {"StatusMEM":{"ProgramSize":378,"Free":624,"Heap":32,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164020","FlashMode":0,"Features":["00000809","80082581","040003A2","00000002","010000C0","00000000","00000020"],"Drivers":"1,2","Sensors":""}}

arendst, 2020-05-04:
Your flashmode is wrong! 

It appears to be QIO (0) mode while Tasmota only supports DOUT (3) mode.

You'll have to reflash your device using serial connection and make sure to use DOUT.

adharyanto, 2020-05-04:
I tried to re-flash again from v8.2.0.4, using SPI Mode = DOUT, Flash Sizze = 8Mbit (1MB):
Then update OTA to v8.2.0.6 Minimal = it works, (same as before)
Apply command setoption65 1
Then update OTA to v8.2.0.6 Full = it still does not work, WiFi lost (same as before).

20:27:28 CMD: status 4
20:27:28 MQT: StepperM1/stat/STATUS4 = {"StatusMEM":{"ProgramSize":582,"Free":420,"Heap":27,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164020","FlashMode":3,"Features":["00000809","8FDAE797","043683A0","000000CD","010013C0","C000F981","00000024"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}

00:00:15 CMD: setoption65
00:00:15 MQT: StepperM1/stat/RESULT = {"SetOption65":"ON"}

00:00:30 CMD: status 2
00:00:30 MQT: StepperM1/stat/STATUS2 = {"StatusFWR":{"Version":"8.2.0.6(c66f14b-minimal)","BuildDateTime":"2020-05-04T15:00:35","Boot":5,"Core":"2_7_0","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"345/699"}}

Btw, I also tried to compile using GitPod, v8.2.0.6, flash firmware.bin directly via usb, it works fine...
But this does not solve my original issue of updating OTA from v8.2.0.4 to v8.2.0.6

arendst, 2020-05-04:
Why do you load -minimal first and then the final?

Why don't you open the console and just type ``upload 1`` and wait while looking at the console what happens. It will load minimal first and then the final version. No need to interrupt.

Try this and let me know if that worked.

arendst, 2020-05-04:
Just did an upload using the GUI so minimal first, then after the gui returns the final version. Does work as expected.

```
15:46:21 HTP: Firmware Upgrade
15:46:34 CFG: Saved to flash at FB, Count 25, Bytes 4096
15:46:34 UPL: File tasmota-minimal (2).bin ...
................................................................................
................................................................................
..............................
15:46:45 OTA: Version 0x08020006, Compatible 0x07010006
15:46:45 UPL: Successful 387136 bytes. Restarting
15:46:45 HTP: Upload done
15:46:45 SRC: WebGui from 192.168.2.1
15:46:45 MQT: Attempting connection...
15:46:45 MQT: Connected
15:46:45 MQT: tele/wemos4/LWT = Online (retained)
15:46:45 MQT: cmnd/wemos4/POWER = 
15:46:45 MQT: Subscribe to cmnd/wemos4/#
15:46:45 MQT: Subscribe to cmnd/tasmotas/#
15:46:45 MQT: Subscribe to cmnd/DVES_83BB10_fb/#
15:46:46 APP: Restarting

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3656, room 16 
tail 8
chksum 0x0c
csum 0x0c
v9c56ed1f
@cp:0
ld
00:00:00 CFG: Loaded from flash at FB, Count 25
00:00:00 SRC: Restart
00:00:00 Project tasmota Wemos4 Version 8.2.0.6(c66f14b-minimal)-2_7_0
00:00:00 WARNING This version does not support persistent settings
00:00:00 WIF: Attempting connection...
00:00:01 WIF: Network (re)scan started...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Attempting connection...
00:00:03 WIF: Attempting connection...
00:00:04 WIF: Network 0, AP2, SSId indebuurt_IoT, Channel 11, BSSId 18:E8:29:CA:17:C1, RSSI -34, Encryption 1
00:00:04 WIF: Connecting to AP2 indebuurt_IoT Channel 11 BSSId 18:E8:29:CA:17:C1 in mode 11N as wemos4...
00:00:04 WIF: Attempting connection...
00:00:05 WIF: Connected
00:00:05 HTP: Web server active on wemos4 with IP address 192.168.2.223
15:47:01 NTP: UTC 2020-05-04T13:47:00, DST 2020-03-29T02:00:00, STD 2020-10-25T03:00:00
15:47:02 MQT: Attempting connection...
15:47:02 MQT: Connected
15:47:02 MQT: tele/wemos4/LWT = Online (retained)
15:47:02 MQT: cmnd/wemos4/POWER = 
15:47:02 MQT: Subscribe to cmnd/wemos4/#
15:47:02 MQT: Subscribe to cmnd/tasmotas/#
15:47:02 MQT: Subscribe to cmnd/DVES_83BB10_fb/#
15:47:02 MQT: tele/wemos4/INFO1 = {"Module":"Generic","Version":"8.2.0.6(c66f14b-minimal)","FallbackTopic":"cmnd/DVES_83BB10_fb/","GroupTopic":"cmnd/tasmotas/"}
15:47:02 MQT: tele/wemos4/INFO2 = {"WebServerMode":"Admin","Hostname":"wemos4","IPAddress":"192.168.2.223"}
15:47:02 MQT: tele/wemos4/INFO3 = {"RestartReason":"Software/System restart"}
15:47:02 MQT: stat/wemos4/RESULT = {"POWER1":"OFF"}
15:47:02 MQT: stat/wemos4/POWER1 = OFF
15:47:02 MQT: stat/wemos4/RESULT = {"POWER2":"OFF"}
15:47:02 MQT: stat/wemos4/POWER2 = OFF
15:47:02 MQT: stat/wemos4/RESULT = {"POWER3":"OFF"}
15:47:02 MQT: stat/wemos4/POWER3 = OFF
15:47:02 MQT: stat/wemos4/RESULT = {"POWER4":"OFF"}
15:47:02 MQT: stat/wemos4/POWER4 = OFF
15:47:02 MQT: stat/wemos4/RESULT = {"POWER5":"OFF"}
15:47:02 MQT: stat/wemos4/POWER5 = OFF
15:47:03 APP: Boot Count 8
15:47:06 MQT: tele/wemos4/STATE = {"Time":"2020-05-04T15:47:06","Uptime":"0T00:00:13","UptimeSec":13,"Vcc":2.973,"Heap":33,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","POWER5":"OFF","Wifi":{"AP":2,"SSId":"indebuurt_IoT","BSSId":"18:E8:29:CA:17:C1","Channel":11,"RSSI":100,"Signal":-38,"LinkCount":1,"Downtime":"0T00:00:07"}}
15:47:09 HTP: Main Menu
15:47:14 HTP: Firmware Upgrade
[... 43 lines cut ...]
15:48:01 QPC: Reset
15:48:02 MQT: Attempting connection...
15:48:02 MQT: Connected
15:48:02 MQT: tele/wemos4/LWT = Online (retained)
15:48:02 MQT: cmnd/wemos4/POWER = 
15:48:02 MQT: Subscribe to cmnd/wemos4/#
15:48:02 MQT: Subscribe to cmnd/tasmotas/#
15:48:02 MQT: Subscribe to cmnd/DVES_83BB10_fb/#
15:48:02 MQT: tele/wemos4/INFO1 = {"Module":"Generic","Version":"8.2.0.6(c66f14b-tasmota)","FallbackTopic":"cmnd/DVES_83BB10_fb/","GroupTopic":"cmnd/tasmotas/"}
15:48:02 MQT: tele/wemos4/INFO2 = {"WebServerMode":"Admin","Hostname":"wemos4","IPAddress":"192.168.2.223"}
15:48:02 MQT: tele/wemos4/INFO3 = {"RestartReason":"Software/System restart"}
15:48:02 MQT: stat/wemos4/RESULT = {"POWER1":"OFF"}
15:48:02 MQT: stat/wemos4/POWER1 = OFF
15:48:02 MQT: stat/wemos4/RESULT = {"POWER2":"OFF"}
15:48:02 MQT: stat/wemos4/POWER2 = OFF
15:48:02 MQT: stat/wemos4/RESULT = {"POWER3":"OFF"}
15:48:02 MQT: stat/wemos4/POWER3 = OFF
15:48:02 MQT: stat/wemos4/RESULT = {"POWER4":"OFF"}
15:48:02 MQT: stat/wemos4/POWER4 = OFF
15:48:02 MQT: stat/wemos4/RESULT = {"POWER5":"OFF"}
15:48:02 MQT: stat/wemos4/POWER5 = OFF
15:48:03 APP: Boot Count 8
15:48:03 CFG: Saved to flash at FA, Count 26, Bytes 4096
15:48:06 MQT: tele/wemos4/STATE = {"Time":"2020-05-04T15:48:06","Uptime":"0T00:00:13","UptimeSec":13,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","POWER5":"OFF","Wifi":{"AP":2,"SSId":"indebuurt_IoT","BSSId":"18:E8:29:CA:17:C1","Channel":11,"RSSI":100,"Signal":-38,"LinkCount":1,"Downtime":"0T00:00:07"}}
```

adharyanto, 2020-05-04:
Tried to give command "upload 1", then it is updated to v8.2.0.6 Minimal, then after more than 5 minutes, it keeps on Minimal firmware, (never continue to Full update)....
I do not see the APP: Boot Count 8 (as yours)...
Because it keeps in Minimal, I tried to update to FULL via the GUI, then the WiFi lost connection.....

00:00:00 CFG: Loaded from flash at FB, Count 2616
00:00:00 Project tasmota WeMos D1 Mini Version 8.2.0.6(c66f14b-minimal)-2_7_0
00:00:00 WARNING This version does not support persistent settings
00:00:00 WIF: Connecting to AP1 omahklodran Channel 11 BSSId 0C:37:47:93:63:DF in mode 11N as StepperM1-7217...
00:00:03 WIF: Connected
00:00:03 HTP: Web server active on StepperM1-7217 with IP address 192.168.1.40
20:53:51 MQT: Attempting connection...
20:53:51 MQT: Connected
20:53:51 MQT: StepperM1/tele/LWT = Online (retained)
20:53:51 MQT: StepperM1/cmnd/POWER = 
20:53:51 MQT: StepperM1/tele/INFO1 = {"Module":"Generic","Version":"8.2.0.6(c66f14b-minimal)","FallbackTopic":"cmnd/DVES_BAFC31_fb/","GroupTopic":"sonoffs/cmnd/"}
20:53:51 MQT: StepperM1/tele/INFO2 = {"WebServerMode":"Admin","Hostname":"StepperM1-7217","IPAddress":"192.168.1.40"}
20:53:51 MQT: StepperM1/tele/INFO3 = {"RestartReason":"Software/System restart"}
20:53:51 MQT: StepperM1/stat/RESULT = {"POWER1":"OFF"}
20:53:51 MQT: StepperM1/stat/POWER1 = OFF
20:53:51 MQT: StepperM1/stat/RESULT = {"POWER2":"OFF"}
20:53:51 MQT: StepperM1/stat/POWER2 = OFF
20:53:51 MQT: StepperM1/stat/RESULT = {"POWER3":"OFF"}
20:53:51 MQT: StepperM1/stat/POWER3 = OFF
20:53:51 MQT: StepperM1/stat/RESULT = {"POWER4":"OFF"}
20:53:51 MQT: StepperM1/stat/POWER4 = OFF
20:53:55 MQT: StepperM1/tele/STATE = {"Time":"2020-05-04T20:53:55","Uptime":"0T00:00:11","UptimeSec":11,"Vcc":3.021,"Heap":31,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":33,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"omahklodran","BSSId":"0C:37:47:93:63:DF","Channel":11,"RSSI":46,"Signal":-77,"LinkCount":1,"Downtime":"0T00:00:05"}}
20:54:05 MQT: StepperM1/tele/STATE = {"Time":"2020-05-04T20:54:05","Uptime":"0T00:00:21","UptimeSec":21,"Vcc":3.021,"Heap":33,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"omahklodran","BSSId":"0C:37:47:93:63:DF","Channel":11,"RSSI":48,"Signal":-76,"LinkCount":1,"Downtime":"0T00:00:05"}}
20:54:15 MQT: StepperM1/tele/STATE = {"Time":"2020-05-04T20:54:15","Uptime":"0T00:00:31","UptimeSec":31,"Vcc":3.021,"Heap":33,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"omahklodran","BSSId":"0C:37:47:93:63:DF","Channel":11,"RSSI":46,"Signal":-77,"LinkCount":1,"Downtime":"0T00:00:05"}}
20:54:25 MQT: StepperM1/tele/STATE = {"Time":"2020-05-04T20:54:25","Uptime":"0T00:00:41","UptimeSec":41,"Vcc":3.020,"Heap":33,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"omahklodran","BSSId":"0C:37:47:93:63:DF","Channel":11,"RSSI":46,"Signal":-77,"LinkCount":1,"Downtime":"0T00:00:05"}}
20:54:35 MQT: StepperM1/tele/STATE = {"Time":"2020-05-04T20:54:35","Uptime":"0T00:00:51","UptimeSec":51,"Vcc":3.021,"Heap":33,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"omahklodran","BSSId":"0C:37:47:93:63:DF","Channel":11,"RSSI":44,"Signal":-78,"LinkCount":1,"Downtime":"0T00:00:05"}}

arendst, 2020-05-04:
Do the same again but first enable more logging with command ``Seriallog 3``. Provide the complete log.

adharyanto, 2020-05-04:
I gave commands:
- seriallog 3
- setoption65 1
- upload 1
(Still keeps on the Minimal firmware, never continue to Full firmware...)

00:00:00 CFG: Loaded from flash at F9, Count 2613
00:00:00 Project tasmota WeMos D1 Mini Version 8.2.0.4(abdd398-tasmota)-STAGE
00:00:00 WIF: Connecting to AP1 omahklodran Channel 7 BSSId 0C:37:47:93:63:DF in mode 11N as StepperM1-7217...
00:00:08 WIF: Connected
00:00:08 HTP: Web server active on StepperM1-7217 with IP address 192.168.1.40
00:00:13 MQT: Attempting connection...
00:00:13 MQT: Connected
00:00:13 MQT: StepperM1/tele/LWT = Online (retained)
00:00:13 MQT: StepperM1/cmnd/POWER = 
00:00:13 MQT: StepperM1/tele/INFO1 = {"Module":"Generic","Version":"8.2.0.4(abdd398-tasmota)","FallbackTopic":"cmnd/DVES_BAFC31_fb/","GroupTopic":"sonoffs/cmnd/"}
00:00:13 MQT: StepperM1/tele/INFO2 = {"WebServerMode":"Admin","Hostname":"StepperM1-7217","IPAddress":"192.168.1.40"}
00:00:13 MQT: StepperM1/tele/INFO3 = {"RestartReason":"External System"}
00:00:13 MQT: StepperM1/stat/RESULT = {"POWER1":"OFF"}
00:00:13 MQT: StepperM1/stat/POWER1 = OFF
00:00:13 MQT: StepperM1/stat/RESULT = {"POWER2":"OFF"}
00:00:13 MQT: StepperM1/stat/POWER2 = OFF
00:00:13 MQT: StepperM1/stat/RESULT = {"POWER3":"OFF"}
00:00:13 MQT: StepperM1/stat/POWER3 = OFF
00:00:13 MQT: StepperM1/stat/RESULT = {"POWER4":"OFF"}
00:00:13 MQT: StepperM1/stat/POWER4 = OFF
00:00:17 MQT: StepperM1/tele/STATE = {"Time":"1970-01-01T00:00:17","Uptime":"0T00:00:15","UptimeSec":15,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":51,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"omahklodran","BSSId":"0C:37:47:93:63:DF","Channel":11,"RSSI":46,"Signal":-77,"LinkCount":1,"Downtime":"0T00:00:09"}}
00:00:17 MQT: StepperM1/tele/SENSOR = {"Time":"1970-01-01T00:00:17","ANALOG":{"A0":2}}
00:00:20 CMD: seriallog 3
00:00:20 MQT: StepperM1/stat/RESULT = {"SerialLog":{"3":{"Active":"3"}}}
00:00:26 CMD: setoption65 1
00:00:26 MQT: StepperM1/stat/RESULT = {"SetOption65":"ON"}
00:00:27 MQT: StepperM1/tele/STATE = {"Time":"1970-01-01T00:00:27","Uptime":"0T00:00:25","UptimeSec":25,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":31,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"omahklodran","BSSId":"0C:37:47:93:63:DF","Channel":11,"RSSI":42,"Signal":-79,"LinkCount":1,"Downtime":"0T00:00:09"}}
00:00:27 MQT: StepperM1/tele/SENSOR = {"Time":"1970-01-01T00:00:27","ANALOG":{"A0":2}}
00:00:37 MQT: StepperM1/tele/STATE = {"Time":"1970-01-01T00:00:37","Uptime":"0T00:00:35","UptimeSec":35,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":21,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"omahklodran","BSSId":"0C:37:47:93:63:DF","Channel":11,"RSSI":36,"Signal":-82,"LinkCount":1,"Downtime":"0T00:00:09"}}
00:00:37 MQT: StepperM1/tele/SENSOR = {"Time":"1970-01-01T00:00:37","ANALOG":{"A0":2}}
00:00:47 MQT: StepperM1/tele/STATE = {"Time":"1970-01-01T00:00:47","Uptime":"0T00:00:45","UptimeSec":45,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"omahklodran","BSSId":"0C:37:47:93:63:DF","Channel":11,"RSSI":42,"Signal":-79,"LinkCount":1,"Downtime":"0T00:00:09"}}
00:00:47 MQT: StepperM1/tele/SENSOR = {"Time":"1970-01-01T00:00:47","ANALOG":{"A0":2}}
00:00:57 MQT: StepperM1/tele/STATE = {"Time":"1970-01-01T00:00:57","Uptime":"0T00:00:55","UptimeSec":55,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":26,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"omahklodran","BSSId":"0C:37:47:93:63:DF","Channel":11,"RSSI":44,"Signal":-78,"LinkCount":1,"Downtime":"0T00:00:09"}}
00:00:57 MQT: StepperM1/tele/SENSOR = {"Time":"1970-01-01T00:00:57","ANALOG":{"A0":2}}
00:01:02 CMD: upload 1
00:01:02 MQT: StepperM1/stat/RESULT = {"Upload":"Version 8.2.0.4 from http://thehackbox.org/tasmota/tasmota-minimal.bin"}
00:01:05 HTP: Web server stopped
00:01:36 HTP: Web server active on StepperM1-7217 with IP address 192.168.1.40
00:01:37 MQT: Attempting connection...
00:01:37 MQT: Connected
00:01:37 MQT: StepperM1/tele/LWT = Online (retained)
00:01:37 MQT: StepperM1/cmnd/POWER = 
00:01:37 MQT: StepperM1/tele/STATE = {"Time":"1970-01-01T00:01:37","Uptime":"0T00:01:05","UptimeSec":65,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":1720,"MqttCount":2,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"omahklodran","BSSId":"0C:37:47:93:63:DF","Channel":11,"RSSI":48,"Signal":-76,"LinkCount":1,"Downtime":"0T00:00:09"}}
00:01:37 MQT: StepperM1/tele/SENSOR = {"Time":"1970-01-01T00:01:37","ANALOG":{"A0":2}}

===============

00:00:00 CFG: Loaded from flash at FB, Count 2617
00:00:00 Project tasmota WeMos D1 Mini Version 8.2.0.6(c66f14b-minimal)-2_7_0
00:00:00 WARNING This version does not support persistent settings
00:00:00 WIF: Connecting to AP1 omahklodran Channel 11 BSSId 0C:37:47:93:63:DF in mode 11N as StepperM1-7217...
00:00:03 WIF: Connected
00:00:03 HTP: Web server active on StepperM1-7217 with IP address 192.168.1.40
21:17:52 MQT: Attempting connection...
21:17:52 MQT: Connected
21:17:52 MQT: StepperM1/tele/LWT = Online (retained)
21:17:52 MQT: StepperM1/cmnd/POWER = 
21:17:52 MQT: StepperM1/tele/INFO1 = {"Module":"Generic","Version":"8.2.0.6(c66f14b-minimal)","FallbackTopic":"cmnd/DVES_BAFC31_fb/","GroupTopic":"sonoffs/cmnd/"}
[... cut ...]
21:17:52 MQT: StepperM1/stat/POWER3 = OFF
21:17:52 MQT: StepperM1/stat/RESULT = {"POWER4":"OFF"}
21:17:52 MQT: StepperM1/stat/POWER4 = OFF
21:17:56 MQT: StepperM1/tele/STATE = {"Time":"2020-05-04T21:17:56","Uptime":"0T00:00:11","UptimeSec":11,"Vcc":3.021,"Heap":33,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"omahklodran","BSSId":"0C:37:47:93:63:DF","Channel":11,"RSSI":48,"Signal":-76,"LinkCount":1,"Downtime":"0T00:00:05"}}

arendst, 2020-05-04:
You're not providing seriallog as the extra information is not in your logs. Pls connect a serial terminal and do the actions again.

adharyanto, 2020-05-04:
I do not see any Serial Terminal on Tasmota GUI software, where I can find it?

arendst, 2020-05-04:
https://tasmota.github.io/docs/Getting-Started/#using-serial-terminal

adharyanto, 2020-05-04:
This below is the Termite serial log report.
Giving the same commands; (seriallog 3, setoption65 1, upload 1).
Result is = only up to MINIMAL firmware, after that only keeps sending ".../tele/STATE = ...", and never continue to go to FULL firmware update.

=================

sl[00]$Üß|[00]Ä$à<[03][04][0C][0C]Ä[04]lä[04]c|‡ƒ[03]ì[1B]“;ÛcŒ[04]#Œû''Ÿ$'gœãì[0C]c[1C]pì‡dsl;dpóoà[10][03][0C][04]ƒ[0C]$[04]Œœ[04][04][0C]b[04]gã|[03]ì[0C]sœ[0C]c„óonï[00]lÄŽd`[03]˜[1B][13]o'[0C]l`[03][0E][03]n;ŽÛÛ'[0C][04][03][0C][0F]l[07];›“o[04][04][03][04]Ž[07]$œ{ä‡[1C][03]{lœpŒl`[03]üÃ'œ[03]00:00:00 CFG: Loaded from flash at F9, Count 2613
00:00:00 Project tasmota WeMos D1 Mini Version 8.2.0.4(abdd398-tasmota)-STAGE
00:00:00 WIF: Connecting to AP1 omahklodran Channel 7 BSSId 0C:37:47:93:63:DF in mode 11N as StepperM1-7217...
00:00:08 WIF: Connect failed with AP timeout
00:00:08 WIF: Connecting to AP1 omahklodran in mode 11N as StepperM1-7217...
00:00:15 WIF: Connected
00:00:15 HTP: Web server active on StepperM1-7217 with IP address 192.168.1.40
22:04:08 MQT: Attempting connection...
22:04:08 MQT: Connected
22:04:08 MQT: StepperM1/tele/LWT = Online (retained)
22:04:08 MQT: StepperM1/cmnd/POWER = 
22:04:08 MQT: StepperM1/tele/INFO1 = {"Module":"Generic","Version":"8.2.0.4(abdd398-tasmota)","FallbackTopic":"cmnd/DVES_BAFC31_fb/","GroupTopic":"sonoffs/cmnd/"}
22:04:08 MQT: StepperM1/tele/INFO2 = {"WebServerMode":"Admin","Hostname":"StepperM1-7217","IPAddress":"192.168.1.40"}
22:04:08 MQT: StepperM1/tele/INFO3 = {"RestartReason":"External System"}
22:04:08 MQT: StepperM1/stat/RESULT = {"POWER1":"OFF"}
22:04:08 MQT: StepperM1/stat/POWER1 = OFF
22:04:08 MQT: StepperM1/stat/RESULT = {"POWER2":"OFF"}
22:04:08 MQT: StepperM1/stat/POWER2 = OFF
22:04:08 MQT: StepperM1/stat/RESULT = {"POWER3":"OFF"}
22:04:08 MQT: StepperM1/stat/POWER3 = OFF
22:04:08 MQT: StepperM1/stat/RESULT = {"POWER4":"OFF"}
22:04:08 MQT: StepperM1/stat/POWER4 = OFF
22:04:12 MQT: StepperM1/tele/STATE = {"Time":"2020-05-04T22:04:12","Uptime":"0T00:00:22","UptimeSec":22,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":24,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"omahklodran","BSSId":"0C:37:47:93:63:DF","Channel":11,"RSSI":40,"Signal":-80,"LinkCount":1,"Downtime":"0T00:00:16"}}
22:04:12 MQT: StepperM1/tele/SENSOR = {"Time":"2020-05-04T22:04:12","ANALOG":{"A0":2}}
seriallog 3
22:04:13 CMD: seriallog 3
22:04:13 MQT: StepperM1/stat/RESULT = {"SerialLog":{"3":{"Active":"3"}}}
22:04:13 CFG: Saved to flash at F6, Count 2616, Bytes 4096
setoption65 1
22:04:19 CMD: setoption65 1
22:04:19 SRC: Serial
22:04:19 CMD: Group 0, Index 65, Command "SETOPTION", Data "1"
22:04:19 MQT: StepperM1/stat/RESULT = {"SetOption65":"ON"}
22:04:19 CFG: Saved to flash at F5, Count 2617, Bytes 4096
22:04:22 MQT: StepperM1/tele/STATE = {"Time":"2020-05-04T22:04:22","Uptime":"0T00:00:32","UptimeSec":32,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":34,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"omahklodran","BSSId":"0C:37:47:93:63:DF","Channel":11,"RSSI":30,"Signal":-85,"LinkCount":1,"Downtime":"0T00:00:16"}}
22:04:22 MQT: StepperM1/tele/SENSOR = {"Time":"2020-05-04T22:04:22","ANALOG":{"A0":2}}
upload 1
22:04:23 CMD: upload 1
22:04:23 SRC: Serial
22:04:23 CMD: Group 0, Index 1, Command "UPLOAD", Data "1"
22:04:23 MQT: StepperM1/stat/RESULT = {"Upload":"Version 8.2.0.4 from http://thehackbox.org/tasmota/tasmota-minimal.bin"}
22:04:24 CFG: Saved to flash at FB, Count 2618, Bytes 4096
22:04:26 HTP: Web server stopped
22:04:26 UPL: http://thehackbox.org/tasmota/tasmota-minimal.bin
22:04:54 HTP: Web server active on StepperM1-7217 with IP address 192.168.1.40
22:04:54 MQT: Attempting connection...
22:04:54 MQT: Connected
22:04:54 MQT: StepperM1/tele/LWT = Online (retained)
22:04:54 MQT: StepperM1/cmnd/POWER = 
22:04:54 MQT: Subscribe to StepperM1/cmnd/#
22:04:54 MQT: Subscribe to sonoffs/cmnd/#
22:04:54 MQT: Subscribe to cmnd/DVES_BAFC31_fb/#
22:04:56 OTA: Version 0x08020006, Compatible 0x07010006
22:04:56 MQT: StepperM1/stat/UPGRADE = {"Upgrade":"Successful. Restarting"}
22:04:57 APP: Restarting

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 3456, room 16 
tail 0
chksum 0x84
csum 0x84
v372a3ec2
@cp:0
[... 10 lines cut ...]
00:00:03 WIF: Connected
00:00:03 HTP: Web server active on StepperM1-7217 with IP address 192.168.1.40
22:05:08 NTP: UTC 2020-05-04T15:05:07, DST 2020-01-26T00:00:00, STD 2020-01-26T00:00:00
22:05:09 MQT: Attempting connection...
22:05:09 MQT: Connected
22:05:09 MQT: StepperM1/tele/LWT = Online (retained)
22:05:09 MQT: StepperM1/cmnd/POWER = 
22:05:09 MQT: Subscribe to StepperM1/cmnd/#
22:05:09 MQT: Subscribe to sonoffs/cmnd/#
22:05:09 MQT: Subscribe to cmnd/DVES_BAFC31_fb/#
22:05:09 MQT: StepperM1/tele/INFO1 = {"Module":"Generic","Version":"8.2.0.6(c66f14b-minimal)","FallbackTopic":"cmnd/DVES_BAFC31_fb/","GroupTopic":"sonoffs/cmnd/"}
22:05:09 MQT: StepperM1/tele/INFO2 = {"WebServerMode":"Admin","Hostname":"StepperM1-7217","IPAddress":"192.168.1.40"}
22:05:09 MQT: StepperM1/tele/INFO3 = {"RestartReason":"Software/System restart"}
22:05:09 MQT: StepperM1/stat/RESULT = {"POWER1":"OFF"}
22:05:09 MQT: StepperM1/stat/POWER1 = OFF
22:05:09 MQT: StepperM1/stat/RESULT = {"POWER2":"OFF"}
22:05:09 MQT: StepperM1/stat/POWER2 = OFF
22:05:09 MQT: StepperM1/stat/RESULT = {"POWER3":"OFF"}
22:05:09 MQT: StepperM1/stat/POWER3 = OFF
[... cut ...] ahklodran","BSSId":"0C:37:47:93:63:DF","Channel":11,"RSSI":48,"Signal":-76,"LinkCount":1,"Downtime":"0T00:00:05"}}
22:05:33 MQT: StepperM1/tele/STATE = {"Time":"2020-05-04T22:05:33","Uptime":"0T00:00:31","UptimeSec":31,"Vcc":3.021,"Heap":33,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"omahklodran","BSSId":"0C:37:47:93:63:DF","Channel":11,"RSSI":34,"Signal":-83,"LinkCount":1,"Downtime":"0T00:00:05"}}

arendst, 2020-05-04:
That's expected when you set the OTAUrl to ``http://thehackbox.org/tasmota/tasmota-minimal.bin``

Now the final step.

Do it all again but FIRST set OtaUrl to ``http://thehackbox.org/tasmota/tasmota.bin``

adharyanto, 2020-05-04:
Great.... now it works.......!!!! Thank you....
So, do we have to update via Serial Terminal, because if we click "Start Upgrade" on the Tasmota GUI it will stop the WiFi, and upgrade to Full firmware become unsuccessful....(???)
So, if we have to do it via Serial Terminal, then how we can update the other Devices that are not connected via serial port? The Devices are away/remote locations, must be (OTA). Will it work this way as well?

=============

seriallog 3
22:27:58 CMD: seriallog 3
22:27:58 MQT: StepperM1/stat/RESULT = {"SerialLog":{"3":{"Active":"3"}}}
22:27:59 CFG: Saved to flash at F6, Count 2616, Bytes 4096
22:27:59 MQT: StepperM1/tele/STATE = {"Time":"2020-05-04T22:27:59","Uptime":"0T00:00:55","UptimeSec":55,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":26,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"omahklodran","BSSId":"0C:37:47:93:63:DF","Channel":11,"RSSI":42,"Signal":-79,"LinkCount":1,"Downtime":"0T00:00:07"}}
22:27:59 MQT: StepperM1/tele/SENSOR = {"Time":"2020-05-04T22:27:59","ANALOG":{"A0":2}}
setoption65 1
22:28:06 CMD: setoption65 1
22:28:06 SRC: Serial
22:28:06 CMD: Group 0, Index 65, Command "SETOPTION", Data "1"
22:28:06 MQT: StepperM1/stat/RESULT = {"SetOption65":"ON"}
22:28:06 CFG: Saved to flash at F5, Count 2617, Bytes 4096
22:28:09 MQT: StepperM1/tele/STATE = {"Time":"2020-05-04T22:28:09","Uptime":"0T00:01:05","UptimeSec":65,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"omahklodran","BSSId":"0C:37:47:93:63:DF","Channel":11,"RSSI":42,"Signal":-79,"LinkCount":1,"Downtime":"0T00:00:07"}}
22:28:09 MQT: StepperM1/tele/SENSOR = {"Time":"2020-05-04T22:28:09","ANALOG":{"A0":2}}
upload 1
22:28:10 CMD: upload 1
22:28:10 SRC: Serial
22:28:10 CMD: Group 0, Index 1, Command "UPLOAD", Data "1"
22:28:10 MQT: StepperM1/stat/RESULT = {"Upload":"Version 8.2.0.4 from http://thehackbox.org/tasmota/tasmota-minimal.bin"}
22:28:11 CFG: Saved to flash at FB, Count 2618, Bytes 4096
22:28:13 HTP: Web server stopped
22:28:13 UPL: http://thehackbox.org/tasmota/tasmota-minimal.bin
22:28:31 HTP: Web server active on StepperM1-7217 with IP address 192.168.1.40
22:28:31 MQT: Attempting connection...
22:28:31 MQT: Connected
22:28:31 MQT: StepperM1/tele/LWT = Online (retained)
22:28:31 MQT: StepperM1/cmnd/POWER = 
22:28:31 MQT: Subscribe to StepperM1/cmnd/#
22:28:31 MQT: Subscribe to sonoffs/cmnd/#
22:28:31 MQT: Subscribe to cmnd/DVES_BAFC31_fb/#
22:28:33 OTA: Version 0x08020006, Compatible 0x07010006
22:28:33 MQT: StepperM1/stat/UPGRADE = {"Upgrade":"Successful. Restarting"}
22:28:34 APP: Restarting

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 3456, room 16 
tail 0
chksum 0x84
csum 0x84
v372a3ec2
@cp:0
ld
00:00:00 CFG: Loaded from flash at FB, Count 2618
00:00:00 CFG: CR 345/699
00:00:00 SRC: Restart
00:00:00 Project tasmota WeMos D1 Mini Version 8.2.0.6(a32682b-minimal)-2_7_0
00:00:00 WARNING This version does not support persistent settings
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Connecting to AP1 omahklodran Channel 11 BSSId 0C:37:47:93:63:DF in mode 11N as StepperM1-7217...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Attempting connection...
00:00:03 WIF: Connected
00:00:03 HTP: Web server active on StepperM1-7217 with IP address 192.168.1.40
22:28:45 NTP: UTC 2020-05-04T15:28:44, DST 2020-01-26T00:00:00, STD 2020-01-26T00:00:00
22:28:46 MQT: Attempting connection...
22:28:46 MQT: Connected
22:28:46 MQT: StepperM1/tele/LWT = Online (retained)
22:28:46 MQT: StepperM1/cmnd/POWER = 
22:28:46 MQT: Subscribe to StepperM1/cmnd/#
22:28:46 MQT: Subscribe to sonoffs/cmnd/#
22:28:46 MQT: Subscribe to cmnd/DVES_BAFC31_fb/#
22:28:46 MQT: StepperM1/tele/INFO1 = {"Module":"Generic","Version":"8.2.0.6(a32682b-minimal)","FallbackTopic":"cmnd/DVES_BAFC31_fb/","GroupTopic":"sonoffs/cmnd/"}
22:28:46 MQT: StepperM1/tele/INFO2 = {"WebServerMode":"Admin","Hostname":"StepperM1-7217","IPAddress":"192.168.1.40"}
[... 49 lines cut ...]
00:00:07 MQT: StepperM1/tele/LWT = Online (retained)
00:00:07 MQT: StepperM1/cmnd/POWER = 
00:00:07 MQT: Subscribe to StepperM1/cmnd/#
00:00:07 MQT: Subscribe to sonoffs/cmnd/#
00:00:07 MQT: Subscribe to cmnd/DVES_BAFC31_fb/#
00:00:07 MQT: StepperM1/tele/INFO1 = {"Module":"Generic","Version":"8.2.0.6(a32682b-tasmota)","FallbackTopic":"cmnd/DVES_BAFC31_fb/","GroupTopic":"sonoffs/cmnd/"}
00:00:07 MQT: StepperM1/tele/INFO2 = {"WebServerMode":"Admin","Hostname":"StepperM1-7217","IPAddress":"192.168.1.40"}
00:00:07 MQT: StepperM1/tele/INFO3 = {"RestartReason":"Software/System restart"}
00:00:07 MQT: StepperM1/stat/RESULT = {"POWER1":"OFF"}
00:00:07 MQT: StepperM1/stat/POWER1 = OFF
00:00:07 MQT: StepperM1/stat/RESULT = {"POWER2":"OFF"}
00:00:07 MQT: StepperM1/stat/POWER2 = OFF
00:00:07 MQT: StepperM1/stat/RESULT = {"POWER3":"OFF"}
00:00:07 MQT: StepperM1/stat/POWER3 = OFF
00:00:07 MQT: StepperM1/stat/RESULT = {"POWER4":"OFF"}
00:00:07 MQT: StepperM1/stat/POWER4 = OFF
00:00:08 APP: Boot Count 270
00:00:08 CFG: Saved to flash at FA, Count 2620, Bytes 4096
00:00:11 HTP: Main Menu
00:00:11 MQT: StepperM1/tele/STATE = {"Time":"1970-01-01T00:00:11","Uptime":"0T00:00:13","UptimeSec":13,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":41,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"omahklodran","BSSId":"0C:37:47:93:63:DF","Channel":11,"RSSI" [... cut ...]
00:00:21 MQT: StepperM1/tele/SENSOR = {"Time":"1970-01-01T00:00:21","ANALOG":{"A0":3}}
00:00:31 MQT: StepperM1/tele/STATE = {"Time":"1970-01-01T00:00:31","Uptime":"0T00:00:33","UptimeSec":33,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":36,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"omahklodran","BSSId":"0C:37:47:93:63:DF","Channel":11,"RSSI":46,"Signal":-77,"LinkCount":1,"Downtime":"0T00:00:07"}}
00:00:31 MQT: StepperM1/tele/SENSOR = {"Time":"1970-01-01T00:00:31","ANALOG":{"A0":2}}

arendst, 2020-05-04:
> So, do we have to update via Serial Terminal, because if we click "Start Upgrade" on the Tasmota GUI it will stop the WiFi, and upgrade to Full firmware become unsuccessful....(???)
So, if we have to do it via Serial Terminal, then how we can update the other Devices that are not connected via serial port? The Devices are away/remote locations, must be (OTA). Will it work this way as well?

No there is no reason for anyone to use serial for upgrade. YOU needed serial to find out what went wrong.

If you retrace what you did you'll notice that the main reason your upgrades failed is because your OtaUrl was set wrong. It should never be set to -minimal as that is just an intermediate step. It should always contain the final image you want to load.

I suggest you find answers to your questions by executing the same from the GUI. You'll see it all works as designed as long as OtaUrl is set right.

THERE ARE NO KNOWN ISSUES WITH OTA UPGRADES.

adharyanto, 2020-05-04:
Ok, I repeat again the steps from beginning:

1. Reflash my Wemos D1 Mini back to v8.2.0.4 (DOUT, 8Mbit/1MB) = reflash using PC via USB cable
2. Unplug the usb cable from pc, and connect the it to battery/power bank = to make sure Wemos D1 Mini is connected via WiFi
3. Execute console command = setoption65 1
4. Edit the link = http://thehackbox.org/tasmota/tasmota.bin
5. Click "Start upgrade" from Tasmota GUI = directly without minimal firmware
6. Wait until it reboots = and then started in v8.2.0.6
7. Now it works fine...... my problems are solved, thank you.

Just wondering why I did not have any issues with all my Sonoff modules, (even with Minimal first), problems only with my Wemos D1 Mini and my ESP-12F... (???)
Let's see how about the others, who have lost their GoSund SP111, Obi Socket, Shelly 2.5 while upgrading to v8.2.0.6 = need to know how they will resolve their issues.

adharyanto, 2020-05-04:
Fyi, additional information:
- tried to repeat again the above steps, now started by flashing v8.2.0.4 (QIO, 32Mbit/8MB), result is the same  = also working fine.

closed 2020-05-04 by arendst