arendst/Tasmota #8330: Add LG ThinQ HVAC support

aph3rson, 2020-05-04:
**Have you looked for this feature in other issues and in the docs?**  Yes

**Is your feature request related to a problem? Please describe.**  
I'm looking to add support for the ThinQ protocol described [here](https://pastebin.com/8x1CAG9u). This would start with HVAC support, but could be expanded to other devices which use the same LCW005/ThinQ dongle.

**Describe the solution you'd like**  
Implementing a new device type and appropriate driver, including a basic mini-split HVAC UI and support for the climate.mqtt integration in HA.

**Describe alternatives you've considered**  
Automations in HA - easier to implement, but not as clean as implementing it on the MQTT client (Tasmota).

**Additional context**  
I'm planning on working this, looking for an issue number.

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-05-04:
Tasmota support of IR Protocols is  based and uses the IR Lib from https://github.com/crankyoldgit/IRremoteESP8266
If you want support for your HVAC it has to be added here.

aph3rson, 2020-05-04:
@Jason2866 The ThinQ protocol does not use IR in any fashion - it's designed for replacing a dongle which communicates with the controlled device over a 5V UART connection. You can take a peek at the FCC documentation for the device [on fccid.io](https://fccid.io/BEJ-LCW005).

I'm not sure the IR library is relevant here, unless there's something I'm missing here.

Jason2866, 2020-05-04:
Ahh, yes. In this case maybe someone does jump in the boat and makes a PR.

aph3rson, 2020-05-05:
@ascillato2 @Jason2866
> **Additional context**
> I'm planning on working this, looking for an issue number.

aph3rson, 2020-05-24:
@ascillato2 while I'm developing this, I'm having some questions on style for Tasmota drivers and kosher form for the project. It appears that the Discord server is pretty lacking in devs for the Tasmota project - is there a better channel for asking these questions (to reduce the necessary changes in my eventual PR)?

ascillato, 2020-05-25:
Hi,

You can tag or PM any of us in Discord or you can ask here too if you want.

aph3rson, 2020-05-25:
@ascillato I've been in the #general channel looking for folks, I don't see you in the Discord. Would you mind tagging me when you have a second, `@aph3rson`?

ascillato2, 2020-06-09:
Hi, any news on this?
As soon as you asked I have sent you a PM.

aph3rson, 2020-06-09:
@ascillato2 I'm still in-progress - having some issues with some of the finer points of the protocol. Currently have a PoC somewhat working with console commands, still need to implement MQTT handling and the UI afterwards.

aph3rson, 2020-07-04:
Hello, yes, I am still working on this. There's some problems with the reverse-engineered protocol spec, and I've been investigating how best to tackle those.

Current progress is being made, though - there are local commands on the device to turn the unit on/off, set mode, set target temp, and fan speed.

closed 2020-08-08 by stale[bot]

infinit13, 2021-02-07:
Hi, I'm writing again after a long time.  In the meantime, I've been working on similar research to this, and I've come back to solve the problem. Finally, I will restart my own thread. Thank you.

evlo, 2023-11-09:
I'm guessing Linbus

evlo, 2023-11-09:
https://fccid.io/BEJPCRCUDT2/User-Manual/User-Manual-1911954.pdf maybe helps?

evlo, 2024-01-03:
every other connecter in the unit is jst 2.54, i wanted to sniff cn_WF uart traffic, but is 2.0 or 1.5 which i did not have at hand, figures