arendst/Tasmota #8332: KNX empty UDP packets sent to the network

oblaise, 2020-05-04:
### PROBLEM DESCRIPTION
Empty KNX packets are sent after receiving a valid KNX UDP routing packet.

Each time the device receive a valid KNX routing indication packet it generates an empty KNX routing packet (UDP payload len=0) that is send to the network.

This issue happened after the 6.6.0 -> 6.7.1 upgrade and is always present in the latest firmware.

I observed the issue when KNX related software started to issue warnings due to invalid packets after the firmware upgrade.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff S26 + development board
- [x] Tasmota binary firmware version number used: 6.6.0, 6.7.1, 8.2.0 & 8.2.0.6
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [x ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
06:26:07 RSL: RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":1,"BASE":18}
06:26:07 RSL: RESULT = {"Module":{"18":"Generic"}}
06:26:08 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"56":"Led1i"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"21":"Relay1"},"GPIO14":{"17":"Button1"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
06:27:04 RSL: STATUS = {"Status":{"Module":18,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
06:27:04 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff-knx.bin","RestartReason":"Software/System restart","Uptime":"0T00:01:31","StartupUTC":"2020-05-04T05:25:33","Sleep":50,"CfgHolder":4617,"BootCount":11,"BCResetTime":"2020-05-04T06:17:13","SaveCount":114,"SaveAddress":"F4000"}}
06:27:04 RSL: STATUS2 = {"StatusFWR":{"Version":"8.2.0.6(d410420-knx)","BuildDateTime":"2020-05-03T19:01:09","Boot":31,"Core":"2_7_0","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"371/699"}}
06:27:04 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["xxxxxx",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["82008001","2805C8000100060000005AFF000000000000","00000000","00000000"]}}
06:27:04 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":585,"Free":416,"Heap":23,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"16405E","FlashMode":3,"Features":["00000809","9FDAE597","042683B0","000000CD","010013C0","8000F181","00000024"],"Drivers":"1,2,3,4,5,6,7,8,9,10,11,12,16,18,19,22,24,26,27,30,35,37","Sensors":"1,2,3,4,5,6"}}
06:27:04 RSL: STATUS5 = {"StatusNET":{"Hostname":"sonoff-3425","IPAddress":"172.x.x.x","Gateway":"172.x.x.x","Subnetmask":"255.255.255.0","DNSServer":"172.xx.x.x","Mac":"DC:4F:22:21:4D:61","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
06:27:04 RSL: STATUS7 = {"StatusTIM":{"UTC":"2020-05-04T05:27:04","Local":"2020-05-04T06:27:04","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"05:23","Sunset":"20:10"}}
06:27:04 RSL: STATUS10 = {"StatusSNS":{"Time":"2020-05-04T06:27:04"}}
06:27:04 RSL: STATUS11 = {"StatusSTS":{"Time":"2020-05-04T06:27:04","Uptime":"0T00:01:31","UptimeSec":91,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"ON","Wifi":{"AP":1,"SSId":"xxxxx","BSSId":"00:1D:A1:76:A5:22","Channel":6,"RSSI":56,"Signal":-72,"LinkCount":1,"Downtime":"0T00:00:05"}}}
```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
N/A
```

### TO REPRODUCE
Enable KNX, Define a KNX group address linked to the Output 1 (Output 1 -> KNX group address) and define a KNX group address as input (KNX Group Address -> Output 1). As soon as you have defined the latest settings the issue start to happens on the network.

### EXPECTED BEHAVIOUR
No empty UDP packet should be send to the network

### SCREENSHOTS
[image]

### ADDITIONAL CONTEXT
N/A

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2020-05-04:
Hi, 

Thanks for reporting.

This problem is not in Tasmota but in the core. It was reported before in Tasmota at: https://github.com/arendst/Tasmota/issues/5053 and then it goes to the KNX Library: https://github.com/envy/esp-knx-ip/issues/68 but it ends up being an issue at UDP of the arduino core. In that moment was solved by Arduino Core Devs and by the use of the latest arduino core in Tasmota, this problem was solved. As you are using a precompiled bin of the latest Tasmota version, means that you are using latest arduino core too. We should investigate it and report again to the Arduino core.

Thanks for reporting.

ascillato2, 2020-05-18:
https://github.com/arendst/Tasmota/pull/8459#issuecomment-630248149

PR #8459 (2020-05-18): Fix: KNX empty UDP packets sent to the network
## Description:

The flush() call after read() causes empty packets being sent as response
for all UDP packets received on the KNX UDP port. In my tests with sonoff devices, the zero length packets mentioned in #8332 vanished after removing that flush() call.

Since I do not have any ESP32 device, it would be nice if someone else could test PR.

**Related issue (if applicable):** fixes #8332

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added  [... cut]
closed 2020-05-18 by ascillato2