arendst/Tasmota #8334: OTA update fails with XMC Flash chip

Jason2866, 2020-05-04:
OTA update fails (used latest gz Tasmota firmware) when a XMC Flash is on the ESP board.
Module Model: ESP-12F Vendor: DOITING
Flash Chip Id 0x164020

See https://github.com/esp8266/Arduino/issues/7267

To get it working (again) a serial flash is needed

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): _____
- [x] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [x] Self-compiled
    - [x] IDE / Compiler used: PlatformIO
- [x] Flashing tools used: Esptool.py  v.2.8
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
09:59:37 CMD: Backlog Template; Module; GPIO 255
09:59:37 MQT: stat/sonoff-E1242D/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
09:59:37 MQT: stat/sonoff-E1242D/RESULT = {"Module":{"18":"Generic"}}
09:59:37 MQT: stat/sonoff-E1242D/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```
10:00:27 CMD: status 0
10:00:27 MQT: stat/sonoff-E1242D/STATUS = {"Status":{"Module":18,"FriendlyName":["Tasmota"],"Topic":"sonoff-E1242D","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
10:00:27 MQT: stat/sonoff-E1242D/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"sonoffs","OtaUrl":"http://192.168.2.20/data/firmwares/tasmota.bin","RestartReason":"External System","Uptime":"0T00:06:35","StartupUTC":"2020-05-04T07:53:52","Sleep":50,"CfgHolder":4617,"BootCount":4,"BCResetTime":"2020-05-04T09:53:01","SaveCount":7,"SaveAddress":"F5000"}}
10:00:27 MQT: stat/sonoff-E1242D/STATUS2 = {"StatusFWR":{"Version":"8.2.0.6(tasmota)","BuildDateTime":"2020-05-03T23:12:17","Boot":31,"Core":"2_7_0","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"361/699"}}
10:00:27 MQT: stat/sonoff-E1242D/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Jason_Home_WLAN","Jason_Home_WLAN"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00208009","2805C8000100060000005A00000000000000","00008000","00000000"]}}
10:00:27 MQT: stat/sonoff-E1242D/STATUS4 = {"StatusMEM":{"ProgramSize":528,"Free":472,"Heap":23,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164020","FlashMode":3,"Features":["00000809","8F8AE787","040003A1","00640487","001026C0","0440A000","00000020"],"Drivers":"1,2,3,4,5,6,8,9,10,20,26,30","Sensors":"1,3,6,9,15,18,19,33,35,40,60,62"}}
10:00:27 MQT: stat/sonoff-E1242D/STATUS5 = {"StatusNET":{"Hostname":"sonoff-E1242D-1069","IPAddress":"192.168.2.146","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.1","Mac":"50:02:91:E1:24:2D","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
10:00:27 MQT: stat/sonoff-E1242D/STATUS6 = {"StatusMQT":{"MqttHost":"mqtt.lan","MqttPort":1883,"MqttClientMask":"sonoff-%06X","MqttClient":"sonoff-E1242D","MqttUser":"openhabian","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
10:00:27 MQT: stat/sonoff-E1242D/STATUS7 = {"StatusTIM":{"UTC":"2020-05-04T08:00:27","Local":"2020-05-04T10:00:27","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":99,"Sunrise":"05:46","Sunset":"20:32"}}
10:00:27 MQT: stat/sonoff-E1242D/STATUS10 = {"StatusSNS":{"Time":"2020-05-04T10:00:27"}}
10:00:27 MQT: stat/sonoff-E1242D/STATUS11 = {"StatusSTS":{"Time":"2020-05-04T10:00:27","Uptime":"0T00:06:35","UptimeSec":395,"Vcc":3.385,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"Jason_Home_WLAN","BSSId":"8C:53:C3:85:17:96","Channel":9,"RSSI":100,"Signal":-39,"LinkCount":1,"Downtime":"0T00:00:04"}}}
```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
Flash Tasmota on a device with a XMC flash and do a OTA update

### EXPECTED BEHAVIOUR
Working OTA update

arendst, 2020-05-04:
Does it only fail on GZip or also on a normal binary?

Jason2866, 2020-05-04:
Yes for both, PR https://github.com/esp8266/Arduino/pull/6725 causes the issue

closed 2020-05-04 by ascillato2