arendst/Tasmota #8341: I2C PT2257 Digital Potentiometer Support

ceinem, 2020-05-04:
**Have you looked for this feature in other issues and in the docs?**  
Yes

**Is your feature request related to a problem? Please describe.**  
I am building an IoT Music System and would like to control it using Tasmota. For controlling the volume I'm using a PT2257 Digital Potentiometer which can be controlled using I2C.

**Describe the solution you'd like**  
Simple addition of I2C support for the PT2257. Commands include increasing and decreasing volume on one or both channels or muting them altogether. 

**Describe alternatives you've considered**  
Software defined Volume control, but this limits my possibilities as I would only be able to control digital input signals and not analog audio input. 

**Additional context**  
Datasheet can be found [here](https://www.rcscomponents.kiev.ua/datasheets/pt2257-datasheet.pdf) and an existing Arduino Library [here](https://github.com/victornpb/Evc_pt2257)

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-05-04:
Just for reference, this feature has been requested before ( https://github.com/arendst/Tasmota/issues/6421 )

ceinem, 2020-05-04:
Thanks for the info. I would like to try to implement the feature myself. I have working code to control it (without the linked library, just onewire), but so far I'm controlling it with Blynk. 
Are there any similar devices already supported by tasmota where I could use the code as a base? 
I would be thankful for any guidance in this issue

closed 2020-06-05 by stale[bot]

segaura, 2020-08-16:
> Thanks for the info. I would like to try to implement the feature myself. I have working code to control it (without the linked library, just onewire), but so far I'm controlling it with Blynk.
> Are [...]

I'm in a similar boat, I want to drive a MPC41010 which needs only elementary SPI code.  My understanding is that I ought to create a new Tasmota "peripheral" (and then made it available through a template) but I don't know how to start from.

ceinem, 2020-08-16:
Yeah, I tried for a couple of hours, but couldn't find enough documentation/help, so I gave up and wrote my own firmware. so now I have one device without Tasmota...

segaura, 2020-08-16:
Yes, I can also go the short way, in my case with a headless solution reading an MQTT topic and driving the digital pot accordingly, but thought that having this working in the tasmota ecosystem would have been long cooler.  Especially because I did a very simple and repeatable D1 mini shield.
I will try with my "two hours", if no one hints me meanwhile, thank you.

ceinem, 2020-08-18:
yeah, I'm in the same situation. Let me know if you make any progress!

macgeeknz, 2022-11-08:
I guess this is quite old now and possibly not much use, but I was looking to use Tasmota with a DS1882 digipot when I came across your posting. I see Tasmota supports DS3502 chip, have you considered changing your hardware to use that?

ceinem, 2022-11-11:
No, i didn't. Still running my custom firmware. Maybe one day I'll put in the work to make it nice and tidy.

macgeeknz, 2022-11-11:
Hey if it works it works eh :) I'm running a custom sketch using a DS1882 digipot myself, but I think I'll have a go at submitting a patch to Tasmota to include support because it looks like it shouldn't be too much work to modify the 3502 driver to work with the 1882.