arendst/Tasmota #8345: Is there a need for the OpenTherm integration?

yury-sannikov, 2020-05-04:
Hi here.
I made an OpenTherm integration for my home project. I can be used with https://github.com/jpraus/arduino-opentherm or http://ihormelnyk.com/opentherm_adapter boards, but not with OTG.

Is there any interest in such integration so I invest some time to prepare a PR against the tasmota repo?

https://github.com/yury-sannikov/Sonoff-Tasmota-Dimmer/blob/8.2.0_local_ot/tasmota/xsns_85_opentherm.ino
https://github.com/yury-sannikov/Sonoff-Tasmota-Dimmer/blob/8.2.0_local_ot/tasmota/xsns_85_opentherm_protocol.ino

Also, have HomeAssistant complimentary component https://github.com/yury-sannikov/ha-mqtt-water-heater to wire it up if someone needed.

closed 2020-05-04 by close-issue-app[bot]

nonyhaha, 2020-12-17:
What, really? Nobody replied? 
Please, at least make a clearer instructiuon on how to setup everything :) 
I see on this page https://tasmota.github.io/docs/OpenTherm/ that we need to add a line to a specific file and compile our own firmware. Also, there is something mentioned about gpio0, but I do not understand what it means by "The integration attaches ISR to the RX GPIO to avoid pooling. That GPIO should support interrupts. Please consider this while working on the schematics."

ascillato, 2020-12-17:
Please, see above, there is a link to the complete issue. You just reached an empty issue closed by the bot because was the issue template was deleted while opening. Then another issue (linked above) was submitted and all the discussion is there. Check that please. Thanks.