arendst/Tasmota #8346: Is there a need for the OpenTherm integration?

yury-sannikov, 2020-05-04:
**Have you looked for this feature in other issues and in the docs?**  
Yes

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
Hi here.
I made an OpenTherm integration for my home project. I can be used with https://github.com/jpraus/arduino-opentherm or http://ihormelnyk.com/opentherm_adapter boards, but not with OTG.

Is there any interest in such integration so I invest some time to prepare a PR against the tasmota repo?

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._

Component code:
https://github.com/yury-sannikov/Sonoff-Tasmota-Dimmer/blob/8.2.0_local_ot/tasmota/xsns_85_opentherm.ino
https://github.com/yury-sannikov/Sonoff-Tasmota-Dimmer/blob/8.2.0_local_ot/tasmota/xsns_85_opentherm_protocol.ino

Also, have HomeAssistant complimentary component https://github.com/yury-sannikov/ha-mqtt-water-heater to wire it up if someone needed.

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._
none

**Additional context**  
_Add any other context or screenshots about the feature request here._
[image]

**(Please, remember to close the issue when the problem has been addressed)**

nagyrobi, 2020-05-04:
In what aspect is opentherm better than a simple thermostat? Can it be used with any boiler?

meingraham, 2020-05-05:
Any commonality with this PR that could be merged or development effort shared?

https://github.com/arendst/Tasmota/pull/8212

arijav, 2020-05-05:
> Any commonality with this PR that could be merged or development effort shared?
> 
> #8212

I see no overlap between my PR and OpenTherm, what I see is good possibility to integrate inputs coming from the boiler into my driver. Getting the boiler temperature as well as whether we get heating water or cooling one could enrich the performance of the driver (have to evaluate).

yury-sannikov, 2020-05-06:
Agree with @arijav It's better to keep them completely separate. I think the best approach would be to use Rules to get PID output from telemetry and set boiler temperature based on it. I have a special command designed for the PID controllers. It does not write temperature to the Settings saving flash writes.

arendst, 2020-05-06:
I would also like to keep the two seperate. Especially as OpenTherm will be rules based and therefore lightweight (I hope).

ascillato2, 2020-05-06:
Just for reference for anyone searching in other issues, this was already requested in old issues at https://github.com/arendst/Tasmota/issues/8345 and https://github.com/arendst/Tasmota/issues/2295

arijav, 2020-05-06:
> I would also like to keep the two seperate. Especially as OpenTherm will be rules based and therefore lightweight (I hope).

Yes, as I see my driver and OpenTherm are meant for different things:

1- Thermostat Driver: Is meant to act as a replacement or bypass of a feature reach room thermostat (Something like this: https://www.busch-jaeger.de/produkt/future-linear/knx-raumtemperaturregler-mit-tastsensor)

2- OpenTherm: Is meant to interface with the boiler, provide interfaces for telemetry data and commands (such set point).

They can both complement themselves over rules. My driver can be used via a timer for setting the room temperature and OpenTherm for driving down the boiler if f.i. no heating is needed in the rooms or regulating the temperature depending on the load.

Of course if there is need in OpenTherm of a PI to regulate the set point softly of the boiler I could provide an interface in the driver.

nagyrobi, 2020-05-06:
> 1- Thermostat Driver: Is meant to act as a replacement or bypass of a feature reach room thermostat. It outputs a PWM to actuate heating valves for instance. Something like this: https://www.busch-jaeger.de/produkt/future-linear/knx-raumtemperaturregler-mit-tastsensor

Interesting. Until now, having a Thermostat built-in Tasmota [was](https://github.com/arendst/Tasmota/issues/481) a [taboo](https://github.com/arendst/Tasmota/issues/818).

yury-sannikov, 2020-05-07:
> Of course if there is need in OpenTherm of a PI to regulate the set point softly of the boiler I could provide an interface in the driver.

IMHO, we can't foresee all the use cases ppl might come up with. One universal "interface" might be a telemetry output. So ppl can do something as I mentioned in the comment https://github.com/arendst/Tasmota/pull/8212#issuecomment-625204237 by setting up rules on the telemetry output.

https://tasmota.github.io/docs/Rules/#rule-trigger see 
```
Tele-[SensorName]#[ValueName]
```

closed 2020-05-07 by arendst

nonyhaha, 2020-12-18:
Hello, 
I am having trouble understanding a statement on the tasmota opentherm instructions page. 
It say that: "The integration attaches ISR to the RX GPIO to avoid pooling. That GPIO should support interrupts. Please consider this while working on the schematics." 

How should I proceed? If I tie something to gpio0, it will not boot. Should I use another gpio instead? 

THanks!

PvdGulik, 2021-01-21:
Hello,

I'm very interested in this project. I would like to receive the open therm messages via MQTT. I have a OTGW with a nodemcu.
Is this possible with Tasmota? 

Thanks

PR #8373 (2020-05-07): OpenTherm sensor implementation
## Description:

OpenTherm sensor implementation. Acts as an OpenTherm master device. It can be used with https://github.com/jpraus/arduino-opentherm or http://ihormelnyk.com/opentherm_adapter shields. It _does not_ supports another OpenTherm master device such as a remote panel or physical OpenTherm thermostat.

Features:
- Control Central Heating temperature (CH)
- Control Domestic Hot Water temperature (DHW)
- Read boiler error codes
- Read boiler telemetry information
- Automatically detect supported OpenTherm commands
- Designed to be used with PID applicat [... cut]