arendst/Tasmota #8347: resolve DNS doesn't work in version 8.2.0

garvarma, 2020-05-04:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

Hi,

I'm using differents versions of tasmota (6.5.0 and 8.2.0) in differents devices. For example in led strip light (https://templates.blakadder.com/blitzwolf_BW-LT11.html) or 3 power strip (https://templates.blakadder.com/s2199EU.html) i'm using 6.5.0. In 1 jinvoo smart valve (https://templates.blakadder.com/jinvoo_SM-AW713-v2.html) i'm using version 8.2.0.

The problem is that in version 8.2.0 (also i checked in version 7.1.1) the dns doesn't work correctly.

To be exacly, i has configured the devices with DHCP, and DNS server is the correct on devices. But i the MQTT server i have configured de name, and in versions 7.1.1 or 8.2.0, device can't connect to the broker.
If i put the IP then works well. The same configuration un 6.5.0 works well.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
![imagen](https://user-images.githubusercontent.com/18529175/80996341-f142eb80-8e3f-11ea-809f-b78d10dd49bb.png)
![imagen](https://user-images.githubusercontent.com/18529175/80996362-f9029000-8e3f-11ea-9820-55bfa966f861.png)

But when i see console, i can see that device with version 8.2.0, can't connect to the mqtt server by dns, and in both cases have the same configuration by dns server

imagen.

If i put the IP in the device with version 8.2.0, as y said, then works well.

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-05-04:
Do NOT use `.local` it will not work. `.local` is reserved for mDNS service.
Mixing mDNS in a network where a DNS server is active will result in problems.
There is NO issue in Tasmota. I use DNS with all my Tasmota devices. 
I have not a single IP address in any device

garvarma, 2020-05-04:
@Jason2866 ,

Thanks for reply, so un older versions (like 6.5.0) the use of `.local` works and in newest doesn't?

meingraham, 2020-05-04:
@garvarma  Have you tried `hostname`.lan?

garvarma, 2020-05-04:
@meingraham,

Just now i tried, and doesn't work. I need to change in my DNS server de domain name.

`19:45:53 CMD: Status 0
19:45:53 RSL: stat/tledhab4/STATUS = {"Status":{"Module":0,"FriendlyName":["TLedhab4"],"Topic":"tledhab4","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
19:45:53 RSL: stat/tledhab4/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:33","StartupUTC":"2020-05-04T18:45:20","Sleep":50,"CfgHolder":4617,"BootCount":14,"SaveCount":36,"SaveAddress":"F9000"}}
19:45:53 RSL: stat/tledhab4/STATUS2 = {"StatusFWR":{"Version":"7.1.1(basic)","BuildDateTime":"2019-12-01T13:01:19","Boot":4,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX"}}
19:45:53 RSL: stat/tledhab4/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["IRONMAN2",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
19:45:53 RSL: stat/tledhab4/STATUS4 = {"StatusMEM":{"ProgramSize":458,"Free":544,"Heap":31,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440C8","FlashMode":3,"Features":["00000809","8F082383","041083A0","20021706","010001C0","00000000"],"Drivers":"1,2,3,4,9,10,16,20,21","Sensors":"3"}}
19:45:53 RSL: stat/tledhab4/STATUS5 = {"StatusNET":{"Hostname":"tledhab4-5781","IPAddress":"192.168.1.171","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.100","Mac":"CC:50:E3:14:96:95","Webserver":2,"WifiConfig":4}}
19:45:53 RSL: stat/tledhab4/STATUS6 = {"StatusMQT":{"MqttHost":"mqttserver.lan","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_149695","MqttUser":"homeassistant","MqttCount":0,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
19:45:53 RSL: stat/tledhab4/STATUS7 = {"StatusTIM":{"UTC":"Mon May 04 18:45:53 2020","Local":"Mon May 04 19:45:53 2020","StartDST":"Sun Mar 29 02:00:00 2020","EndDST":"Sun Oct 25 03:00:00 2020","Timezone":"+01:00","Sunrise":"05:23","Sunset":"20:10"}}
19:45:53 RSL: stat/tledhab4/STATUS10 = {"StatusSNS":{"Time":"2020-05-04T19:45:53"}}
19:45:53 RSL: stat/tledhab4/STATUS11 = {"StatusSTS":{"Time":"2020-05-04T19:45:53","Uptime":"0T00:00:33","UptimeSec":33,"Vcc":3.509,"Heap":31,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":22,"MqttCount":0,"POWER":"OFF","Dimmer":10,"Color":"1A1A1A00","HSBColor":"0,0,10","Channel":[10,10,10,0],"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"IRONMAN2","BSSId":"F2:9F:C2:7A:92:8A","Channel":1,"RSSI":66,"LinkCount":1,"Downtime":"0T00:00:06"}}}
19:46:01 MQT: Attempting connection...
19:46:01 MQT: Connect failed to mqttserver.lan:1883, rc -2. Retry in 10 sec
19:46:12 MQT: Attempting connection...
19:46:12 MQT: Connect failed to mqttserver.lan:1883, rc -2. Retry in 10 sec`

Jason2866, 2020-05-04:
Yes, in older versions the use of `.local` may have worked. This is not a function from Tasmota
Tasmota uses the Arduino ESP8266 core (API). The used Arduino core has changed to actual
version since many bugs are fixed. The behaviour of mDNS has changed too. Now mDNS
acts as recommended in the RFCs

ascillato2, 2020-05-06:
Hi, any news on this?

Jason2866, 2020-05-08:
Try latest Tasmota development version. It is build with Arduin core 2.7.1 which has change for mDNS (again) https://github.com/esp8266/Arduino/commit/bf718c39afe88f9fc2721a25028ca96665b006d7

ascillato2, 2020-05-09:
Hi,

When you have some time, please tell us if using latest version solves your issue. Thanks.

closed 2020-05-09 by ascillato2