arendst/Tasmota #8355: more pcf8574 generate a issue on mqtt publishing

megamarco833, 2020-05-05:
### PROBLEM DESCRIPTION
connecting more than one PCF8574 to tasmota it generate an mqtt unexpectet publishing

### REQUESTED INFORMATION
- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): NodeMCU
- [x] Tasmota binary firmware version number used: 8.0.2
  - [ ] Pre-compiled
  - [x] Self-compiled
    - [x] IDE / Compiler used: GITPOD
- [x] Flashing tools used: espflasher
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
19:55:22 CMD: Backlog Template; Module; GPIO 255
19:55:22 MQT: stat/tasmota_esp_160/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
19:55:22 MQT: stat/tasmota_esp_160/RESULT = {"Module":{"18":"Generic"}}
19:55:23 MQT: stat/tasmota_esp_160/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"6":"I2C SDA"},"GPIO5":{"5":"I2C SCL"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
no rules active

```
- [ ] Provide the output of this command: `Status 0`:
```
19:56:53 CMD: Status 0
19:56:53 MQT: stat/tasmota_esp_160/STATUS = {"Status":{"Module":18,"FriendlyName":["Tasmota","Tasmota2","Tasmota3","Tasmota4","","","",""],"Topic":"tasmota_esp_160","ButtonTopic":"0","Power":16777215,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
19:56:53 MQT: stat/tasmota_esp_160/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Exception","Uptime":"0T00:02:22","StartupUTC":"2020-05-05T18:54:31","Sleep":50,"CfgHolder":4617,"BootCount":12,"BCResetTime":"2020-05-05T03:04:33","SaveCount":57,"SaveAddress":"F9000"}}
19:56:53 MQT: stat/tasmota_esp_160/STATUS2 = {"StatusFWR":{"Version":"8.2.0.6(tasmota)","BuildDateTime":"2020-05-05T17:53:28","Boot":31,"Core":"2_7_0","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"340/699"}}
19:56:53 MQT: stat/tasmota_esp_160/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["NETGEAR",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","80000000","00000000"]}}
19:56:53 MQT: stat/tasmota_esp_160/STATUS4 = {"StatusMEM":{"ProgramSize":587,"Free":416,"Heap":17,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164020","FlashMode":2,"Features":["00000809","8FDAE797","043683A0","000000CD","010013C0","C000FB81","00000024"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,28,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
19:56:53 MQT: stat/tasmota_esp_160/STATUS5 = {"StatusNET":{"Hostname":"tasmota_esp_160-2924","IPAddress":"192.168.0.160","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.5","Mac":"50:02:91:E1:4B:6C","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
19:56:53 MQT: stat/tasmota_esp_160/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.105","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_E14B6C","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
19:56:53 MQT: stat/tasmota_esp_160/STATUS7 = {"StatusTIM":{"UTC":"2020-05-05T18:56:53","Local":"2020-05-05T19:56:53","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"05:22","Sunset":"20:11"}}
19:56:53 MQT: stat/tasmota_esp_160/STATUS10 = {"StatusSNS":{"Time":"2020-05-05T19:56:53"}}
19:56:53 MQT: stat/tasmota_esp_160/STATUS11 = {"StatusSTS":{"Time":"2020-05-05T19:56:53","Uptime":"0T00:02:22","UptimeSec":142,"Heap":16,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":24,"MqttCount":1,"POWER1":"ON","POWER2":"ON","POWER3":"ON","POWER4":"ON","POWER5":"ON","POWER6":"ON","POWER7":"ON","POWER8":"ON","POWER9":"ON","POWER10":"ON","POWER11":"ON","POWER12":"ON","POWER13":"ON","POWER14":"ON","POWER15":"ON","POWER16":"ON","POWER17":"ON","POWER18":"ON","POWER19":"ON","POWER20":"ON","POWER21":"ON","POWER22":"ON","POWER23":"ON","POWER24":"ON","Wifi":{"AP":1,"SSId":"NETGEAR","BSSId":"80:3F:5D:6E:37:67","Channel":5,"RSSI":50,"Signal":-75,"LinkCount":1,"Downtime":"0T00:00:03"}}}
19:56:53 MQT: stat/tasmota_esp_160/STATUS12 = {"StatusSTK":{"Exception":29,"Reason":"Exception","EPC":["4000df64","00000000","00000000"],"EXCVADDR":"00000000","DEPC":"00000000","CallChain":["401015bc","402606c7","4026065c","40260603","4025f760","4025f789","4025d1f4","401010b8","4026016b","4025955d","401061fd","4000050c","4025deeb","4024f2e8","40101cd2","4025d95e","402670af","4026696f","40245308","40000f49","40000f49","40000e19","40001878","40105a5d","40105a63","400060d0","4010000d","40284354","40265858","40265809","401010b8"]}}

```
- [x] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
I installed on tasmota n°3 pcf8574. They are recognized, and set as output, but when i trigger relays, for example:
power20 1
i see that tasmota (only with some power<x>) is sending a mqtt topic domoticz/in 
it happened not with all power<x> ( but with many power<x>) 
i do not compile nothing inside webpage of domoticz (all value at 0) and i see that the idx that are used to publish are very strange...in some case with negative idx.
also the numbers of idx seams strange (very high number, in only one case the idx is 30 that's the worst case because someone can have an avaliable device at idx=30 that can be trigger involontary)

it happened only with 2nd pcf (relay from9 to 16) and 3rd pcf (relay from 17 to 24)
below some logs as example:
```
18:41:06 CMD: power16 1
18:41:06 MQT: stat/tasmota_esp_160/RESULT = {"POWER16":"ON"}
18:41:06 MQT: stat/tasmota_esp_160/POWER16 = ON
18:41:14 CMD: power14 1
18:41:14 MQT: stat/tasmota_esp_160/RESULT = {"POWER14":"ON"}
18:41:14 MQT: stat/tasmota_esp_160/POWER14 = ON
18:41:21 CMD: power18 1
18:41:21 MQT: domoticz/in = {"idx":1966090,"nvalue":1,"svalue":"","Battery":100,"RSSI":6}
18:41:21 MQT: stat/tasmota_esp_160/RESULT = {"POWER18":"ON"}
18:41:21 MQT: stat/tasmota_esp_160/POWER18 = ON
18:41:30 CMD: power24 1
18:41:30 MQT: domoticz/in = {"idx":-2147483648,"nvalue":1,"svalue":"","Battery":100,"RSSI":5}
18:41:30 MQT: stat/tasmota_esp_160/RESULT = {"POWER24":"ON"}
18:41:30 MQT: stat/tasmota_esp_160/POWER24 = ON
[... 64 lines cut ...]
19:05:23 MQT: domoticz/in = {"idx":-2147483648,"nvalue":1,"svalue":"","Battery":100,"RSSI":5}
19:05:23 MQT: stat/tasmota_esp_160/RESULT = {"POWER24":"ON"}
19:05:23 MQT: stat/tasmota_esp_160/POWER24 = ON

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
just try to trigger relay of PCFs installer using comand power<x> 1 or power<x> 0

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
here the expecation are two different thing:
1) if tasmota's web page of domoticz is NOT filled:
     - tasmota should not publish in topic domoticz/in nothing.
2) if tasmota's web page of domoticz is filled:
    - tasmota should publish inside domoticz/in the idx written inside the dedicated webpage.
i tested to fill the idx inside the webpage but tasmota not publish :(
I notice that idx allow for domotic inside dedicated web page are only 1 to 4 ....is it possible to extend?
if 1 pcf8574 => 8 idxs are required (one for every output)
if 2 pcf => 16 idxs are required
if 3 pcf => 24 idxs
...and so on.

THANKS !!

arendst, 2020-05-06:
Let's start off to say that Tasmota only supports ONE PCF8574.

That rules out all of your problems.

I'll have a quick look why you start seeing Domoticz messages above 16 relays but you probably could deduct that yourselves; there must be something with uint16_t instead of uint32_t.

Also, Tasmota will NEVER SUPPORT more than 32 relays/devices simply because there is no room planned for more; the used uint32_t can only hold 32 relays which I think is already way too much.

arendst, 2020-05-06:
There is really something wring with Domoticz array range checking (no checking ;-().

I will fix this now.

sfromis, 2020-05-06:
If only supporting ONE PCF8574, I'm surprised by the comments I see in the xdrv_28_pcf8574.ino driver like
```
// Max numbers of devices comming from PCF8574 modules
// Max numbers of PCF8574 modules
// suport up to 8 boards PCF8574
```
Seeing it generate up to POWER24 seemed to confirm my impression of the setup being supported, even if I initially expected just one chip to fit.

arendst, 2020-05-06:
With this fix at least the Domoticz message is fixed.

Max relays as still 32 so the amount of "PCF8574 boards" is at max 4. The PCF8574 driver supports 8 boards, Tasmota does not. It all comes down to who writes what code.

sfromis, 2020-05-06:
Would it then make sense to reduce MAX_PCF8574 in tasmota.h to 4?
`const uint8_t MAX_PCF8574 = 8;              // Max number of PCF8574 devices`

arendst, 2020-05-06:
Yes. I'll do it now.

closed 2020-05-06 by ascillato2

megamarco833, 2020-05-06:
> 
> 
> Yes. I'll do it now.

```
hi thans for getting into this and fix.
i compile and flash:
19:59:05 CMD: power24 1
19:59:05 MQT: stat/tasmota_esp_160/RESULT = {"POWER24":"ON"}
19:59:05 MQT: stat/tasmota_esp_160/POWER24 = ON
19:59:14 CMD: power24 0
19:59:14 MQT: stat/tasmota_esp_160/RESULT = {"POWER24":"OFF"}
19:59:14 MQT: stat/tasmota_esp_160/POWER24 = OFF
19:59:59 CMD: power19 1
19:59:59 MQT: stat/tasmota_esp_160/RESULT = {"POWER19":"ON"}
19:59:59 MQT: stat/tasmota_esp_160/POWER19 = ON
20:00:23 CMD: power10 1
20:00:23 MQT: stat/tasmota_esp_160/RESULT = {"POWER10":"ON"}
20:00:23 MQT: stat/tasmota_esp_160/POWER10 = ON
```
it's seams working.
as i understood from discords, due to a memory limitation is not allow to support more than 4 idx.
so to use more relays (i need 18) i need to use something else that can make "talk" tasmota and domoticz, correct?

last question, i tested one pcf where i connect 8relay
i use it to control my blinds.
but in this case is not working the idx set in tasmota. it not comunicate properly with domoticz device (set as dummy device type blinds with percentage inverted)

thanks again

Protoncek, 2020-12-25:
(after a while...)
Hello guys! I'm playing with PCF8574 a bit and i just wonder, is it this chip obsolete? 
I'm trying to use it with home assistant with no luck... I can't even read inputs in web console (or anywhere...). 
I can toggle outputs in web console, but neither inputs or outputs are available as sensors or switches in home assistant.
What am i doing wrong? MCP23017 works great with inputs, but not so much with outputs (doesn't publish change until next teleperiod interval), but pcf8574 doesn't work at all, except integrated relay (i'm using sonoff re5v1 module).

Any help would do...

stefanbode, 2020-12-27:
I did the PCF driver and 8 is theoretically. Also you can use the PCF to detect high low as buttons (not yet implemented). Anyhow Arendt is right that 32 is the absolutely max. Even is supporting up to 8 boards. This is more or less to have access to all the addresses the board can assign.

stefanbode, 2020-12-27:
And the MCG is also my favorite board now. Moved away Frommheit PCF.

Protoncek, 2020-12-27:
Yeah, i went to MCP, too. It’s way more flexible and it uses interrupts to see input changes. I arranged it so that i have 8 inputs and 8 outputs (via ULN2803).
I just wanted to test PCF chip since i have them at home. On PCF i can switch output with mqtt message, but if i choose to be input there is no way to see input state change unless i would constantly check with mqtt messages. 
Oh, and i agree totally that 32 is max, i can’t imagine where i would need so much ports in a single place anyway. I ordeded a bunch of very cheap sonoff re5v1 modules, some bme280 and ds1820 modules and now i’m placing them around my property... At the moment only place i need MCE is my alarm control panel.
So, i guess that we could indeed say PCF to be “kinda” obsolete...

turin400, 2021-03-01:
> I did the PCF driver and 8 is theoretically. Also you can use the PCF to detect high low as buttons (**not yet implemented)**. Anyhow Arendt is right that 32 is the absolutely max. Even is supporting up to 8 boards. This is more or less to have access to all the addresses the board can **assign.**

I would like to know if you are going to finish configuring the pcf8574 driver to accept inputs. Recently I have created a pcb thinking that they work and having problems I have seen that it was still not supported.
Thanks.