arendst/Tasmota #8356: ENERGY_Today value suddenly jumps by +~5.2kWh, repeatably - without consumer

jensb, 2020-05-05:
### PROBLEM DESCRIPTION
On a Gosund SP112 socket, the "ENERGY_Today" value suddenly jumps up +5.2kWh in the middle of the night (i.e. when no device is drawing any current). This has been repeated twice in consecutive nights.

### REQUESTED INFORMATION

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Gosund SP112
- [X] Tasmota binary firmware version number used: 8.2.0
  - [X] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [X] Flashing tools used: tuya-convert
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
19:29:02 CMD: Backlog Template; Module; GPIO 255
19:29:02 MQT: stat/tasmota/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
19:29:02 MQT: stat/tasmota/RESULT = {"Module":{"55":"Gosund SP1 v23"}}
19:29:03 MQT: stat/tasmota/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"158":"LedLinki"},"GPIO2":{"0":"None"},"GPIO3":{"17":"Button1"},"GPIO4":{"134":"BL0937 CF"},"GPIO5":{"132":"HLWBL CF1"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"131":"HLWBL SELi"},"GPIO13":{"56":"Led1i"},"GPIO14":{"21":"Relay1"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}
1
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
19:30:50 MQT: stat/tasmota/STATUS = {"Status":{"Module":55,"FriendlyName":["Arbeitszimmer-PC-Jens"],"Topic":"tasmota","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
19:30:50 MQT: stat/tasmota/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:17:04","StartupUTC":"2020-05-05T18:13:46","Sleep":50,"CfgHolder":4617,"BootCount":72,"BCResetTime":"2020-05-05T14:23:21","SaveCount":205,"SaveAddress":"FA000"}}
19:30:50 MQT: stat/tasmota/STATUS2 = {"StatusFWR":{"Version":"8.2.0(tasmota)","BuildDateTime":"2020-03-20T14:45:23","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8285","CR":"412/699"}}
19:30:50 MQT: stat/tasmota/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["FRITZ!Box 7590 UN",""],"TelePeriod":10,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
19:30:50 MQT: stat/tasmota/STATUS4 = {"StatusMEM":{"ProgramSize":577,"Free":424,"Heap":24,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","000000CD","010013C0","C000F981","00000004"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
19:30:50 MQT: stat/tasmota/STATUS5 = {"StatusNET":{"Hostname":"tasmota-4121","IPAddress":"192.168.178.63","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.178.1","Mac":"2C:F4:32:A2:10:19","Webserver":2,"WifiConfig":2,"WifiPower":17.0}}
19:30:50 MQT: stat/tasmota/STATUS6 = {"StatusMQT":{"MqttHost":"raspberrypi.fritz.box","MqttPort":1883,"MqttClientMask":"Arbeitszimmer_PC_Jens","MqttClient":"Arbeitszimmer_PC_Jens","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
19:30:50 MQT: stat/tasmota/STATUS7 = {"StatusTIM":{"UTC":"2020-05-05T18:30:50","Local":"2020-05-05T19:30:50","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"05:22","Sunset":"20:11"}}
19:30:50 MQT: stat/tasmota/STATUS9 = {"StatusPTH":{"PowerDelta":0,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
19:30:50 MQT: stat/tasmota/STATUS10 = {"StatusSNS":{"Time":"2020-05-05T19:30:50","ENERGY":{"TotalStartTime":"2020-03-19T21:45:24","Total":44.763,"Yesterday":0.740,"Today":5.924,"Power":128,"ApparentPower":182,"ReactivePower":129,"Factor":0.70,"Voltage":234,"Current":0.778}}}
19:30:50 MQT: stat/tasmota/STATUS11 = {"StatusSTS":{"Time":"2020-05-05T19:30:50","Uptime":"0T00:17:04","UptimeSec":1024,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"FRITZ!Box 7590 UN","BSSId":"F0:B0:14:AF:7B:31","Channel":11,"RSSI":100,"Signal":-47,"LinkCount":1,"Downtime":"0T00:00:06"}}}
19:30:58 MQT: tele/tasmota/STATE = {"Time":"2020-05-05T19:30:58","Uptime":"0T00:17:12","UptimeSec":1032,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"FRITZ!Box 7590 UN","BSSId":"F0:B0:14:AF:7B:31","Channel":11,"RSSI":100,"Signal":-47,"LinkCount":1,"Downtime":"0T00:00:06"}}
19:30:58 MQT: tele/tasmota/SENSOR = {"Time":"2020-05-05T19:30:58","ENERGY":{"TotalStartTime":"2020-03-19T21:45:24","Total":44.763,"Yesterday":0.740,"Today":5.924,"Period":0,"Power":128,"ApparentPower":182,"ReactivePower":130,"Factor":0.70,"Voltage":235,"Current":0.775}}
```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
(not available, the issue happened at night, when nobody was watching - I only found out about it later)
```

### TO REPRODUCE
No special steps are required. The device just began showing this behaviour with v8.1.x, I updated to v8.2.0 but the behaviour stayed the same.

### EXPECTED BEHAVIOUR
ENERGY_Today should not suddenly jump by any amount without the current actually being drawn. And this was definitely not the case, I would notice 5 additional kWh on our house meter.

### SCREENSHOTS

### ADDITIONAL CONTEXT

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2020-05-06:
Hi,

For troubleshooting we need more information. You will need to set a SYSLOG to save everything your device does. Please set it to level 4. On how to set a SYSLOG, please check the Tasmota docs.

With that log we should be able to see the moment in which it jumps to search for the problem (just an issue in your hardware, bad internal power supply; or in the software like that the data received from the power measurement chip had an CRC error and was not correctly decoded)

Not related to your issue but you have `"SerialLog":2` enabled and you are using TX and RX GPIOs for a led and a button. Serilallog should be 0 for your gpio settings.

jensb, 2020-05-06:
OK. I set SysLog to 4, opened my raspberry syslogd for UDP messages and I am now receiving data from the plug.

```
May  6 21:34:32 tasmota-4121 ESP-MQT: tele/tasmota/STATE = {"Time":"2020-05-06T20:34:32","Uptime":"1T01:20:46","UptimeSec":91246,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":7,"POWER":"ON","Wifi":{"AP":1,"SSId":"xx","BSSId":"xx","Channel":11,"RSSI":100,"Signal":-47,"LinkCount":1,"Downtime":"0T00:00:06"}}
May  6 21:34:32 tasmota-4121 ESP-MQT: tele/tasmota/SENSOR = {"Time":"2020-05-06T20:34:32","ENERGY":{"TotalStartTime":"2020-03-19T21:45:24","Total":45.681,"Yesterday":6.223,"Today":0.619,"Period":0,"Power":98,"ApparentPower":148,"ReactivePower":111,"Factor":0.66,"Voltage":236,"Current":0.626}}
```
Will post the syslog output if this happens again.
Thanks!

jensb, 2020-05-09:
It happened today again with v8.2, but different compared to before the upgrade.

At ~10:45 I noticed the web interface was unavailable. The device still reported state via syslog though and it was apparently working. Due to lack of other options I eventually power cycled it. After the reboot, the "ENERGY_Today" value had again jumped by ~5.8kWh.

```
May  9 10:44:57 tasmota-4121 ESP-MQT: tele/tasmota/SENSOR = {"Time":"2020-05-09T09:44:57","ENERGY":{"TotalStartTime":"2020-03-19T21:45:24","Total":47.360,"Yesterday":0.638,"Today":0.067,"Period":0,"Power":44,"ApparentPower":54,"ReactivePower":32,"Factor":0.80,"Voltage":235,"Current":0.231}}
May  9 10:45:18 tasmota-4121 ESP-MQT: tele/tasmota/SENSOR = {"Time":"2020-05-09T09:45:17","ENERGY":{"TotalStartTime":"2020-03-19T21:45:24","Total":47.361,"Yesterday":0.638,"Today":0.068,"Period":0,"Power":76,"ApparentPower":116,"ReactivePower":88,"Factor":0.65,"Voltage":235,"Current":0.493}}
May  9 10:45:20 tasmota-4121 ESP-WIF: Checking connection...
May  9 10:45:45 tasmota-4121 ESP-MQT: tele/tasmota/STATE = {"Time":"2020-05-09T09:45:45","Uptime":"3T14:31:59","UptimeSec":311519,"Heap":17,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":64,"MqttCount":12,"POWER":"ON","Wifi":{"AP":1,"SSId":"FRITZ!Box 7590 UN","BSSId":"98:9B:CB:79:5C:75","Channel":11,"RSSI":12,"Signal":-94,"LinkCount":6,"Downtime":"0T00:00:24"}}
May  9 10:45:52 tasmota-4121 ESP-MQT: Attempting connection...
May  9 10:46:14 tasmota-4121 ESP-MQT: Connect failed to raspberrypi.fritz.box:1883, rc -2. Retry in 10 sec
May  9 10:46:30 tasmota-4121 ESP-RSL: tele/tasmota/STATE = {"Time":"2020-05-09T09:46:30","Uptime":"3T14:32:44","UptimeSec":311564,"Heap":8,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":12,"POWER":"ON","Wifi":{"AP":1,"SSId":"FRITZ!Box 7590 UN","BSSId":"98:9B:CB:79:5C:75","Channel":11,"RSSI":10,"Signal":-95,"LinkCount":6,"Downtime":"0T00:00:24"}}
May  9 10:47:20 tasmota-4121 ESP-RSL: tele/tasmota/STATE = {"Time":"2020-05-09T09:47:20","Uptime":"3T14:33:34","UptimeSec":311614,"Heap":5,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":12,"POWER":"ON","Wifi":{"AP":1,"SSId":"FRITZ!Box 7590 UN","BSSId":"98:9B:CB:79:5C:75","Channel":11,"RSSI":18,"Signal":-91,"LinkCount":6,"Downtime":"0T00:00:24"}}
May  9 10:47:50 tasmota-4121 ESP-RSL: tele/tasmota/SENSOR = {"Time":"2020-05-09T09:47:50","ENERGY":{"TotalStartTime":"2020-03-19T21:45:24","Total":47.364,"Yesterday":0.638,"Today":0.071,"Period":0,"Power":29,"ApparentPower":68,"ReactivePower":61,"Factor":0.43,"Voltage":236,"Current":0.287}}
May  9 10:48:14 tasmota-4121 ESP-WIF: Connected
May  9 10:48:14 tasmota-4121 ESP-HTP: Web server active on tasmota-4121 with IP address 192.168.178.63
May  9 10:48:15 tasmota-4121 ESP-NTP: UTC 2020-05-09T08:48:14, DST 2020-03-29T02:00:00, STD 2020-10-25T03:00:00
May  9 10:48:15 tasmota-4121 ESP-MQT: Attempting connection...
May  9 10:48:15 tasmota-4121 ESP-MQT: Connected
May  9 10:48:15 tasmota-4121 ESP-MQT: tele/tasmota/LWT = Online (retained)
May  9 10:48:15 tasmota-4121 ESP-MQT: cmnd/tasmota/POWER = 
May  9 10:48:15 tasmota-4121 ESP-MQT: Subscribe to cmnd/tasmota/#
May  9 10:48:15 tasmota-4121 ESP-MQT: Subscribe to cmnd/tasmotas/#
May  9 10:48:15 tasmota-4121 ESP-MQT: Subscribe to cmnd/Arbeitszimmer_PC_Jens_fb/#
May  9 10:48:15 tasmota-4121 ESP-MQT: tele/tasmota/INFO1 = {"Module":"Gosund SP1 v23","Version":"8.2.0(tasmota)","FallbackTopic":"cmnd/Arbeitszimmer_PC_Jens_fb/","GroupTopic":"cmnd/tasmotas/"}
May  9 10:48:15 tasmota-4121 ESP-MQT: tele/tasmota/INFO2 = {"WebServerMode":"Admin","Hostname":"tasmota-4121","IPAddress":"192.168.178.63"}
May  9 10:48:15 tasmota-4121 ESP-MQT: tele/tasmota/INFO3 = {"RestartReason":"Power On"}
May  9 10:48:15 tasmota-4121 ESP-MQT: stat/tasmota/RESULT = {"POWER":"ON"}
May  9 10:48:15 tasmota-4121 ESP-MQT: stat/tasmota/POWER = ON
May  9 10:48:16 tasmota-4121 ESP-QPC: Reset
May  9 10:48:17 tasmota-4121 ESP-APP: Boot Count 73
May  9 10:48:18 tasmota-4121 ESP-CFG: Saved to flash at F4, Count 211, Bytes 4096
May  9 10:48:19 tasmota-4121 ESP-MQT: tele/tasmota/STATE = {"Time":"2020-05-09T09:48:19","Uptime":"0T00:00:12","UptimeSec":12,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"FRITZ!Box 7590 UN","BSSId":"F0:B0:14:AF:7B:31","Channel":11,"RSSI":100,"Signal":-50,"LinkCount":1,"Downtime":"0T00:00:06"}}
May  9 10:48:19 tasmota-4121 ESP-MQT: tele/tasmota/SENSOR = {"Time":"2020-05-09T09:48:19","ENERGY":{"TotalStartTime":"2020-03-19T21:45:24","Total":53.178,"Yesterday":0.638,"Today":5.884,"Period":0,"Power":14,"ApparentPower":44,"ReactivePower":41,"Factor":0.31,"Voltage":236,"Current":0.185}}
```
Maybe the saved value in the flash memory is misinterpreted or saved incorrectly?

jensb, 2020-05-10:
Thank you. Note this is a unique problem. I have ten of these devices in my home, all configured identically, just collecting energy usage data, and only this one has this bug, but it is not used in any special way different from the others (that I can see).

arendst, 2020-05-10:
Pls give the latest dev version a try (Wait before it is available at http://thehackbox.org/tasmota/ with above commit number) or compile yourselves

closed 2020-05-10 by ascillato2

jensb, 2020-05-31:
Tried v8.2, v8.3 and v8.3.1. This still happens. Note that the increase in consumption is always 5.24kWh, and it only happens once a day, and it even happens when nobody is at home (so nobody actually uses the device, no current was consumed today).