arendst/Tasmota #8361: The Github wiki is broken

nuxeh, 2020-05-06:
Hi, the Github wiki links persistent in issues on Github and elsewhere are broken. Thanks.

closed 2020-05-06 by close-issue-app[bot]

AlexTransit, 2020-05-06:
wiki here
https://tasmota.github.io/docs/

nuxeh, 2020-05-06:
Thanks @AlexTransit , I had seen that, but this is pretty incomplete, where are for example the list of console commands? This just detracts from the usability of this project, imo.

meingraham, 2020-05-06:
@nuxeh 

Menu bar on the left, click on Commands?

Similar for other parts of the documentation. Also, there are additional major sections linked across the top menu bar.

nuxeh, 2020-05-06:
Thanks @meingraham , my bad.

AlexTransit, 2020-05-07:
@meingraham  
use search. (top page)

meingraham, 2020-05-07:
@AlexTransit 

Huh?  My question mark was not me asking but rather inquiring if that was not what  @nuxeh needed. He got my meaning.