arendst/Tasmota #8362: Relay wired to pin D0 not works

fabry891, 2020-05-06:
### PROBLEM DESCRIPTION
I've installed the lastest tasmota firmware on nodemcu to control 3 relay and one BME280 I2C sensor.
The issue is on relay that is wired to D0 pin (it power a transformer 220V ac to 12V AC), in works very well when the transformer isn't powered, but if is powered, node mcu non respond very well at my command, both via web page also via mqtt command and home assistant. It seems rebooted and lost wifi connectiont for some seconds.
I've try with other relay and also change another pin, but nothings change.
I think that when the transfomer is switch on to off and viceversa, it interferences with nodemcu.
I've search this issue on this forum, but I've not found a solution.

Can anyone helps me?
Thanks.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): NodeMCU
- [x] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [x] Self-compiled
    - [x] IDE / Compiler used: Atom/PlatformIO
- [x] Flashing tools used: Atom/PlatformIO
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
13:36:26 MQT: stat/tasmotaSerra/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
13:36:26 MQT: stat/tasmotaSerra/RESULT = {"Module":{"18":"Generic"}}
13:36:27 MQT: stat/tasmotaSerra/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"6":"I2C SDA"},"GPIO5":{"5":"I2C SCL"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"22":"Relay2"},"GPIO13":{"23":"Relay3"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"21":"Relay1"}}

```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
13:37:08 MQT: stat/tasmotaSerra/RESULT = {"Rule1":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
13:37:08 MQT: stat/tasmotaSerra/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
13:37:08 MQT: stat/tasmotaSerra/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Free":510,"Rules":"`"}

```
- [x] Provide the output of this command: `Status 0`:
```
13:37:41 MQT: stat/tasmotaSerra/STATUS = {"Status":{"Module":18,"FriendlyName":["Tasmota1","Tasmota2","Tasmota3"],"Topic":"tasmotaSerra","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
13:37:41 MQT: stat/tasmotaSerra/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Exception","Uptime":"0T00:27:49","StartupUTC":"2020-05-06T12:09:52","Sleep":50,"CfgHolder":4617,"BootCount":226,"BCResetTime":"2020-05-01T15:51:33","SaveCount":2799,"SaveAddress":"F5000"}}
13:37:41 MQT: stat/tasmotaSerra/STATUS2 = {"StatusFWR":{"Version":"8.2.0.5(tasmota)","BuildDateTime":"2020-05-01T16:50:23","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"376/699"}}
13:37:41 MQT: stat/tasmotaSerra/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["WifiCasaTim",""],"TelePeriod":60,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
13:37:41 MQT: stat/tasmotaSerra/STATUS4 = {"StatusMEM":{"ProgramSize":594,"Free":408,"Heap":23,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"16405E","FlashMode":3,"Features":["00000809","8FDAE797","043683A0","00000CCD","010013C0","C000F981","00000024"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6,9"}}
13:37:41 MQT: stat/tasmotaSerra/STATUS5 = {"StatusNET":{"Hostname":"tasmotaSerra-6945","IPAddress":"192.168.1.208","Gateway":"192.168.1.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.254","Mac":"50:02:91:C3:3B:21","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
13:37:41 MQT: stat/tasmotaSerra/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.111","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_C33B21","MqttUser":"MQTT_user","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
13:37:41 MQT: stat/tasmotaSerra/STATUS7 = {"StatusTIM":{"UTC":"2020-05-06T12:37:41","Local":"2020-05-06T13:37:41","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"05:20","Sunset":"20:13"}}
13:37:41 MQT: stat/tasmotaSerra/STATUS10 = {"StatusSNS":{"Time":"2020-05-06T13:37:41","BME280":{"Temperature":19.5,"Humidity":58.8,"DewPoint":11.2,"Pressure":999.5},"PressureUnit":"hPa","TempUnit":"C"}}
13:37:41 MQT: stat/tasmotaSerra/STATUS11 = {"StatusSTS":{"Time":"2020-05-06T13:37:41","Uptime":"0T00:27:49","UptimeSec":1669,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"ON","POWER2":"OFF","POWER3":"OFF","Wifi":{"AP":1,"SSId":"WifiCasaTim","BSSId":"C4:EA:1D:11:C0:C3","Channel":11,"RSSI":100,"Signal":-48,"LinkCount":1,"Downtime":"0T00:00:03"}}}
13:37:41 MQT: stat/tasmotaSerra/STATUS12 = {"StatusSTK":{"Exception":3,"Reason":"Exception","EPC":["4025889f","00000000","00000000"],"EXCVADDR":"4025c3d5","DEPC":"00000000"}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
Relay wired to pin D0 works well, whe the transformer is power on.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-05-06:
Relais connected to GPIO 16 is not a good idea. GPIO 16 is a special GPIO and has limitations

ascillato2, 2020-05-06:
That GPIO is used by the ESP8266 chip for deepsleep. It is how the hardware works. For more information on GPIO restrictions of the ESP8266 chip, check the Tasmota Docs at https://tasmota.github.io/docs/Peripherals/#gpio-overview

Gpio restrictions are hardware and not controlled by Tasmota Software.

If you need further assistance, please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. 

Thanks.

---------------------------

### Support Information 
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-05-06 by ascillato2