arendst/Tasmota #8374: Sonoff SV module type

SavageAUS, 2020-05-07:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Setting the module type in tasmota GUI does not stick
[image]
Updated to latest dev to see if issue still exists from 8.2 and indeed it does. Device module says sonoff basic when set to sonoff sv

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): _____sonoff sv
- [ ] Tasmota binary firmware version number used: _____8.2 & 8.2.0.6
  - [x ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ x] Flashing tools used: _____OTA
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
``12:03:22 CMD: Backlog Template; Module; GPIO 255
12:03:22 RSL: stat/tasmota_208B42/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
12:03:22 RSL: stat/tasmota_208B42/RESULT = {"Module":{"3":"Sonoff SV"}}
12:03:23 RSL: stat/tasmota_208B42/RESULT = {"GPIO0":{"17":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"21":"Relay1"},"GPIO13":{"56":"Led1i"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [x ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here: 
12:05:02 CMD: Status 0
12:05:02 RSL: stat/tasmota_208B42/STATUS = {"Status":{"Module":3,"FriendlyName":["Tasmota"],"Topic":"tasmota_208B42","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
12:05:02 RSL: stat/tasmota_208B42/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"https://thehackbox.org/tasmota/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:10:07","StartupUTC":"2020-05-07T10:54:55","Sleep":50,"CfgHolder":4617,"BootCount":10,"BCResetTime":"2020-05-07T11:22:50","SaveCount":22,"SaveAddress":"FB000"}}
12:05:02 RSL: stat/tasmota_208B42/STATUS2 = {"StatusFWR":{"Version":"8.2.0.6(88dbc8f-tasmota)","BuildDateTime":"2020-05-07T10:00:17","Boot":31,"Core":"2_7_0","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"325/699"}}
12:05:02 RSL: stat/tasmota_208B42/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["MyRepublic CBD5",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
12:05:02 RSL: stat/tasmota_208B42/STATUS4 = {"StatusMEM":{"ProgramSize":584,"Free":416,"Heap":25,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"146085","FlashMode":3,"Features":["00000809","8FDAE797","043683A1","000000CD","010013C0","C000F981","00000024"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
12:05:02 RSL: stat/tasmota_208B42/STATUS5 = {"StatusNET":{"Hostname":"tasmota_208B42-2882","IPAddress":"192.168.2.130","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.2","Mac":"3C:71:BF:20:8B:42","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
12:05:02 RSL: stat/tasmota_208B42/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_208B42","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
12:05:02 RSL: stat/tasmota_208B42/STATUS7 = {"StatusTIM":{"UTC":"2020-05-07T11:05:02","Local":"2020-05-07T12:05:02","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"05:19","Sunset":"20:14"}}
12:05:02 RSL: stat/tasmota_208B42/STATUS10 = {"StatusSNS":{"Time":"2020-05-07T12:05:02"}}
12:05:02 RSL: stat/tasmota_208B42/STATUS11 = {"StatusSTS":{"Time":"2020-05-07T12:05:02","Uptime":"0T00:10:07","UptimeSec":607,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"MyRepublic CBD5","BSSId":"AC:84:C6:42:12:1F","Channel":9,"RSSI":50,"Signal":-75,"LinkCount":1,"Downtime":"0T00:00:06"}}}
12:05:02 APP: Serial logging disabled

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2020-05-07:
As designed:

[image]

What is means is that in the default configuration Module type ``Sonoff Basic`` is stored. It just tries to tell you the default module, NOT the module you just selected and configured. Considering the amount of configurable GPIOs you can see the difference between The SV and Basic.

effelle, 2020-05-07:
Closing this issue as it has been answered.

Support Information
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-05-07 by effelle