arendst/Tasmota #8378: TSL2651 with NODEMCU problems

XxJooO, 2020-05-07:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

Hi all,

I used 8.1 tasmota-DE.bin, not self compiled and I get proper measurement by connecting GPIO4 to I2C SDA (6) and GPIO5 to I2C SCL (5) to a NODEMCU. By now it shows 10.660lx

Now I updated to 8.2 tasmota-sensors.bin and the TSL2651 is still connected like mentioned above. By now I restarted a lot and powered off and on again. But now I still get no proper measurements.

I´m still using Module type Generic (18). Is there a workaround for this problem?

Thank you for your great work!!!

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x] Device used (e.g., Sonoff Basic): ___NODEMCU__
- [ x] Tasmota binary firmware version number used: _8.2 sensors.bin____
  - [ x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ x] Flashing tools used: __OTA___
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
``17:03:15 MQT: stat/Sonoff-Luxmeter-topic/RESULT = {"NAME":"Jo","GPIO":[255,255,255,255,5,6,255,255,255,255,255,255,255],"FLAG":0,"BASE":18}
17:03:15 MQT: stat/Sonoff-Luxmeter-topic/RESULT = {"Module":{"18":"Generic"}}
17:03:16 MQT: stat/Sonoff-Luxmeter-topic/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"6":"I2C SDA"},"GPIO5":{"5":"I2C SCL"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}`
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
17:05:19 MQT: stat/Sonoff-Luxmeter-topic/STATUS = {"Status":{"Module":18,"FriendlyName":["Sonoff-Luxmeter"],"Topic":"Sonoff-Luxmeter-topic","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":0,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
17:05:19 MQT: stat/Sonoff-Luxmeter-topic/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff-DE.bin","RestartReason":"Software/System restart","Uptime":"0T00:16:30","StartupUTC":"2020-05-07T14:48:49","Sleep":50,"CfgHolder":4617,"BootCount":137,"BCResetTime":"2020-05-07T16:17:09","SaveCount":1527,"SaveAddress":"F4000"}}
17:05:19 MQT: stat/Sonoff-Luxmeter-topic/STATUS2 = {"StatusFWR":{"Version":"8.2.0(sensors)","BuildDateTime":"2020-03-20T14:48:25","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"363/699"}}
17:05:19 MQT: stat/Sonoff-Luxmeter-topic/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["xXWLANXx","xXWLANXx"],"TelePeriod":10,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005AFF000000000000","00000000","00000000"]}}
17:05:19 MQT: stat/Sonoff-Luxmeter-topic/STATUS4 = {"StatusMEM":{"ProgramSize":597,"Free":404,"Heap":21,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164020","FlashMode":3,"Features":["00000809","8FDAA387","0415A3A4","B7FFBFCD","01DA9BC4","24367CC7","00000002"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,14,16,17,20,21,24,29,34","Sensors":"1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,51,52,55,56,58,59,62,64"}}
17:05:19 MQT: stat/Sonoff-Luxmeter-topic/STATUS5 = {"StatusNET":{"Hostname":"Sonoff-Luxmeter","IPAddress":"192.168.178.28","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.178.10","Mac":"84:F3:EB:31:26:27","Webserver":2,"WifiConfig":0,"WifiPower":17.0}}
17:05:19 MQT: stat/Sonoff-Luxmeter-topic/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.178.6","MqttPort":1883,"MqttClientMask":"Sonoff-Luxmeter","MqttClient":"Sonoff-Luxmeter","MqttUser":"Sonoff","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
17:05:19 MQT: stat/Sonoff-Luxmeter-topic/STATUS7 = {"StatusTIM":{"UTC":"2020-05-07T15:05:19","Local":"2020-05-07T17:05:19","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":99,"Sunrise":"06:19","Sunset":"21:14"}}
17:05:19 MQT: stat/Sonoff-Luxmeter-topic/STATUS10 = {"StatusSNS":{"Time":"2020-05-07T17:05:19","TSL2561":{"Illuminance":99999.999}}}
17:05:19 MQT: stat/Sonoff-Luxmeter-topic/STATUS11 = {"StatusSTS":{"Time":"2020-05-07T17:05:19","Uptime":"0T00:16:30","UptimeSec":990,"Heap":21,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"xXWLANXx","BSSId":"C8:0E:14:0B:B2:1D","Channel":6,"RSSI":44,"Signal":-78,"LinkCount":1,"Downtime":"0T00:00:06"}}}
17:05:22 MQT: tele/Sonoff-Luxmeter-topic/STATE = {"Time":"2020-05-07T17:05:22","Uptime":"0T00:16:33","UptimeSec":993,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":26,"MqttCount":1,"Wifi":{"AP":1,"SSId":"xXWLANXx","BSSId":"C8:0E:14:0B:B2:1D","Channel":6,"RSSI":42,"Signal":-79,"LinkCount":1,"Downtime":"0T00:00:06"}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

![image](https://user-images.githubusercontent.com/47712364/81310996-1d02e300-9085-11ea-8b76-e4f8c81611c2.png)

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

joba-1, 2020-05-07:
nearly missed this one... It is TSL2561, not TSL2651 :)
I'm the author of the TSL2561 lib and I'll support, if I can.

"Illuminance":99999.999 from your log means, sensor got saturated, i.e.
full raw light level or ir raw light level are out of spec range.
This usually only happens, if you do not use auto range (the sensor has
different sensitivity modes) or expose to direct sunlight or very, very
close to bright leds.
Maybe it could happen if the sensor is defect, but I didn't see this so far.
Have to check if tasmota still uses auto range as I initially wrote it - a
lot has changed since...
There was no change in the library for a very long time - surely not
between tasmota 8.1 and 8.2

schrej-zz, 2020-05-08:
In 8.2, a lot of sensors are now only in tasmota-sensors.bin. Did you check  if you sensor is involved?

joba-1, 2020-05-08:
From the log: Version":"8.2.0(sensors)
Also, the value 99999.999 comes from tsl2561 lib code, so it is included and used.

Btw. I'd recommend displaying "out of range" in this case (I just can't do that in the lib, that just returns integers)

rt400, 2020-05-13:
try to run in console 
`i2cdriver13 0`
make restart and see if it fix your problem

XxJooO, 2020-05-14:
Thank you, I will give it a try and reply

XxJooO, 2020-05-16:
It does not work at all. The values jump between the 999999.999, real values like 10240 and then suddenly it goes to 0 and stays there. So I will change the sensor and the NodeMCU and try it again. Maybe a hardware problem...

XxJooO, 2020-05-16:
Hey guys, by now it seems to work - I think it was a bad connection cable. The sensor is still giving values since the last start 68 minutes before. And they still seem to be valid. Thank you for your help.

Jo

closed 2020-05-16 by XxJooO

joba-1, 2020-05-16:
Very good. Thanks for the update.