arendst/Tasmota #8385: Support Tokens in Friendly Name

seidler2547, 2020-05-07:
**Have you looked for this feature in other issues and in the docs?**  
Yes.

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
When used with Home Assistent and auto-discovery, the friendly name will be used as entity name in Home Assistant. Since the Friendly Name is static, this creates a lot of entities with similar names.

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._
If Friendly Name supported Tokens, it could at least be made unique for each new device and thus not create device name clashes in Home Assistant.

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._
Manual configuration in HA is of course an option, but then the autodiscovery feature can also be removed as it becomes useless.

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-05-07:
Friendly name is not static. You can change like you want

seidler2547, 2020-05-07:
But only after Tasmota is installed. Not in the build: https://github.com/arendst/Tasmota/issues/8382

Jason2866, 2020-05-07:
Yes, and there are a many options to configure Tasmota settings when it is running.

seidler2547, 2020-05-07:
Please read the description of the feature request. The name sent in the discovery for Home Assistant will always be "Tasmota". Even if I change it later, it won't update the device ID in HA.

seidler2547, 2020-05-07:
> Friendly name main usage is for controlling via Alexa
And apparently device ID and name in HA when autodiscovery is enabled. Of course that would be another option to send for example the MQTT topic as device ID to HA.

Jason2866, 2020-05-07:
This was discussed internal already. Tasmota would need extra space to support different names
just for HA Autodiscovery. Theo decided not to do.

effelle, 2020-05-07:
And I second what @Jason2866 said: no other changes will be made over the last one, where `Friendlyname1` was removed as device **ROOT** name on latest devs and soon will be the default on the new master 8.3. The name of the module/template is used instead and was changed after the request of a Home Assistant developer. 
On Home Assistant integration page you can easily change the root name of the device, Home Assistant will ask you if you want to change all the name of the related sensors at once.

seidler2547, 2020-05-07:
So in 8.3 the problem will be alleviated?

effelle, 2020-05-07:
`Friendlyname1` is the name of the first relay and was not correct using it on the first place or at least not anymore. 
Is still the name you will see on the webui, because when Tasmota was created was when the market was hitted by the Sonoff Basic, with just one relay. At the time was not a bad choice. 
As I said on latest devs the module name is equivalent of the root name for HA. 
If you create a template adhoc you will have also a meaningful name for each of your devices.
If you don't like mess with the modules/template use the renaming function under HA.
The next stable release, scheduled for next Thursday (May 14) if you don't like to play with dev bin.
Closing this issue as it has been answered, thanks for your feedback.

Support Information
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-05-07 by effelle