arendst/Tasmota #8386: New switchmode 13 & 14 as 'tweaked' version of switchmode 11 & 12 (Virtual one-button Zigbee switch)

htvekov, 2020-05-07:
**Have you looked for this feature in other issues and in the docs?**  

Yep. Can't find anyone requesting same switchmode features.

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._

I'm using new switchmode 11 setting to apply dimming capabilities to my 'dumb' wall switches made 'smart' with Tasmotized Shelly Ones behind switch in wall socket. Automation part, for dimming attached smart light through HA MQTT sensor, is Xavi's ControllerX appdeamon app and Tasmota rules. Some 'tweaks' and 'tricks' in rules are needed to get this working with switchmode 11.
Switchmode 11 was clearly not developed for this purpose, but specifically for dimming a RGBW light strip. For my use I've two issues preventing me from using this switchmode directly as a virtual version of a physical Zigbee one-button switch (Eg. Xiaomi Aqara WXKG11LM).

1. HOLD (inc-dec) function sends continuous commands until RELEASE (clear) is send
2. Release (clear) is not send immediately after button is released but hooked up to same delay as HOLD function defined by setoption32

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._

1. HOLD (inc-dec) only send once (after delay set by setoption32)
2. RELEASE (clear) send immediately when button is released (no delay at all)
3. Remaining two funtion calls (toggle and inv) as they are today with Switchmode 11

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

I'm very happy that Switchmode11 was done and merged, so I could proceed with my 'Tasmotizing' wall swithches project 😁 Not only adding toggle, on and off function to connected smart light, but improving with dimming capabilities as well. Also improved the WAF quite a lot, not having extra Zigbee swithches next to the 'dumb' wall switch. And be able to implement easy direct relay control if MQTT server is not available, also helps on the WAF 😉

But to use Switchmode11 for this use case, involves somewhat more than basic user knowledge. This dimming switchmode feature would be much more 'accessible' for other users and easier to implement with HA apps like ControllerX or own automations. To 'tweak' Swithmode 11 to emulate a real physical one-button Zigbee switch I use following Tasmota Rules for a working solution. But I still like to see the 100 millisecond unneeded delay disappear 😉

```
Rule1 on switch1#state=2 do backlog publish tasmota_topic toggle;publish tasmota_topic endon on switch1#state=6 do backlog publish tasmota_topic clear;setoption32 10;rule3 1;publish tasmota_topic endon    

Rule2 on Mqtt#Disconnected do backlog rule1 0 endon on Mqtt#Connected do backlog rule1 1 endon

Rule3 5 on switch1#state=4 do backlog publish tasmota_topic inc-dec;setoption32 1;publish tasmota_topic;rule3 0 endon  
```    

RULE1 sends the TOGGLE command and afterwards immediately clears the sensor state with an empty payload (just like a physical button does). It also handles the CLEAR command when button is released, resets HOLD delay to 1 sec., enable RULE3 and finally clears payload

RULE2 handles the 'fall back' to direct switch control if MQTT server is unavailable. Nice WAF feature, if HA/MQTT server is down 😉 Then lights can still be toggled on/off using the wall switch.

RULE3 sends the HOLD command, set wait delay to lowest possible minimum 0,1 sec. (for next release button event), disables itselves (to ensure only one HOLD command is send) and finally clears payload.
RULE3 is set as 'run once' to prevent some extra HOLD commands gets send before rule is disabled.

**Additional context**  
_Add any other context or screenshots about the feature request here._

Link to documentation for Tasmota switchmode11 setup with ControllerX: https://xaviml.github.io/controllerx/examples/tasmota-switchmode11

Here you'll find some more details about my use case with Tasmota switchmode 11

**(Please, remember to close the issue when the problem has been addressed)**

effelle, 2020-05-07:
I must admit I never understood the use of SwitchMode 11 and 12 but is also true I don't use any kind of dimmer or RGB light .
As for my understanding rules are  pretty necessary to have it working as expected have a look to PR #7603.
A note about your rules:
- On Rule 2  backlog is not necessary since you are adding just one event;
- On rule 3 you have `SetOption32` set to 1 and this is a very short time, you could accidentally trigger the factory reset procedure.

Last but not least, I'm pinging @RemiDing since is the one who added those special SwitchModes probably is more suited to help you in finding a better solution.

RemiDing, 2020-05-08:
Hi, 
thanks for the ping @effelle
I will look into this issue more closely over the weekend.
Maybe I can add code to switchmode 11 to help your project @htvekov
For the soulution you recommended:
1. HOLD (inc-dec) only send once (after delay set by setoption32)
Switchmode 11 needs repeatet HOLD (inc_dec) messages for a dimmer to increment or decrement the light. To send it only once is not possible in switchmode 11. Maybe its possible to ignore the repeated messages?
2. RELEASE (clear) send immediately when button is released (no delay at all)
Switchmode 11 could send a DIFFERENT (new) command immediately when the button is released. Would that help?

htvekov, 2020-05-08:
Hi' @effelle & @RemiDing 

Thank you both for your quick replies ! 🚀😊

Perhaps I should have included the needed Tasmota 'setup' commands I use as well in the request. And also emphasize that I HAVE achieved a fully working solution.

It just needs so many tweaks and special rules (difficult for non experienced users), that this really nice switchmode probably would NOT be used by many users.  As it simply is complicated to implement a working automation for this. That's the reason for the 'improvement issue' to get this dimming switchmode spread out to more users in the community. I believe many actually are not aware of the real potential in this switchmode to do exactly what I've done with 'dumb' wall switches. Easy, reliable and really cheap 'dumb' wall switch conversion at the cost of only a Shelly One or similar switch 😎 What in the past has kept me from doing this, was the fact that i KNEW automations would be a real pain. Now with Tasmota switchmode11 and ControllerX it's actually an easy setup - but still requires an overall experienced user.

Here are the commands I use in conjuction to the three rules to get everything going:

**Powerretain1 1               Retain Tasmota power settings i HA in event of power outage
Setoption1 1                   Allow only single, double and hold press button actions
Setoption32 10		Delay for HOLD button (in 0,1 sec.). I use 10 = 1 sec.
Setoption34 50               Minimize delay between backlog commands to 50 ms. (Default 200 ms.)
Switchtopic 0                  Disable switchtopic. Needed when using rules in Tasmota.
Switchmode1 11             Set Tasmota to switchmode11**

@effelle 
- I actually did check the original PR to get the needed information for doing my implementation 😉
- Setoption1 1 should hopefully secure aginst factory resets with the very low delay in setoption32
- Your're absolute right about the not needed backlog command 😊 Never really noticed that it actually only contained one event.

@RemiDing 
- I do, via rules, ignore the repeated inc-dec messages. It's just complicated to filter out all, but the first message. I both have to disable rule3 handling the event AND set the rule as 'run once'. Only disabling the rule, will still let some 2-3 messages through (probably already in que ?), before disabling the rule 'kicks in'. With the 'run once' option, only one event is send. I know that you in switchmode 11 rely on the repeated inc-dec messages. That's why I suggested a 'tweaked' new alternative switchmode version with slightly different behaviour 😊

- A different message to be send immediately (no delay at all) when button is released would be great !! 😃 That's the only thing I can't work my way out of in the rules. I simply have to wait the minimum extra 100 ms. (setoption32 1. Lowest possible value), before automation realise that button actually has been released.

RemiDing, 2020-05-09:
@htvekov  please try 
https://github.com/RemiDing/Tasmota/tree/development/build_output/firmware/tasmota.bin
in switchmode 11 and 12 it will send immediately a 0 when the button is released.
So you dont need to change setoption32 to 1

htvekov, 2020-05-10:
Hi' @RemiDing 

Sorry about the late reply...

Great ! 😎

Have updated to your dev version and tested - unfortunately a 'no go'  👎 ?
I assume by **'send immediately a 7'** you mean switch1#state=7 should be used as trigger in rules or... ???

RemiDing, 2020-05-11:
Yes. I dont have a zigbee switch here but I think this rules should work

RULE1 on switch1#state=2 do backlog publish tasmota_topic toggle;publish tasmota_topic endon
on switch1#state=7 do backlog publish tasmota_topic clear; rule3 1;publish tasmota_topic endon

RULE2 on Mqtt#Disconnected do backlog rule1 0 endon on Mqtt#Connected do backlog rule1 1 endon

RULE3 5 on switch1#state=4 do backlog publish tasmota_topic inc-dec; publish tasmota_topic; rule3 0 endon

can you be more specific why this is a `no go` ?

htvekov, 2020-05-11:
Hi' @RemiDing 

Of cource 👍

No events are simply fired upon release of button using switch1#state=7 instead of switch1#state=6.

`Toggle` and `inc-dec` fires as expected.
Switch1#state=6 also still works as usual.

Ciao !

htvekov, 2020-05-11:
Here's the log (using weblog 4 level) for 'standard' clear using switch1#state=6.
I'm holding button and releasing it again:

```
22:21:15 RUL: SWITCH1#STATE=4 performs "backlog publish kontor_loftlys_cstm_mqtt/cmnd inc-dec;setoption32 1;publish kontor_loftlys_cstm_mqtt/cmnd;rule3 0"
22:21:15 SRC: Rule
22:21:15 CMD: Group 0, Index 1, Command "BACKLOG", Data "publish kontor_loftlys_cstm_mqtt/cmnd inc-dec;setoption32 1;publish kontor_loftlys_cstm_mqtt/cmnd;rule3 0"
22:21:15 MQT: kontor_loftlys/stat/SWITCH1T = {"TRIG":"ON"}
22:21:15 SRC: Backlog
22:21:15 CMD: Group 0, Index 1, Command "PUBLISH", Data "kontor_loftlys_cstm_mqtt/cmnd inc-dec"
22:21:15 MQT: kontor_loftlys_cstm_mqtt/cmnd = inc-dec
22:21:15 SRC: Backlog
22:21:15 CMD: Group 0, Index 32, Command "SETOPTION", Data "1"
22:21:15 MQT: kontor_loftlys/stat/RESULT = {"SetOption32":1}
22:21:15 SRC: Backlog
22:21:15 CMD: Group 0, Index 1, Command "PUBLISH", Data "kontor_loftlys_cstm_mqtt/cmnd"
22:21:15 MQT: kontor_loftlys_cstm_mqtt/cmnd = 
22:21:16 MQT: kontor_loftlys/stat/SWITCH1T = {"TRIG":"ON"}
22:21:16 MQT: kontor_loftlys/stat/SWITCH1T = {"TRIG":"ON"}
22:21:16 SRC: Backlog
22:21:16 CMD: Group 0, Index 3, Command "RULE", Data "0"
22:21:16 MQT: kontor_loftlys/stat/RESULT = {"Rule3":"OFF","Once":"ON","StopOnError":"OFF","Free":370,"Rules":"on switch1#state=4 do backlog publish kontor_loftlys_cstm_mqtt/cmnd inc-dec;setoption32 1;publish kontor_loftlys_cstm_mqtt/cmnd;rule3 0 endon"}
22:21:16 CFG: Saved to flash at FB, Count 641, Bytes 4096
22:21:17 MQT: kontor_loftlys/stat/SWITCH1 = {"STATE":"OFF"}
22:21:17 RUL: SWITCH1#STATE=6 performs "backlog publish kontor_loftlys_cstm_mqtt/cmnd clear;setoption32 10;rule3 1;publish kontor_loftlys_cstm_mqtt/cmnd"
22:21:17 SRC: Rule
22:21:17 CMD: Group 0, Index 1, Command "BACKLOG", Data "publish kontor_loftlys_cstm_mqtt/cmnd clear;setoption32 10;rule3 1;publish kontor_loftlys_cstm_mqtt/cmnd"
22:21:17 MQT: kontor_loftlys/stat/SWITCH1T = {"TRIG":"ON"}
22:21:17 SRC: Backlog
22:21:17 CMD: Group 0, Index 1, Command "PUBLISH", Data "kontor_loftlys_cstm_mqtt/cmnd clear"
22:21:17 MQT: kontor_loftlys_cstm_mqtt/cmnd = clear
22:21:17 SRC: Backlog
22:21:17 CMD: Group 0, Index 32, Command "SETOPTION", Data "10"
22:21:17 MQT: kontor_loftlys/stat/RESULT = {"SetOption32":10}
22:21:17 SRC: Backlog
22:21:17 CMD: Group 0, Index 3, Command "RULE", Data "1"
22:21:17 MQT: kontor_loftlys/stat/RESULT = {"Rule3":"ON","Once":"ON","StopOnError":"OFF","Free":370,"Rules":"on switch1#state=4 do backlog publish kontor_loftlys_cstm_mqtt/cmnd inc-dec;setoption32 1;publish kontor_loftlys_cstm_mqtt/cmnd;rule3 0 endon"}
22:21:17 SRC: Backlog
22:21:17 CMD: Group 0, Index 1, Command "PUBLISH", Data "kontor_loftlys_cstm_mqtt/cmnd"
22:21:17 MQT: kontor_loftlys_cstm_mqtt/cmnd =

```

And here's the log for same procedure using switch1#state=7. Haven't removed the setoption32 commands yet:

```
22:29:18 RUL: SWITCH1#STATE=4 performs "backlog publish kontor_loftlys_cstm_mqtt/cmnd inc-dec;setoption32 1;publish kontor_loftlys_cstm_mqtt/cmnd;rule3 0"
22:29:18 SRC: Rule
22:29:18 CMD: Group 0, Index 1, Command "BACKLOG", Data "publish kontor_loftlys_cstm_mqtt/cmnd inc-dec;setoption32 1;publish kontor_loftlys_cstm_mqtt/cmnd;rule3 0"
22:29:19 MQT: kontor_loftlys/stat/SWITCH1T = {"TRIG":"ON"}
22:29:19 SRC: Backlog
22:29:19 CMD: Group 0, Index 1, Command "PUBLISH", Data "kontor_loftlys_cstm_mqtt/cmnd inc-dec"
22:29:19 MQT: kontor_loftlys_cstm_mqtt/cmnd = inc-dec
22:29:19 SRC: Backlog
22:29:19 CMD: Group 0, Index 32, Command "SETOPTION", Data "1"
22:29:19 MQT: kontor_loftlys/stat/RESULT = {"SetOption32":1}
22:29:19 SRC: Backlog
22:29:19 CMD: Group 0, Index 1, Command "PUBLISH", Data "kontor_loftlys_cstm_mqtt/cmnd"
22:29:19 MQT: kontor_loftlys_cstm_mqtt/cmnd = 
22:29:19 MQT: kontor_loftlys/stat/SWITCH1T = {"TRIG":"ON"}
22:29:19 MQT: kontor_loftlys/stat/SWITCH1T = {"TRIG":"ON"}
22:29:19 SRC: Backlog
22:29:19 CMD: Group 0, Index 3, Command "RULE", Data "0"
22:29:19 MQT: kontor_loftlys/stat/RESULT = {"Rule3":"OFF","Once":"ON","StopOnError":"OFF","Free":370,"Rules":"on switch1#state=4 do backlog publish kontor_loftlys_cstm_mqtt/cmnd inc-dec;setoption32 1;publish kontor_loftlys_cstm_mqtt/cmnd;rule3 0 endon"}
22:29:19 CFG: Saved to flash at F8, Count 644, Bytes 4096
22:29:20 MQT: kontor_loftlys/stat/SWITCH1 = {"STATE":"OFF"}
22:29:21 MQT: kontor_loftlys/stat/SWITCH1T = {"TRIG":"ON"}
```

htvekov, 2020-05-12:
Hi' @RemiDing 

I've downloaded your dev firmware again and updated once more.
All works perfectly now with immediate clear command using swithc1#state=7 😎😁

Really don't know how I managed to completely f*ck up the first update ?? 🙄
I noticed that my version changed from 8.2.0 to 8.2.0.6. and thought everything must be fine then. But somehow I must have downloaded 'standard' dev branch v8.2.0.6 and not yours... 🤔

I'm really sorry for have wasted your time testing and debugging 'non existing' errors 😣😫😖🥺

And I'm REALLY pleased that you did make this enhancement for me 😀
Thanks you so much, RemiDing !! 🤗

Any idea on usual timeframe for this update/PR to appear in next Tasmota release (dev/stable) ?

RemiDing, 2020-05-13:
No time wasted. Debugging is always good. When you look for non existing errors you maybe find real ones :-)
I do some more tests and then start a pull request. Please feel free to do additional tests also.
I have no idea about the timeframe. Normally it goes within a week.
Maybe you can later update https://xaviml.github.io/controllerx/examples/tasmota-switchmode11

htvekov, 2020-05-13:
As I'm in the process of installing Shelly's behind all wall switches in the house and changing them from regular swithes to push buttons (they're prepared to be fitted with a spring), I'm already doing 'live' tests 24/7 of this solution 😉

For now I'm done with 7 switches where 3 of them are running with the switchmode 11 controlling smart lights. Both as single bulbs and light groups and both on z2m and Hue bridge HA integration.

All smooth sailing so far! 😎

Lag is also very low (some hundreds of milliseconds I guess on average for action) and fully acceptable. I'm currently running HA as hassio install on a rpi3+ with ssd disc, a good handfull of apps, HACS custom add-ons and some 800 lines of automation. I believe it's time for me to leave the rpi platform shortly and go NUC! 🚀🚀 I expect HA to be even more responsive and some further lag improvement shifting to a more powerfull hardware platform.

And yes, reason for asking about the expected timeframe, was indeed to update documentation and examples for the ControllerX solution with Tasmota switchmode11 😊

Once again, thank you, @RemiDing 👍🙋‍♂️

ascillato2, 2020-05-16:
Thanks for working on this.

Closing this issue as the PR #8450 has been done for reviewing.

closed 2020-05-16 by ascillato2

htvekov, 2020-06-20:
Documentation is now updated with Xavi's release of ControllerX v2.8.0

https://xaviml.github.io/controllerx/examples/tasmota-switchmode11