arendst/Tasmota #8388: Change the PWM minimum frequency limit from 100 to lower

dan-cristian, 2020-05-07:
**Have you looked for this feature in other issues and in the docs?**  
Y
**Is your feature request related to a problem? Please describe.**  
_I need to control a heater via PWM and a SSR, the PWM frequency for this needs to be around 50 Hz. Currently Tasmota only allows 100Hz as minimum frequency _es

**Describe the solution you'd like**  
_Reduce the frequency limit to 1. Why is it set to 100 anyways?_

**Describe alternatives you've considered**  
_I made this work with micropython but is not an easy solution_

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

effelle, 2020-05-07:
Have a look on PR #8255 to know why. 
If I remember correctly there is a long story behind this PR, better ask to @s-hadinger for more details.

s-hadinger, 2020-05-08:
I don't know why the minimum was set to 100Hz and I don't see a reason why we couldn't lower this minimum.

Let me check with the Arduino code what is the absolute minimum frequency.

s-hadinger, 2020-05-08:
I tried going to very low frequencies, below 20Hz it's getting chaotic. I suppose some timers overflow. I propose to make a quick patch in Tasmota to lower the minimum to 20Hz for now.

Please don't use low frequencies for LEDs, it's flashing like hell :)

arendst, 2020-05-08:
That's why I choose 100Hz ;-)

closed 2020-05-08 by arendst

dan-cristian, 2020-05-08:
40 as minimum works well for me, many thanks!

PR #8394 (2020-05-08): Lower minimum PWMFrequency to 40Hz
## Description:

Allow to lower PWMFrequency to 40Hz instead of 100Hz. Don't use low frequencies for LEDs, it will blink like hell.

**Related issue (if applicable):** fixes #8388

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR.
  - [x] The code change is tested and works on core ESP8266 V.2.7.1
  - [ ] The code change is tested and works on core ESP32 V.1.12.0
  - [x] I accept the [CLA](https://github.com/arendst/Tasmota/blob/development/CONTR [... cut]