arendst/Tasmota #8400: Have a sane way of checking current time of the module from web interface

martinpaljak, 2020-05-09:
**Have you looked for this feature in other issues and in the docs?**  
I think so.

**Is your feature request related to a problem? Please describe.**  
It is a pain to get the time of the module right, or find out what the right time is. 

**Describe the solution you'd like**  
Show module time (and timezone) in the info screen. And if possible, make timezone configuration a bit more sensible.

**Describe alternatives you've considered**  
Have the default at UTC at least

**Additional context**  

**(Please, remember to close the issue when the problem has been addressed)**

meingraham, 2020-05-09:
@martinpaljak 

> Show module time (and timezone) in the info screen.

As in when you click the `Information` button on the web UI?

It's not on the main screen because that page does not refresh. If the time was output when the page was initially rendered, I'm sure there would be complaints that the displayed time is incorrect. This applies to displaying it when the `Information` page loads as well. Or most of the Tasmota UI.

However, if you click on the `Console` web UI button, there you will see the logged messages and each message is preceded by the device's timestamp. Would this meet you needs?

martinpaljak, 2020-05-09:
Yes I mean the information button, `/in` url. It does show the build date, which is much less useful, IMHO.

My needs are met, but the thing I've been annoyed about most is the time (and somewhat flaky network via web, but mqtt is rocksolid).

The default that seems to be +1 timezone is also weird, if anything set to UTC.

meingraham, 2020-05-10:
Yeah, the pages aren't going to add time of day that remains static once the page loads... because inevitably there will be the report (complaint?) that the time doesn't update.

+1 time zone... the project owner doesn't live in the UK... so it's logical that he built the software with the default settings for himself :wink: One of the first things a Tasmota user would typically do is to make changes to settings to "localize" the device - i.e., Time Zone, Lat/Lon. After flashing the pre-compiled software, setting these is quite simple. Everything is well documented in the Commands article. And, if you are compiling your own binary, then have a `user_config_override.h` that you can use for all your devices with the localization "hard coded".

I will go ahead and close this issue since it is answered... perhaps not to your satisfaction, but answered nevertheless.

closed 2020-05-10 by meingraham

martinpaljak, 2020-05-10:
I dont live in UK either, that’s why a neutral default like UTC would make sense, something log analyzers or sensor data collectors use anyway.