arendst/Tasmota #8402: HASS discover MQTT Lights in Template schema instead of Default schema - to support transitions

maichai, 2020-05-09:
**Have you looked for this feature in other issues and in the docs?**  

yes

**Is your feature request related to a problem? Please describe.**  

Home Assistant discovery of lights uses default schema, which does not support transitions.

**Describe the solution you'd like**  

HASS Discover lights in Template schema.

**Describe alternatives you've considered**  

I tried also JSON chema but that does not work with the command structure of Tasmota.

**Additional context**  

If you would like to implement this I can provide MQTT Light Template schema descriptions for RGBW and RBW+CCT. Wou don't need to figure it out on your own. It took me all evening to get those to work...

**(Please, remember to close the issue when the problem has been addressed)**

effelle, 2020-05-10:
A practical example?

maichai, 2020-05-11:
I dont understand the question. What so far was missing: it is not a template schema. In template schema you can do this: 

`
command_on_template: >
....
      {%- if transition is defined -%}
      fade 1; speed {{ transition |round |int * 2 }}; 
      {%- endif -%}
....
`

effelle, 2020-05-11:
The question is simple: If you want me to add something on the discovery you need to show the advantage of the edit. 
Then again, explain with a practical example why should I use a template over the normal configuration? What will I gain? What will I lose?

maichai, 2020-05-11:
See above: Home Assistant discovery of lights uses default schema, which does not support transitions. You lose nothing. You gain transitions. If that is not interesting, no problem, I can use it manually.

emontnemery, 2020-05-11:
I don't think Home Assistant MQTT light template schema is a good match for Tasmota, it requires all commands to be sent on a single topics and all state to be received on a single topic.
OTOH, transition support could be added to Home Assistant MQTT light basic schema.

maichai, 2020-05-11:
You have a point regarding the command, but note that you do NOT need to send all parameters, just the ones that have changed. See my example above. AFAIK the Default schema does NOT support transitions [(see)](https://www.home-assistant.io/integrations/light.mqtt/). If you can get it to work with Default schema that would be great. I just went down the Template chema road because the docs say that the Default schema does not support them. I tried the JSON schema, but it is not compatible with the JSON dictionary provided by Tasmota. Maybe I don't know well enough how to use the JSON schema, maybe one can set the dictionary to conform with Tasmota. That would also work then.

effelle, 2020-05-11:
Ok @emontnemery, I'll wait an eventual edit made by you under Home Assistant and then will see if I can merge in Tasmota. For now the Relay/Light void stays as it is.

closed 2020-05-11 by effelle