arendst/Tasmota #8404: Sonoff S31, Domoticz: voltage and current only reported on restart

gknops, 2020-05-10:
### PROBLEM DESCRIPTION
It seems the voltage and current info are only provided on restart. The console log below illustrates it (note how idx 242, 243 and 245 are only reported once, while 241 is reported every 10 seconds (updated interval):

00:00:00 CFG: Loaded from flash at F6, Count 62
00:00:00 Project tasmota Tasmota Version 8.2.0(tasmota)-STAGE
00:00:00 SER: Set to 8E1 4800 bit/s
00:00:00 SNS: Hardware Serial
00:00:00 WIF: Connecting to AP1 Marillia_IoT in mode 11N as tasmota_00543D-5181...
00:00:05 WIF: Connected
00:00:05 HTP: Web server active on tasmota_00543D-5181 with IP address 192.168.10.137
07:18:38 MQT: Attempting connection...
07:18:38 MQT: Connected
07:18:38 MQT: tele/tasmota_00543D/LWT = Online (retained)
07:18:38 MQT: cmnd/tasmota_00543D/POWER = 
07:18:38 MQT: tele/tasmota_00543D/INFO1 = {"Module":"Sonoff S31","Version":"8.2.0(tasmota)","FallbackTopic":"cmnd/DVES_00543D_fb/","GroupTopic":"cmnd/tasmotas/"}
07:18:38 MQT: tele/tasmota_00543D/INFO2 = {"WebServerMode":"Admin","Hostname":"tasmota_00543D-5181","IPAddress":"192.168.10.137"}
07:18:38 MQT: tele/tasmota_00543D/INFO3 = {"RestartReason":"Software/System restart"}
07:18:38 MQT: stat/tasmota_00543D/RESULT = {"POWER":"ON"}
07:18:38 MQT: stat/tasmota_00543D/POWER = ON
07:18:39 MQT: domoticz/in = {"idx":241,"nvalue":1,"svalue":"","Battery":100,"RSSI":10}
07:18:42 MQT: tele/tasmota_00543D/STATE = {"Time":"2020-05-10T07:18:42","Uptime":"0T00:00:12","UptimeSec":12,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"Marillia_IoT","BSSId":"E2:63:DA:AE:48:6E","Channel":11,"RSSI":100,"Signal":-46,"LinkCount":1,"Downtime":"0T00:00:06"}}
07:18:42 MQT: domoticz/in = {"idx":245,"nvalue":0,"svalue":"0;16.0","Battery":100,"RSSI":10}
07:18:42 MQT: domoticz/in = {"idx":242,"nvalue":0,"svalue":"120","Battery":100,"RSSI":10}
07:18:42 MQT: domoticz/in = {"idx":243,"nvalue":0,"svalue":"0.000","Battery":100,"RSSI":10}
07:18:42 MQT: tele/tasmota_00543D/SENSOR = {"Time":"2020-05-10T07:18:42","ENERGY":{"TotalStartTime":"2020-05-09T23:21:44","Total":0.016,"Yesterday":0.000,"Today":0.016,"Period":0,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":120,"Current":0.000}}
07:18:49 MQT: domoticz/in = {"idx":241,"nvalue":1,"svalue":"","Battery":100,"RSSI":10}
07:18:59 MQT: domoticz/in = {"idx":241,"nvalue":1,"svalue":"","Battery":100,"RSSI":10}
07:19:09 MQT: domoticz/in = {"idx":241,"nvalue":1,"svalue":"","Battery":100,"RSSI":10}
07:19:19 MQT: domoticz/in = {"idx":241,"nvalue":1,"svalue":"","Battery":100,"RSSI":10}
07:19:29 MQT: domoticz/in = {"idx":241,"nvalue":1,"svalue":"","Battery":100,"RSSI":10}

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2020-05-10:
With the information you provided I can only start the following guessing:

- You configured Domoticz Update Timer set to 10 seconds
- Your Idx of 241 is the Idx1 value, being the power state
- Other indexes I can imagine but are not relevant

Your findings are as designed. Domoticz Update timer ONLY sends Power states to Domoticz. Other telemetry data is send every ``teleperiod`` time which is set to 300 seconds by default.

You normally do not need Domoticz Update Timer as all power states are send immediatley when they occur. Wanting more energy updates just change ``teleperiod 30`` to have updates every 30 seconds.

gknops, 2020-05-10:
Hallo Theo,

Thank you so much for the detailed answer! That was exactly it.

Sorry, I am new to Tasmota and did not expect a setting in _Logging_ to affect MQTT, nor did I find much with google searches or in the various HowTos.

Thanks again for making great software available and spending the surely large amount of time to maintain it, I appreciate it!

Gerd

closed 2020-05-10 by gknops