arendst/Tasmota #8414: Multiple devices reset to default config after MQTT server crash

nunojpg, 2020-05-11:
### PROBLEM DESCRIPTION
I have two devices connected flashed with 8.2.0 for several weeks and connected to MQTT with TLS (CA).
Today I heard a device switching unexpectedly and after opening the web interface I noticed the configuration was lost. I checked another device and same happened.
I checked my MQTT server and there had been a kernel crash due to out of memory at the same time.
Apparently when MQTT connections timeout of fail in any way, it's possible that the device resets to default configuration.
This had happened to me before with version <8.2.0 (not sure which, but from December, probably 8.1.0) but at the time I could not investigate. Now I see it was the exact same problem.

Program Version | 8.2.0(tasmota)
-- | --
2020-04-27T11:37:05
STAGE/2.2.2-dev(38a443e)
0T00:20:42
134 at 0xF6000
30
**Blocked Loop**
Tasmota

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ Sonoff TH, BW-SHP6] Device used (e.g., Sonoff Basic): _____
- [8.2.0] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [x] Self-compiled
    - [x] IDE / Compiler used: Docker
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-05-11:
Probably your AC mains was gone several times.
Set `Setoption 65` and see https://tasmota.github.io/docs/Device-Recovery/#fast-power-cycle-device-recovery

nunojpg, 2020-05-11:
> Probably your AC mains was gone several times.
> Set `Setoption 65 1` and see https://tasmota.github.io/docs/Device-Recovery/#fast-power-cycle-device-recovery

Power quality have been stable without interruptions, micro interruptions or voltage drops for the previous 7 days. We have continuous monitoring of the grid.

The previous time this happened I was doing operations on the MQTT server and as it crashed the device also lost Module configuration.

Now MQTT crashed and both devices connected lost configuration. MQTT is at a google instance. (just to add power is independent).

Likely a payload transmission (after the connection to the topic is estabilished) over MQTT hanging causes this.

blakadder, 2020-05-11:
If the device crashes and bootloops the bootloop protection was activated and device incrementally clears settings until it reaches factory default state as protection

ascillato, 2020-05-11:
The only ways that Tasmota returns all its config to default is by:

* Pressing button1 for more than 40 seconds (if button1 is defined and if multipress is not disabled. See `setoption1`)
* BootLoop protection - you can disable by `SetOption36 0`
* Fast power cycle device recovery - You can disable by `SetOption65 1`

As the requested issue template is not completed, I can't know your GPIO config, so guessing and because you claim your power is stable, seems that you have `buttontopic` defined. So, buttontopic and switchtopic have a fallback feature that in case of mqtt or wifi disconnection, they will start controlling your relay. So if you have a button1 defined that is in the state "1" and your mqtt have disconnected, your button is like pressed for more than 40 seconds for Tasmota, making the full reset.

So, please, change `SetOption36 0`, `SetOption65 1`, `buttontopic 0`, `switchtopic 0` and if you were using any of the last 2, use a rule instead as explained in the docs. We saw that some people use switchtopic with PIR sensors and that was not designed for that. For a PIR or similar use case, a rule is needed.

nunojpg, 2020-05-11:
@ascillato I am sorry for having the initial bug report incomplete. I am not sure about buttontopic/switchtopic.

One of the devices that got reset is a plug with this config:

```
13:26:47 CMD: Backlog Template; Module; GPIO 255
13:26:47 MQT: stat/Plug2/RESULT = {"NAME":"BW-SHP6 10A","GPIO":[158,255,56,255,0,134,0,0,131,17,132,21,0],"FLAG":0,"BASE":45}
13:26:48 MQT: stat/Plug2/RESULT = {"Module":{"0":"BW-SHP6 10A"}}
13:26:48 MQT: stat/Plug2/RESULT = {"GPIO0":{"158":"LedLinki"},"GPIO1":{"0":"None"},"GPIO2":{"56":"Led1i"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"134":"BL0937 CF"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"131":"HLWBL SELi"},"GPIO13":{"17":"Button1"},"GPIO14":{"132":"HLWBL CF1"},"GPIO15":{"21":"Relay1"},"GPIO16":{"0":"None"}}
```

The only change from stock configuration is SetOption59.

There is no chance any button was pressed.

This reset does not happen if WiFi/internet/MQTT broker becomes offline, but only when MQTT broker hangs.

@blakadder in fact I assume the underlying bug report is "device crashes" when MQTT payload transfer hangs. How should I further debug it?

blakadder, 2020-05-11:
create a syslog server or connect through serial and watch logs

ascillato, 2020-05-11:
> How should I further debug it?

Please, go to the console and type:

```lua
SetOption36 0
SetOption65 1
buttontopic 0
switchtopic 0
```

then if you can, set a syslog and try to reproduce your issue again.

Jason2866, 2020-05-11:
I would change Gpio13 from Button1 to Switch1. Doing this a long press will not do a reset

nunojpg, 2020-05-12:
> ```lua
> SetOption36 0
> SetOption65 1
> ```

Regardless of the underlying bug this will fix it for me because the device will no longer clear the configuration when it happens.

closed 2020-05-12 by nunojpg

ascillato, 2020-05-12:
> Regardless of the underlying bug ....

If you want to dig deeper and find the root cause of your reset, please, set a syslog and enable again those protections with `SetOption36 1` and `SetOption65 0`
A syslog with loglevel at 4 will tell us exactly why it is resetting.

> Restart reason immediately after is Blocked loop. Wi-Fi and MQTT configurations are kept. Only Module configuration is lost.

If wifi and mqtt credentials are kept means that the reset is not because of the button but rather because of the bootloop protection. Also, as the wifi connection is going to be there, a syslog should catch all the events.

nunojpg, 2020-05-12:
@ascillato I will keep it in mind, but,

1 - The reason why my server crashed was also fixed and as so this rare problem (twice in 4 months) will likely not be triggered again.
2 -  I tried to manually trigger it by trying to starve the server memory and it did not work. I also tried to open a TCP socket with TLS and the device connects and sends the MQTT hello, but nothing bad happens. It probably needs to be first accepted (connected to topic, I don't know MQTT terms), and then blocked. I need to probably dig a little bit into MQTT to just hardcode the expected answer and then leave it hanging. As soon as I manage to do it then I think about extracting the logs.

Thanks

Jason2866, 2025-06-04:
Use latest Tasmota version, there are many fixes done. There is no support for outdated versions.