arendst/Tasmota #8415: Device reboots and settings lost / corrupted when writing long rules.

garci66, 2020-05-11:
### PROBLEM DESCRIPTION
When writing a long rule (or appending to it), sometimes the device reboots and loses all its config. Won't connect to wifi anymore and serial logs show its trying to connect to a network with gibberish names. The problem occurred several time when trying to configure the rules below in one go.

```
{"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":168,"Rules":"on event#MCPINT_D0 do teleperiod 10 endon  on event#MCPINT_D1 do teleperiod 10 endon  on event#MCPINT_D2 do teleperiod 10 endon  on event#MCPINT_D3 do teleperiod 10 endon  on event#MCPINT_D4 do teleperiod 10 endon  on event#MCPINT_D5 do teleperiod 10 endon  on event#MCPINT_D6 do teleperiod 10 endon  on event#MCPINT_D7 do teleperiod 10 endon "}
```

If I tried entering this rule as a single line, the device would most probably immediately reboot and loose all of its settings. At the time it was still connected over serial, so the log would show it trying to connect to whatever random SSID name (each time different). It was just random names / sometimes "255", etc. Trued the 10-times fast power cycle and the log would show that more and more settings would be reset but it never entered into "AP mode"  no matter what.. it always (even after 10 reboots) would still try to connect to the gibberish SSID.

I ended up adding the rules one at a time:
`Rule1 on event#MCPINT_D0 do teleperiod 10 endon` 
and then 
`Rule1 +on event#MCPINT_D1 do teleperiod 10 endon`

and so on.

But even in those steps it could cause a reboot / settings corruption if I pasted two commands too quickly (i.e, before seeing the response to the first command on the console).

With patience (just waiting 5-10 seconds between each paste) I was able to configure this with no issues and now its working perfectly. But not sure what was causing the corruption.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): ESP-01 (with 1meg flash)
- [X] Tasmota binary firmware version number used: 8.2
  - [ ] Pre-compiled
  - [x] Self-compiled
    - [x] IDE / Compiler used: docker-tasmota (needed MCP23017 support) - using tasmota-sensors build
- [x] Flashing tools used: esptools.py
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
11:51:44 CMD: Backlog Template; Module; GPIO 255
11:51:45 MQT: stat/tasmota_0B00F3/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
11:51:45 MQT: stat/tasmota_0B00F3/RESULT = {"Module":{"0":"Generic"}}
11:51:45 MQT: stat/tasmota_0B00F3/RESULT = {"GPIO0":{"5":"I2C SCL"},"GPIO1":{"77":"SDM630 Tx"},"GPIO2":{"6":"I2C SDA"},"GPIO3":{"78":"SDM630 Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}
```
- [X] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
11:52:44 CMD: Rule1
11:52:44 MQT: stat/tasmota_0B00F3/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":168,"Rules":"on event#MCPINT_D0 do teleperiod 10 endon  on event#MCPINT_D1 do teleperiod 10 endon  on event#MCPINT_D2 do teleperiod 10 endon  on event#MCPINT_D3 do teleperiod 10 endon  on event#MCPINT_D4 do teleperiod 10 endon  on event#MCPINT_D5 do teleperiod 10 endon  on event#MCPINT_D6 do teleperiod 10 endon  on event#MCPINT_D7 do teleperiod 10 endon "}
11:52:46 CMD: Rule2
11:52:46 MQT: stat/tasmota_0B00F3/RESULT = {"Rule2":"ON","Once":"OFF","StopOnError":"OFF","Free":164,"Rules":"on event#MCPINT_D8 do teleperiod 10 endon  on event#MCPINT_9 do teleperiod 10 endon  on event#MCPINT_D10 do teleperiod 10 endon  on event#MCPINT_D11 do teleperiod 10 endon on event#MCPINT_D12 do teleperiod 10 endon  on event#MCPINT_D13 do teleperiod 10 endon  on event#MCPINT_D14 do teleperiod 10 endon  on event#MCPINT_D15 do teleperiod 10 endon "}

```
- [X] Provide the output of this command: `Status 0`:
```
11:53:21 MQT: stat/tasmota_0B00F3/STATUS1 = {"StatusPRM":{"Baudrate":9600,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"1T14:59:27","StartupUTC":"2020-05-09T23:53:54","Sleep":50,"CfgHolder":4617,"BootCount":15,"BCResetTime":"2020-04-27T01:35:07","SaveCount":32360,"SaveAddress":"F7000"}}
11:53:21 MQT: stat/tasmota_0B00F3/STATUS2 = {"StatusFWR":{"Version":"8.2.0(sensors)","BuildDateTime":"2020-05-09T23:51:36","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"356/699"}}
11:53:21 MQT: stat/tasmota_0B00F3/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["garci",""],"TelePeriod":10,"Resolution":"558180C0","SetOption":["000A8009","2805C8000100060000005A00000000000000","004002C0","00000000"]}}
11:53:21 MQT: stat/tasmota_0B00F3/STATUS4 = {"StatusMEM":{"ProgramSize":603,"Free":400,"Heap":22,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440E0","FlashMode":3,"Features":["00000809","8FDAA387","0415A3A5","B7FFBFCD","01DA9BD5","24367CC7","00000002"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,14,16,17,20,21,24,29,34","Sensors":"1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,26,28,29,31,34,37,39,40,42,43,51,52,55,56,58,59,62,64"}}
11:53:21 MQT: stat/tasmota_0B00F3/STATUS5 = {"StatusNET":{"Hostname":"tasmota_0B00F3-0243","IPAddress":"192.168.1.216","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"8.8.8.8","Mac":"5C:CF:7F:0B:00:F3","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
11:53:21 MQT: stat/tasmota_0B00F3/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.170","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_0B00F3","MqttUser":"mqtt","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
11:53:21 MQT: stat/tasmota_0B00F3/STATUS7 = {"StatusTIM":{"UTC":"2020-05-11T14:53:21","Local":"2020-05-11T11:53:21","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"-03:00","Sunrise":"01:13","Sunset":"16:20"}}
11:53:21 MQT: stat/tasmota_0B00F3/STATUS9 = {"StatusPTH":{"PowerDelta":0,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
11:53:22 MQT: stat/tasmota_0B00F3/STATUS10 = {"StatusSNS":{"Time":"2020-05-11T11:53:21","ENERGY":{"TotalStartTime":"2020-05-11T11:53:18","Total":86.280,"Yesterday":11.940,"Today":3.420,"Power":[267,null,null],"ApparentPower":[323,null,null],"ReactivePower":[-181,0,0],"Factor":[0.83,null,null],"Voltage":[226,null,null],"Current":[1.430,null,null]},"MCP230XX":{"D0":0,"D1":1,"D2":0,"D3":1,"D4":0,"D5":1,"D6":0,"D7":0,"D8":1,"D9":0,"D10":0,"D11":0,"D12":0,"D13":0,"D14":1,"D15":0}}}
11:53:22 MQT: stat/tasmota_0B00F3/STATUS11 = {"StatusSTS":{"Time":"2020-05-11T11:53:22","Uptime":"1T14:59:28","UptimeSec":140368,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"garci","BSSId":"80:2A:A8:44:00:1A","Channel":11,"RSSI":100,"Signal":-44,"LinkCount":1,"Downtime":"0T00:00:07"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
not available

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2020-05-11:
You're on target! 

I just revisited the serial handling of long commands. Give the latest development branch a try and let me know if that solves your issue.

A quick paste of 
```
rule3 on event#MCPINT_D0 do teleperiod 10 endon  on event#MCPINT_D1 do teleperiod 10 endon  on event#MCPINT_D2 do teleperiod 10 endon  on event#MCPINT_D3 do teleperiod 10 endon  on event#MCPINT_D4 do teleperiod 10 endon  on event#MCPINT_D5 do teleperiod 10 endon  on event#MCPINT_D6 do teleperiod 10 endon  on event#MCPINT_D7 do teleperiod 10 endon
```
works fine on my device.

effelle, 2020-05-11:
Sorry @ascillato we changed the labels at the same time :P

arendst, 2020-05-11:
Ah. I see you probably have another problem running low on memory usin ghte sensor binary.

BTW I suppose you meant this as the full command line:
```
rule on event#MCPINT_D0 do teleperiod 10 endon  on event#MCPINT_D1 do teleperiod 10 endon  on event#MCPINT_D2 do teleperiod 10 endon  on event#MCPINT_D3 do teleperiod 10 endon  on event#MCPINT_D4 do teleperiod 10 endon  on event#MCPINT_D5 do teleperiod 10 endon  on event#MCPINT_D6 do teleperiod 10 endon  on event#MCPINT_D7 do teleperiod 10 endon
```
as you cannot enter JSON as a rule option

arendst, 2020-05-11:
Just tried using the latest development branch:

```
17:23:59 CMD: status 0
17:23:59 MQT: stat/wemos4/STATUS = {"Status":{"Module":18,"FriendlyName":["Wemos4","Tasmota2","Tasmota3","Tasmota4",""],"Topic":"wemos4","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
17:23:59 MQT: stat/wemos4/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/tasmota-sensors.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:07:19","StartupUTC":"2020-05-11T15:16:40","Sleep":50,"CfgHolder":4617,"BootCount":54,"BCResetTime":"2020-04-20T17:11:44","SaveCount":154,"SaveAddress":"F9000"}}
17:24:00 MQT: stat/wemos4/STATUS2 = {"StatusFWR":{"Version":"8.2.0.6(001b6e4-sensors)","BuildDateTime":"2020-05-11T16:01:39","Boot":31,"Core":"2_7_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"334/699"}}
17:24:00 MQT: stat/wemos4/STATUS3 = {"StatusLOG":{"SerialLog":3,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"domus1","LogPort":514,"SSId":["indebuurt","indebuurt_IoT"],"TelePeriod":60,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","008000C0","00000000"]}}
17:24:00 MQT: stat/wemos4/STATUS4 = {"StatusMEM":{"ProgramSize":621,"Free":380,"Heap":21,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640E0","FlashMode":3,"Features":["00000809","8FDAA787","0415A3A5","B7FFBFCD","01DA9BC4","64367CC7","00000072"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,14,16,17,20,21,24,29,34","Sensors":"1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,51,52,55,56,58,59,62,64,66,67"}}
17:24:00 MQT: stat/wemos4/STATUS5 = {"StatusNET":{"Hostname":"wemos4","IPAddress":"192.168.2.223","Gateway":"192.168.2.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.27","Mac":"5C:CF:7F:83:BB:10","Webserver":2,"WifiConfig":5,"WifiPower":17.0}}
17:24:00 MQT: stat/wemos4/STATUS6 = {"StatusMQT":{"MqttHost":"domus1","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_83BB10","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
17:24:00 MQT: stat/wemos4/STATUS7 = {"StatusTIM":{"UTC":"2020-05-11T15:24:00","Local":"2020-05-11T17:24:00","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":99,"Sunrise":"05:38","Sunset":"21:17"}}
17:24:00 MQT: stat/wemos4/STATUS10 = {"StatusSNS":{"Time":"2020-05-11T17:24:00"}}
17:24:00 MQT: stat/wemos4/STATUS11 = {"StatusSTS":{"Time":"2020-05-11T17:24:00","Uptime":"0T00:07:20","UptimeSec":440,"Heap":21,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":23,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","POWER5":"OFF","Wifi":{"AP":2,"SSId":"indebuurt_IoT","BSSId":"18:E8:29:CA:17:C1","Channel":11,"RSSI":100,"Signal":-29,"LinkCount":1,"Downtime":"0T00:00:07"}}}
17:24:10 CMD: rule on event#MCPINT_D0 do teleperiod 10 endon  on event#MCPINT_D1 do teleperiod 10 endon  on event#MCPINT_D2 do teleperiod 10 endon  on event#MCPINT_D3 do teleperiod 10 endon  on event#MCPINT_D4 do teleperiod 10 endon  on event#MCPINT_D5 do teleperiod 10 endon  on event#MCPINT_D6 do teleperiod 10 endon  on event#MCPINT_D7 do teleperiod 10 endon
17:24:10 RUL: Stored uncompressed, would compress from 342 to 69 (-80%)
17:24:11 MQT: stat/wemos4/RESULT = {"Rule1":"OFF","Once":"OFF","StopOnError":"OFF","Length":342,"Free":169,"Rules":"on event#MCPINT_D0 do teleperiod 10 endon  on event#MCPINT_D1 do teleperiod 10 endon  on event#MCPINT_D2 do teleperiod 10 endon  on event#MCPINT_D3 do teleperiod 10 endon  on event#MCPINT_D4 do teleperiod 10 endon  on event#MCPINT_D5 do teleperiod 10 endon  on event#MCPINT_D6 do teleperiod 10 endon  on event#MCPINT_D7 do teleperiod 10 endon"}
```
No problem at all.

garci66, 2020-05-11:
Thanks @arendst!! will make a build out of master and let you know. Is the corruption of the settings a known issue when running out of memory or parsing that line? Cause the only way I managed to recover the device was a full flash wipe and re-flash. The "40 second" gpio0 / or 10 fast reboots didnt work...  

The config corruption seemed to be different each time (it happened around 10 times until I was very careful and patient adding the commands).

Will try ASAP. 

And once again, thanks for the amazing work! I have a single ESP01 monitoring 16 digital inputs and a power monitor over modbus... love it!

arendst, 2020-05-11:
Build out of master will provide the current released version (8.2.0)

You'll need to build out of Development to get the corrected version (8.2.0.6)

closed 2020-05-16 by ascillato2