arendst/Tasmota #8416: Add support for new power meter (Schneider PM800)

garci66, 2020-05-11:
**Have you looked for this feature in other issues and in the docs?**  
Yes - found similar other power meters so wanted to implement this one as well. Its a very high quality power meter that can be found on ebay at relatively good prices sometimes. https://www.se.com/us/en/product-range-presentation/918-pm800-series/

**Is your feature request related to a problem? Please describe.**  
No specific problem.

**Describe the solution you'd like**  
 I have several of these meters and have donde an adaptation of the SDM630 module to support this model (by re-writing the modbus register address). I was wondering how to best integrate this into the code. Seems a bit wasteful to have multiple "identical" power meters where the only difference is the modbus registers.

**Describe alternatives you've considered**  
I can keep this as my private branch but wanted to see if there were better options on how to integrate this (and potentially other power meters) where the main factor is just the register address.

**Additional context**  
I can provide a PR but right now its agains the SDM630 module. So I can either add it as a new module or find a way to modularize the register addresses.

Additionally, for 3-phase meters, I've noticed that the floating point registers for the phases tend to be contiguous. For bus eficiency, we could read them in one go, requesting 6 registers instead of 2 on each pass of the loop. Populating all 3-phase currents in one go, then all 3 phase voltages, etc... 

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-05-11:
Great, please provide that PR. Thanks.

kugelkopf123, 2020-05-11:
There is already support for Modbus RTU devices and many other electricity meters. Have a look at: [link](https://tasmota.github.io/docs/Smart-Meter-Interface/)

The important data as well as registers are easy to enter in the Scripter. Further down on the linked page you will find examples with other Modbus devices. You can compare the registers from the manuals of the example devices with the respective examples. This often helps to find out exactly how it works if there are problems.

kugelkopf123, 2020-05-11:
It isnt that difficult. But yes the Docu could be better. 
Perhaps is the SDM630 and the SchneiderPM800 nearly identically with the [sdm530](https://tasmota.github.io/docs/Smart-Meter-Interface/#sdm530)

theoretically you just need to add the following to user_config_override.h:
```
#ifndef USE_SCRIPT
#define USE_SCRIPT  
#endif
#ifdef USE_RULES
#undef USE_RULES
#endif

#define SML_MAX_VARS 30 // Number of Vars you'll need in the Descriptor
#define USE_SML_M
```

Und dann das dementsprechende Script in dem Scripter in den Einstellungen der WEBGUI.

If so, please provide your SML Script (Registers etc...) so I can add it to the Docs for other users.

garci66, 2020-05-11:
Thanks! will try it out!.  I notice that yes, it can be shown int the web-ui.  To send it over mqtt I also use the json prefix, no?

With the SDM630 (the native module where I changed the registers) I didnt have to do anything at all to get all the values in home-assistant.  I totally appreciate the flexibility of this mechanism, don't get me wrong.. but its also a lot easier.

The "SML_MAX_VARS" parameter is for the total number of variables I want to use? (if I had multiple (4, for example) power meters in a row in the same modbus bus, each with a different address and I get, lets say, 15 values each, I would set the max_vars to 60?)

Thanks a  lot!

kugelkopf123, 2020-05-11:
If you set the teleperiod to 10, then every 10 seconds the read out values are transmitted via MQTT. Unfortunately I can only speak of ioBroker. I do not use HassIO. 

No problem. You can of course add the PullReq. I just wanted to show another way how to add a Modbus device. 

Yes SML_MAX_VARS defines the Number of dif. values you have. Like Total_inDev1,TotPower_Dev1,Power_L1Dev1,Power_L2_Dev1...etc. 
If you want to know how it all works in detail you have to ask @gemu2015  about it. So far I don't know anybody who has read more than 2 Modbus devices. Unfortunately the number of characters in the script is limited. So there is a lot to do. 
If only the ID changes with 4 devices, maybe there is a way to put something like this into a loop @gemu2015 ?

More information, what is possible with the scripteditor, you can find in the [wiki](https://tasmota.github.io/docs/Scripting-Language/). There are some Examples.

twinkelbel, 2022-03-28:
hello, i also have a PM800 firmware 12,5  that i want to use with Tasmota and Domoticz, but i not get the scribt to work,
I see tasmota sending data to the PM800 (tx led on rs485 converter)
but the PM800 not answere, did you find out how to get it work?
Could you please share your scribt?
Thank you, Robert

garci66, 2022-03-28:
if the PM800 is not factory reset, you should check its modbus address. Also, did you try reversing the modbus cables (A and B)? Also, are you compiling your own code for tasmota? or using it out of the box?

twinkelbel, 2022-03-28:
hello, thanks for answere.

the com in the PM800 i set to modbus rtu 9600 adress 1, 8N1

i compiled tasmota with 
user_config_override.h:

#ifndef USE_SCRIPT
#define USE_SCRIPT
#endif
#ifndef USE_SML_M
#define USE_SML_M
#endif
#ifdef USE_RULES
#undef USE_RULES
#endif

Then i used the scribt from SDM 530 (modbus)example

>D  
>B  
->sensor53 r
>M 1  
+1,3,m,0,9600,MODBUS,1,1,01040000,01040002,01040004,01040006,01040008,0104000a,0104000c,0104000e,01040010  
1,010404ffffffff@i0:1,Voltage P1,V,Voltage_P1,2  
1,010404ffffffff@i1:1,Voltage P2,V,Voltage_P2,2  
1,010404ffffffff@i2:1,Voltage P3,V,Voltage_P3,2  
1,010404ffffffff@i3:1,Current P1,A,Current_P1,2  
1,010404ffffffff@i4:1,Current P2,A,Current_P2,2  
1,010404ffffffff@i5:1,Current P3,A,Current_P3,2  
1,010404ffffffff@i6:1,Active Power P1,W,Power_P1,2  
1,010404ffffffff@i7:1,Active Power P2,W,Power_P2,2  
1,010404ffffffff@i8:1,Active Power P3,W,Power_P3,2  
#  
i changed it  to test.

>D  
>B  
->sensor53 r
>M 1  
+1,3,m,0,9600,PM800,1,1,01041124,01041143 
1,010404uuUUuuUU@i0:1,Voltage P1,V,Voltage_P1,2  
1,010404ffffffff@i6:1,Active Power Total,W,Power,2  
#  

registerlist PM800
Voltage A-N  (integer) 1124
Real Power, Total, (integer or is it signed integer?) 1143

in console Sensor53 d1, i get raw data

18:55:32.713 HTP: Consoles
18:55:39.259 WIF: Controleer verbinding...
18:55:39.666 HTP: Console
18:55:48.180 CMD: sensor53 d1
18:55:48.182 SRC: WebConsole from 192.168.1.100
18:55:48.183 CMD: Grp 0, Cmd 'SENSOR', Idx 53, Len 2, Pld -99, Data 'd1'
18:55:48.187 RSL: RESULT = {"Time":"2022-03-28T18:55:48","SML":{"CMD":"dump: 1"}}
18:55:48.292 > 01 04 11 24 00 02 34 fc 
18:55:48.388 > 01 04 11 43 00 02 85 23 
18:55:48.492 > 01 04 11 24 00 02 34 fc 
18:55:48.592 > 01 04 11 43 00 02 85 23 
18:55:48.693 > 01 04 11 24 00 02 34 fc 
18:55:48.792 > 01 04 11 43 00 02 85 23 
18:55:48.892 > 01 04 11 24 00 02 34 fc 
18:55:48.993 > 01 04 11 43 00 02 85 23 

Because i get data back i assume the connection is working.

 but after sensor53 d0 still nothing:

I wonder do i have to put a 3 or 4 in front of the registernumber? input register or holding register?
The register number are in DEC, but i read i have to put it in HEX, if to put it in HEX do i also need to put the 3 or 4 also in HEX calculation? 
voltage register 1124, HEX 464 or 31124 HEX 7924 or 41124 HEX a0a4????
also play with 1,010404ffffffff@ and 1,010404uuUUuuUU@
I assume also here i have to change 4 to 3 1,010304ffffffff@ to test.
I already try all kinds of combinations but still no output

thanks for any help

Robert

gemu2015, 2022-03-29:
you do NOT get any response from the meter. what you see is the request sent to the meter
check the LEDs on the interface. TX LED should flash and RX too.
if both flash you should see the response right after the request in dump mode.

twinkelbel, 2022-03-30:
Thank you to help,

yesterday i tried all day to understand how this works.
i tried without the sonoff basic and used usb-ttl converter to test the rs485 adapter and this also not worked, seems the ttl-rs485 adapter was not working, luckily i ordered 2 and the second one worked as expected!

now i see data but the register is offset, do i overlay registers while reading? wrong HEX calculation?

                         regnr pm800               Hex                    in scribt tasmota           value
current ph1           1100                       44c                          44b                         0-32767 integer
current ph2           1101                       44d                          44c                         0-32767 integer
current ph3           1102                       44e                          44d                         0-32767 integer
voltage ph1          1124                       464                          463                         0-32767 integer
voltage ph2          1125                       465                          464                         0-32767 integer
voltage ph3          1126                       466                          465                         0-32767 integer
total real power    1143                      477                           476               -32767-32767 integer

my scribt now looks like:

>D  
>B  
->sensor53 r
>M 1  
+1,3,m,0,9600,PM800,1,1,01030463,01030464,01030465,0103044b,0103044c,0103044d,01030476 
1,010304UUuuxxxx@i0:10,Voltage P1,V,Voltage_P1,2  
1,010304UUuuxxxx@i1:10,Voltage P2,V,Voltage_P2,2
1,010304UUuuxxxx@i2:10,Voltage P3,V,Voltage_P3,2
1,010304UUuuxxxx@i3:100,Current P1,A,Current_P1,2
1,010304UUuuxxxx@i4:100,Current P2,A,Current_P2,2
1,010304UUuuxxxx@i5:100,Current P3,A,Current_P3,2
1,010304SSssxxxx@i6:1,Real energy,Wh,Real_energy,2
#

I used devider because i tricked the PM800 to get higher resolution. 
current transformers are 50-5 but in config PM800 i set 5000-5.
Same with voltage.
now i get watt in/out not Kw on display of PM800

sensor53 d1 i get now:
09:46:04.580 : 01 03 04 01 c2 01 52 da 5e 
09:46:04.633 > 01 03 04 4d 00 02 55 2c 
09:46:04.681 : 01 03 04 01 52 01 06 db 8c 
09:46:04.732 > 01 03 04 76 00 02 24 e1 
09:46:04.781 : 01 03 04 f7 05 fe 5d 59 df 
09:46:04.832 > 01 03 04 63 00 02 35 25 
09:46:04.881 : 01 03 04 09 5d 08 f6 ef fb 
09:46:04.934 > 01 03 04 64 00 02 84 e4 
09:46:04.982 : 01 03 04 08 f6 08 ef 5e 2d 
09:46:05.031 > 01 03 04 65 00 02 d5 24 
09:46:05.081 : 01 03 04 08 f2 00 32 d8 75 
09:46:05.131 > 01 03 04 4b 00 02 b5 2d 
09:46:05.181 : 01 03 04 01 16 01 c0 1b cb 
09:46:05.233 > 01 03 04 4c 00 02 04 ec 
09:46:05.281 : 01 03 04 01 c0 01 50 fa 5f 
09:46:05.332 > 01 03 04 4d 00 02 55 2c 
09:46:05.381 : 01 03 04 01 50 01 06 7a 4c 
09:46:05.433 > 01 03 04 76 00 02 24 e1 
09:46:05.534 > 01 03 04 63 00 02 35 25 
09:46:05.581 : 01 03 04 09 62 08 f2 de 34 
09:46:05.633 > 01 03 04 64 00 02 84 e4 

[image]
Not know if this screenshot from GUI works...

Now i see the real power 1000,00 wh is it possible just to see 1000 wh, no need to see smaller then 1.

next problem is to get all this data into Domoticz with MQTT, if scribt is used it seems not possible to use rules 
to give voltage, current and power his own IDX.

it seems possible in scribt too, but to get this working took me already one day

Thanks for the great help!

Robert