arendst/Tasmota #8419: Please add support for Aqara Opple Wireless Scene Switch

raduseitan, 2020-05-11:
### PROBLEM DESCRIPTION
Aqara Opple Wireless Scene Switch 2 Button (WXCJKG11LM), Aqara Opple Wireless Scene Switch 4 Button (WXCJKG12LM), Aqara Opple Wireless Scene Switch 6 Button (WXCJKG13LM) are not implement in tasmota\xdrv_23_zigbee_5_converters.ino

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): NodeMCU + CC2530 (DL-20)
- [x] Tasmota binary firmware version number used: 8.2.0.6
  - [ ] Pre-compiled
  - [x] Self-compiled
    - [x] IDE / Compiler used: VSCode + PlatformIO
- [x] Flashing tools used: tasmotizer_x86-1.1a
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
21:44:47 CMD: Backlog Template; Module; GPIO 255
21:44:47 SRC: WebConsole from 192.168.1.81
21:44:47 CMD: Group 0, Index 1, Command "BACKLOG", Data "Template; Module; GPIO 255"
21:44:47 SRC: Backlog
21:44:47 CMD: Group 0, Index 1, Command "TEMPLATE", Data ""
21:44:47 MQT: stat/zigbee2tasmota/RESULT = {"NAME":"Zigbee","GPIO":[0,0,0,0,0,0,0,0,0,166,0,165,0],"FLAG":0,"BASE":18}
21:44:47 SRC: Backlog
21:44:47 CMD: Group 0, Index 1, Command "MODULE", Data ""
21:44:47 MQT: stat/zigbee2tasmota/RESULT = {"Module":{"0":"Zigbee"}}
21:44:47 SRC: Backlog
21:44:47 CMD: Group 0, Index 1, Command "GPIO", Data "255"
21:44:47 MQT: stat/zigbee2tasmota/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"166":"Zigbee Rx"},"GPIO14":{"0":"None"},"GPIO15":{"165":"Zigbee Tx"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
21:47:37 CMD: STATUS 0
21:47:37 SRC: WebConsole from 192.168.1.81
21:47:37 CMD: Group 0, Index 1, Command "STATUS", Data "0"
21:47:37 MQT: stat/zigbee2tasmota/STATUS = {"Status":{"Module":0,"FriendlyName":["Zigbee2Tasmota"],"Topic":"zigbee2tasmota","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
21:47:37 MQT: stat/zigbee2tasmota/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"External System","Uptime":"3T05:50:36","StartupUTC":"2020-05-08T14:57:01","Sleep":50,"CfgHolder":4617,"BootCount":19,"BCResetTime":"2020-05-07T21:16:41","SaveCount":43,"SaveAddress":"F9000"}}
21:47:37 MQT: stat/zigbee2tasmota/STATUS2 = {"StatusFWR":{"Version":"8.2.0.6(tasmota)","BuildDateTime":"2020-05-07T21:22:17","Boot":31,"Core":"2_7_0","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"369/699"}}
21:47:37 MQT: stat/zigbee2tasmota/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":3,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["OPC",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["000A8009","2805C8000100060000005A00000000000000","00000200","00000000"]}}
21:47:37 MQT: stat/zigbee2tasmota/STATUS4 = {"StatusMEM":{"ProgramSize":635,"Free":368,"Heap":22,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640EF","FlashMode":2,"Features":["00000809","8FDAE797","047683A1","000000CD","010013C0","C000F981","00000024"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,23,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
21:47:37 MQT: stat/zigbee2tasmota/STATUS5 = {"StatusNET":{"Hostname":"zigbee2tasmota-6779","IPAddress":"192.168.1.90","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"60:01:94:4A:5A:7B","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
21:47:37 MQT: stat/zigbee2tasmota/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.105","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_4A5A7B","MqttUser":"albertmqtt","MqttCount":3,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
21:47:37 MQT: stat/zigbee2tasmota/STATUS7 = {"StatusTIM":{"UTC":"2020-05-11T20:47:37","Local":"2020-05-11T21:47:37","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"05:13","Sunset":"20:20"}}
21:47:37 MQT: stat/zigbee2tasmota/STATUS10 = {"StatusSNS":{"Time":"2020-05-11T21:47:37"}}
21:47:37 MQT: stat/zigbee2tasmota/STATUS11 = {"StatusSTS":{"Time":"2020-05-11T21:47:37","Uptime":"3T05:50:36","UptimeSec":280236,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":22,"MqttCount":3,"Wifi":{"AP":1,"SSId":"OPC","BSSId":"24:F5:A2:A4:14:41","Channel":1,"RSSI":86,"Signal":-57,"LinkCount":1,"Downtime":"0T00:00:06"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
1. From internet:
https://www.zigbee2mqtt.io/devices/WXCJKG11LM.html
[... 126 lines cut ...]
22:20:28 ZIG: {"ZbZNPReceived":"44810000080080A6010101310039E3C300000911D70201550700000080A61D"}
22:20:28 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":8,"srcaddr":"0xA680","srcendpoint":1,"dstendpoint":1,"wasbroadcast":1,"LinkQuality":49,"securityuse":0,"seqnumber":0,"timestamp":12837689,"fc":"0x11","manuf":"0x0000","transact":215,"cmdid":"0x02","payload":"015507000000"}}
22:20:28 ZIG: ZbZCLRawReceived: {"0xA680":{"0008!02":"015507000000","DimmerStep":1}}
22:20:28 MQT: tele/zigbee2tasmota/SENSOR = {"ZbReceived":{"0xA680":{"Name":"Aqara Opple WXCJKG12LM","0008!02":"015507000000","DimmerStep":1,"Endpoint":1,"LinkQuality":49}}}
22:20:29 ZIG: ZbZNPSent 24020280A6000000000000010000010800F2301E050000F2000000
22:20:29 ZIG: {"ZbZNPReceived":"640200"}
22:20:36 ZIG: {"ZbZNPReceived":"4480F001F2"}
22:20:36 MQT: tele/zigbee2tasmota/RESULT = {"ZbConfirm":{"Endpoint":1,"Status":240,"StatusMessage":""}}
```
row 2 right button, long click:
```22:21:59 ZIG: {"ZbZNPReceived":"44810000080080A6010B0144007E41C800000911D90200550700000080A61D"}
22:21:59 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":8,"srcaddr":"0xA680","srcendpoint":1,"dstendpoint":11,"wasbroadcast":1,"LinkQuality":68,"securityuse":0,"seqnumber":0,"timestamp":13123966,"fc":"0x11","manuf":"0x0000","transact":217,"cmdid":"0x02","payload":"005507000000"}}
22:21:59 ZIG: ZbZCLRawReceived: {"0xA680":{"0008!02":"005507000000","DimmerStep":0}}
22:22:00 MQT: tele/zigbee2tasmota/SENSOR = {"ZbReceived":{"0xA680":{"Name":"Aqara Opple WXCJKG12LM","0008!02":"005507000000","DimmerStep":0,"Endpoint":1,"LinkQuality":68}}}
22:22:00 ZIG: {"ZbZNPReceived":"44810000080080A601010144007E41C800000911D90200550700000080A61D"}
22:22:00 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":8,"srcaddr":"0xA680","srcendpoint":1,"dstendpoint":1,"wasbroadcast":1,"LinkQuality":68,"securityuse":0,"seqnumber":0,"timestamp":13123966,"fc":"0x11","manuf":"0x0000","transact":217,"cmdid":"0x02","payload":"005507000000"}}
22:22:00 ZIG: ZbZCLRawReceived: {"0xA680":{"0008!02":"005507000000","DimmerStep":0}}
22:22:00 MQT: tele/zigbee2tasmota/SENSOR = {"ZbReceived":{"0xA680":{"Name":"Aqara Opple WXCJKG12LM","0008!02":"005507000000","DimmerStep":0,"Endpoint":1,"LinkQuality":68}}}
22:22:01 ZIG: ZbZNPSent 24020280A6000000000000010000010800F4301E050000F4000000
22:22:01 ZIG: {"ZbZNPReceived":"640200"}
22:22:02 ZIG: {"ZbZNPReceived":"4480F001F3"}
22:22:02 MQT: tele/zigbee2tasmota/RESULT = {"ZbConfirm":{"Endpoint":1,"Status":240,"StatusMessage":""}}
```

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-05-12:
If you press one button there is one message in sensor topic `22:26:42 MQT: tele/zigbee2tasmota/SENSOR = {"ZbReceived":{"0xA680":{"Name":"Aqara Opple WXCJKG12LM","0006!00":"","Power":0,"Endpoint":1,"LinkQuality":70}}}`
It works as designed

raduseitan, 2020-05-12:
If I press one button there are two messages in  sensor topic:
```
07:54:11 MQT: tele/zigbee2tasmota/SENSOR = {"ZbReceived":{"0xA680":{"Name":"Aqara Opple WXCJKG12LM","0006!00":"","Power":0,"Endpoint":1,"LinkQuality":115}}}
07:54:11 MQT: tele/zigbee2tasmota/SENSOR = {"ZbReceived":{"0xA680":{"Name":"Aqara Opple WXCJKG12LM","0006!00":"","Power":0,"Endpoint":1,"LinkQuality":115}}}
07:54:18 MQT: tele/zigbee2tasmota/RESULT = {"ZbConfirm":{"Endpoint":1,"Status":240,"StatusMessage":""}}
``` 

[image]

s-hadinger, 2020-05-12:
Z2T coordinator opens two endpoints: 0x01 and 0x0B (0x0B is hardcoded in some Xiaomi devices). The device sends two messages, one for each endpoint. Unfortunately there's nothing much I can do, unless you use bindings. I suggest to bind to group addresses, it makes things easier and more flexible.

Let me check the other events.

raduseitan, 2020-05-12:
thank you for your answer. 
I will look into bindings (I'm terrible new to zigbee and HA automations)

ascillato2, 2020-05-12:
Closing this issue as it has been answered.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-05-12 by ascillato2

raduseitan, 2020-05-12:
sorry to bother you, but both sensor topic messages have the same "Endpoint":1.

If there are two endpoints: 0x01 and 0x0B, we shouldn't have to see different endpoints in sensor topic? something like:
```
07:54:11 MQT: tele/zigbee2tasmota/SENSOR = {"ZbReceived":{"0xA680":{"Name":"Aqara Opple WXCJKG12LM","0006!00":"","Power":0,"Endpoint":1,"LinkQuality":115}}}
07:54:11 MQT: tele/zigbee2tasmota/SENSOR = {"ZbReceived":{"0xA680":{"Name":"Aqara Opple WXCJKG12LM","0006!00":"","Power":0,"Endpoint":11,"LinkQuality":115}}}
07:54:18 MQT: tele/zigbee2tasmota/RESULT = {"ZbConfirm":{"Endpoint":1,"Status":240,"StatusMessage":""}}
```

raduseitan, 2020-05-12:
Ok, done my homework on Zigbee Binding.
As far as I understood, binding has sense inside the Zigbee netwok for a device 2 device communication. Useful for a simple button to lamp automation or one button for more lights, etc.

What I really need is to have this sensor information in MQTT from where some other (nonZigbee) devices should process this information.

If the "Endpoint" from sensor topic message is the source endpoint from ZbZCLReceived, could we also have the destination endpoint added to sensor topic  message?

Should I open a new feature request?
Thank you.

s-hadinger, 2020-05-12:
Hi, I pushed a PR to support `"ColorTempStep<Up/Down>` for the double click, and a better `"DimmerStep"` for long clic. Please try with the latest version.

On endpoints, the Aqara device seems to send to both endpoints for greater compatibility reason. Normally, when bound to a group address, it should stop sending to both endpoints and send to the group address instead. The coordinator will receive all messages to all groups addresses. To bind a device to the group address, use the `AddGroup` command.

There are basically two types of messages from devices: 1/ either an attribute report, which is generally a sensor read (temperature, humidity...); in this case you can't use bindings; 2/ commands that are send from a source to a targer (a server to a client in Zigbee terminology); in this case you can use bindings to direct the command to a specific device or to a gourp.

All the logs you show above are commands, not sensor readings.

Full doc on groups addresses and bindings are here: https://tasmota.github.io/docs/Zigbee/#zigbee-groups

PR #8430 (2020-05-12): Zigbee optimized state machine and support for better ZCL Step commands
## Description:

Optimized Zigbee state machine for Routers and End Devices.

Better support for `DimmerStep`, `ColorStep` and `ColorTempStep`.

Potentially breaking change, now reports `"DimmerStepDown":43` for IKEA Remote instead of `"DimmerStep":1`, which gives more information about the size of the step.

**Related issue (if applicable):** fixes #8419

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR.
  -  [... cut]