arendst/Tasmota #8420: sonoff tx t3 on 8.2.0 radom (ghost) action {"TRIG": "TOGGLE"}

BrnEmilio, 2020-05-11:
**sonoff tx t3** on **8.2.0** radom (ghost) action **{"TRIG": "TOGGLE"}**

### PROBLEM DESCRIPTION
I have the same random problem in tasmota, when I use version 8.2 on the Sonoff tx t3 device with 3 keys  (buttons). 

I desconnected mqtt to discard configuration with home assistant.

Notes: When I use the '**Sonoff T1 1CH (28**)' module after the '_reset 5_' command via console, the error does not occur, but I selected the '**Sonoff T1 3CH (30)**' module, but the random error occurs in the POWER action : ON or POWER: OFF _without clicking_ the button.
 
### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): **sonoff tx t3**
- [x] Tasmota binary firmware version number used: **8.2.0 tasmota.bin (Elliot)**
  - [x] Pre-compiled
  - [] Self-compiled
    - [] IDE / Compiler used: 
- [x] Flashing tools used: **Tasmotizer** (https://github.com/tasmota/tasmotizer)
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
23:31:47 RSL: stat/tasmota_779740/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
23:31:47 RSL: stat/tasmota_779740/RESULT = {"Module":{"30":"Sonoff T1 3CH"}}
23:31:48 RSL: stat/tasmota_779740/RESULT = {"GPIO0":{"17":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"23":"Relay3"},"GPIO5":{"22":"Relay2"},"GPIO9":{"18":"Button2"},"GPIO10":{"19":"Button3"},"GPIO12":{"21":"Relay1"},"GPIO13":{"56":"Led1i"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
```
- [x] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
23:33:08 CMD: Status 0
23:33:08 RSL: stat/tasmota_779740/STATUS = {"Status":{"Module":30,"FriendlyName":["Tasmota","Tasmota2","Tasmota3"],"Topic":"tasmota_779740","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
23:33:08 RSL: stat/tasmota_779740/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:01:40","StartupUTC":"2020-05-11T22:31:28","Sleep":50,"CfgHolder":4617,"BootCount":2,"BCResetTime":"2020-05-11T23:21:25","SaveCount":8,"SaveAddress":"F4000"}}
23:33:08 RSL: stat/tasmota_779740/STATUS2 = {"StatusFWR":{"Version":"8.2.0(tasmota)","BuildDateTime":"2020-03-20T14:45:23","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8285","CR":"306/699"}}
23:33:08 RSL: stat/tasmota_779740/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["4u2",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
23:33:08 RSL: stat/tasmota_779740/STATUS4 = {"StatusMEM":{"ProgramSize":577,"Free":424,"Heap":27,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","000000CD","010013C0","C000F981","00000004"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
23:33:08 RSL: stat/tasmota_779740/STATUS5 = {"StatusNET":{"Hostname":"tasmota_779740-5952","IPAddress":"192.168.13.157","Gateway":"192.168.13.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.13.1","Mac":"84:0D:8E:77:97:40","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
23:33:08 RSL: stat/tasmota_779740/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_779740","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
23:33:08 RSL: stat/tasmota_779740/STATUS7 = {"StatusTIM":{"UTC":"2020-05-11T22:33:08","Local":"2020-05-11T23:33:08","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"05:13","Sunset":"20:20"}}
23:33:08 RSL: stat/tasmota_779740/STATUS10 = {"StatusSNS":{"Time":"2020-05-11T23:33:08"}}
23:33:08 RSL: stat/tasmota_779740/STATUS11 = {"StatusSTS":{"Time":"2020-05-11T23:33:08","Uptime":"0T00:01:40","UptimeSec":100,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"MqttCount":0,"POWER1":"ON","POWER2":"OFF","POWER3":"OFF","Wifi":{"AP":1,"SSId":"4u2","BSSId":"0C:80:63:46:79:ED","Channel":5,"RSSI":100,"Signal":-49,"LinkCount":1,"Downtime":"0T00:00:06"}}}
```
- [x] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
23:41:40 CMD: weblog 4
23:41:40 CMD: Group 0, Index 1, Command "WEBLOG", Data "4"
23:41:40 RSL: stat/tasmota_779740/RESULT = {"WebLog":4}
23:41:55 WIF: Checking connection...
23:42:15 WIF: Checking connection...
23:42:16 APP: Button3 multi-press 1
23:42:16 SRC: Button
23:42:16 RSL: stat/tasmota_779740/RESULT = {"POWER3":"ON"}
23:42:16 RSL: stat/tasmota_779740/POWER3 = ON
23:42:17 CFG: Saved to flash at F8, Count 12, Bytes 4096
23:42:35 WIF: Checking connection...
23:42:55 WIF: Checking connection...
23:43:14 APP: Button2 multi-press 1
23:43:15 WIF: Checking connection...
23:43:23 SRC: Button
23:43:23 RSL: stat/tasmota_779740/RESULT = {"POWER2":"ON"}
23:43:23 RSL: stat/tasmota_779740/POWER2 = ON
23:43:24 CFG: Saved to flash at F7, Count 13, Bytes 4096
```

### TO REPRODUCE
_Steps to reproduce the behavior:_

After restart ALL buttons POWER: OFF

[image]

Random POWER ON any button and time When i first click on button 
[image]

### EXPECTED BEHAVIOUR
I expected than all buttons continue **POWER: OFF** because i'm not enable button with click interface

### SCREENSHOTS
_Sonoff device._
[image]

When i connected device with MQTT ocorr  **{"TRIG": "TOGGLE"}**

[image]

effelle, 2020-05-12:
`{"TRIG": "TOGGLE"}` is there because you have HA Discovery enabled.
Try a different value for `ButtonDebounce` (default is 50) and enable more detailed log (weblog 4 - syslog 4) to help you find the issue.
Also, have a look on #5449 .

BrnEmilio, 2020-05-12:
> `{"TRIG": "TOGGLE"}` is there because you have HA Discovery enabled.
> Try a different value for `ButtonDebounce` (default is 50) and enable more detailed log (weblog 4 - syslog 4) to help you find the issue.
> Also, have a look on #5449 .

look on #5449 - Solution go back version firmware to 8.1 (Doris) ?

BrnEmilio, 2020-05-16:
Thanks guys, I install the new release 8.3 (Fred) and run the success without ghosts.

Note: Only installed and without update other configurations

closed 2020-05-16 by BrnEmilio