arendst/Tasmota #8421: Load 99 if DIMMER != 0

stefanbode, 2020-05-12:
### PROBLEM DESCRIPTION
I know that the PWM is operated by software on the ESP8266. Anyhow, I see 99 Load as soon as one PWM is != 0. Is this a normal behavior or do we have an issue here? My assumption is, that some functions like FADE IN/OUT will be negatively impacted if there is this high load. 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
PWM1 defined. Command: DIMMER 10
TELEPERIOD 10

### EXPECTED BEHAVIOUR
load << 99

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

stefanbode, 2020-05-12:
#7179

closed 2020-05-12 by stefanbode

s-hadinger, 2020-05-12:
PWM is software on ESP8266 and driven by interrupts. This is why Tasmota reports a high load, but there is nothing wrong with it. Fade has almost no impact on it.