arendst/Tasmota #8423: PWM Dimmer start 2 times causing some flicker at start

stefanbode, 2020-05-12:
### PROBLEM DESCRIPTION
If FADE=ON and SPEED=40 you can see on a DIMMER 20 command that the Fade process will start 2 times. One directly after the command and one additional after the RSL has been posted. I assure this is the issue that the dimmer works fine from 100->0% with a smooth fade out. But starts quite hard when 0-->100. Looks like it jumps from 0 to 15 and then smoothes up to 100.
```
10:05:48 delay_seconds 8, save_data_counter 1, fade.duration 6177
**Fade: 0 / 6177** - 
10:05:48 LightApplyFade 0 0 0 0 0
10:05:48 RSL: RESULT = {"POWER":"ON","Dimmer":50}
Fade: 17 / 6177 - Fade: 28 / 6177 - Fade: 38 / 6177 - 
10:05:48 delay_seconds 8, save_data_counter 8, fade.duration 6099
**Fade: 0 / 6099** - Fade: 9 / 6099 - Fade: 19 / 6099 - Fade: 30 / 6099 - Fade: 41 / 6099 - Fade: 51 / 6099 - Fade: 62 / 6099
```

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): GENERIC
- [ x] Tasmota binary firmware version number used: 8.2 latest
  - [ x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

stefanbode, 2020-05-12:
Need moe investgation...

closed 2020-05-12 by stefanbode

s-hadinger, 2020-05-12:
Please give more information: do you have rules? What are settings of PowerOnState? Does it happen after or before wifi connection?

Ideally give the whole log from startup to this point.