arendst/Tasmota #8426: NODEMCU restart with exception when using RfSend

byte4geek, 2020-05-12:
### PROBLEM DESCRIPTION
I try to send a code with RfSend but everytime that i send a command using the console, the device restarts with exception.

I using firmware version 8.2.0 on a nodeMCU with rx and tx modules 433Mhz.
this is the commands i use to send:
`cmnd/temp/rfsend {"RFSend":"0x41010C, 1, 340"}`
or
`cmnd/temp/rfsend {"RFSend":"0x41010C,"}`

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): NODEMCU
- [ ] Tasmota binary firmware version number used: 8.2.0
  - [ ] Pre-compiled
  - [x ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: IDE
- [x ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
13:18:33 CMD: Backlog Template; Module; GPIO 255
13:18:33 MQT: stat/temp/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
13:18:33 MQT: stat/temp/RESULT = {"Module":{"0":"Generic"}}
13:18:34 MQT: stat/temp/RESULT = {"GPIO0":{"1":"DHT11"},"GPIO1":{"0":"None"},"GPIO2":{"2":"AM2301"},"GPIO3":{"0":"None"},"GPIO4":{"106":"RFrecv"},"GPIO5":{"105":"RFSend"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
no rules present

```
- [ ] Provide the output of this command: `Status 0`:
```
 13:15:08 MQT: stat/temp/STATUS = {"Status":{"Module":0,"FriendlyName":["TEMP"],"Topic":"temp","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
13:15:08 MQT: stat/temp/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://192.168.1.2:9541/data/firmwares/tasmota.bin","RestartReason":"External System","Uptime":"0T00:21:07","StartupUTC":"2020-05-12T11:54:01","Sleep":50,"CfgHolder":4617,"BootCount":98,"BCResetTime":"2020-03-04T10:39:54","SaveCount":165,"SaveAddress":"F6000"}}
13:15:08 MQT: stat/temp/STATUS2 = {"StatusFWR":{"Version":"8.2.0(tasmota)","BuildDateTime":"2020-04-09T11:39:26","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"323/699"}}
13:15:08 MQT: stat/temp/STATUS3 = {"StatusLOG":{"SerialLog":4,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["HOME-NET",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
13:15:08 MQT: stat/temp/STATUS4 = {"StatusMEM":{"ProgramSize":576,"Free":424,"Heap":24,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","8FDAE397","043783A0","000000CF","010013C0","C000F981","00000004"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,17,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,3,4,5,6"}}
13:15:08 MQT: stat/temp/STATUS5 = {"StatusNET":{"Hostname":"temp-5928","IPAddress":"192.168.1.21","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.2","Mac":"60:01:94:88:97:28","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
13:15:08 MQT: stat/temp/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.2","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_889728","MqttUser":"mqttuser","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
13:15:08 MQT: stat/temp/STATUS7 = {"StatusTIM":{"UTC":"2020-05-12T12:15:08","Local":"2020-05-12T13:15:08","StartDST":"2020-03-29T01:00:00","EndDST":"2020-10-25T02:00:00","Timezone":"+01:00","Sunrise":"05:11","Sunset":"20:21"}}
13:15:08 MQT: stat/temp/STATUS10 = {"StatusSNS":{"Time":"2020-05-12T13:15:08","DHT11-00":{"Temperature":null,"Humidity":null,"DewPoint":0.0},"AM2301-02":{"Temperature":null,"Humidity":null,"DewPoint":0.0},"TempUnit":"C"}}
13:15:08 MQT: stat/temp/STATUS11 = {"StatusSTS":{"Time":"2020-05-12T13:15:08","Uptime":"0T00:21:07","UptimeSec":1267,"Vcc":3.471,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"HOME-NET","BSSId":"DC:39:6F:AB:C2:02","Channel":6,"RSSI":44,"Signal":-78,"LinkCount":1,"Downtime":"0T00:00:06"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
 00:00:00 CFG: Loaded from flash at F6, Count 165
00:00:00 Project tasmota TEMP Version 8.2.0(tasmota)-STAGE
00:00:00 WIF: Connecting to AP1 HOME-NET in mode 11N as temp-5928...
00:00:05 WIF: Connected
00:00:05 HTP: Web server active on temp-5928 with IP address 192.168.1.21
00:00:05 RSL: tele/temp/RESULT = {"Time":"1970-01-01T00:00:05","RfReceived":{"Data":"0x14000C","Bits":24,"Protocol":1,"Pulse":340}}
13:16:09 MQT: Attempting connection...
13:16:09 MQT: Connected
13:16:09 MQT: tele/temp/LWT = Online (retained)
13:16:09 MQT: cmnd/temp/POWER = 
13:16:09 MQT: tele/temp/INFO1 = {"Module":"Generic","Version":"8.2.0(tasmota)","FallbackTopic":"cmnd/DVES_889728_fb/","GroupTopic":"cmnd/tasmotas/"}
13:16:09 MQT: tele/temp/INFO2 = {"WebServerMode":"Admin","Hostname":"temp-5928","IPAddress":"192.168.1.21"}
13:16:09 MQT: tele/temp/INFO3 = {"RestartReason":{"Exception":28,"Reason":"Exception","EPC":["402481b4","00000000","00000000"],"EXCVADDR":"00000000","DEPC":"00000000","CallChain":["40214d48","40280fa4","40278cb0","40248300","402153bd","4010007d","40246318","4000050c","40221c98","4020d9e1","4027aa9c","40248494","40248508","4027a65d","4027a65d","40278cb8","40278cb8","4027a62c","40278cac","40278cb0","40202cd2","40278cac","4022ce96","4027da20","402221c9","4020d9e1","4021f71a","40101424","4000050c","4021f7b7","4021f7b5"]}}
13:16:10 MQT: tele/temp/RESULT = {"Time":"2020-05-12T13:16:10","RfReceived":{"Data":"0x14000C","Bits":24,"Protocol":1,"Pulse":341}}
13:16:13 MQT: tele/temp/STATE = {"Time":"2020-05-12T13:16:13","Uptime":"0T00:00:12","UptimeSec":12,"Vcc":3.472,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"HOME-NET","BSSId":"DC:39:6F:AB:C2:02","Channel":6,"RSSI":40,"Signal":-80,"LinkCount":1,"Downtime":"0T00:00:06"}}
13:16:13 MQT: tele/temp/SENSOR = {"Time":"2020-05-12T13:16:13","DHT11-00":{"Temperature":null,"Humidity":null,"DewPoint":0.0},"AM2301-02":{"Temperature":null,"Humidity":null,"DewPoint":0.0},"TempUnit":"C"}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
send the commands:

`cmnd/temp/rfsend {"RFSend":"0x41010C, 1, 340"}`
or
`cmnd/temp/rfsend {"RFSend":"0x41010C,"}`

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
none

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
no screenshot

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
none

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2020-05-13:
Are you sure that `cmnd/temp/rfsend {"RFSend":"0x41010C, 1, 340"}`is the correct format? The driver xdrv_17_rcswitch.ino has comments about a different format: 
```
// RFsend {"data":0x501014,"bits":24,"protocol":1,"repeat":10,"pulse":350}
// RFsend data, bits, protocol, repeat, pulse
```

meingraham, 2020-05-13:
... and when this is determined, adding the RFSend command syntax to the docs would be nice :wink:

byte4geek, 2020-05-14:
ok when i using this  command:
cmnd/temp/RFsend {"data":0x501014,"bits":24,"protocol":1,"repeat":10,"pulse":350}

it works fine, the problem is that i can't find this command in any page on the internet, instert this command to the documentation maybe very usefull.

thanks a lot

closed 2020-05-14 by byte4geek