arendst/Tasmota #8429: BH1750 Not Initialized on Boot

agmlego, 2020-05-12:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
It appears that with the latest development branch (af32b3f5e58b4af00124ed6a8d8bf413e36561c0), the BH1750 sensor is no longer initialized to a resolution. As such, it reports 0lux indefinitely until either something else on the bus (I used my excamera i2cdriver) does so, or a command is send to the Tasmota devic

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): _D1 Mini_
- [x] Tasmota binary firmware version number used: _8.2.0.6_
  - [ ] Pre-compiled
  - [x] Self-compiled
    - [x] IDE / Compiler used: _VSCode/PlatformIO_
- [x] Flashing tools used: _PlatformIO_
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
18:25:27 MQT: stat/garden_17F3FB/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
18:25:28 MQT: stat/garden_17F3FB/RESULT = {"Module":{"0":"Generic"}}
18:25:28 MQT: stat/garden_17F3FB/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"6":"I2C SDA"},"GPIO5":{"5":"I2C SCL"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
18:26:00 MQT: stat/garden_17F3FB/STATUS = {"Status":{"Module":0,"FriendlyName":["Garden Cloche"],"Topic":"garden_17F3FB","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
18:26:00 MQT: stat/garden_17F3FB/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"External System","Uptime":"0T00:16:59","StartupUTC":"2020-05-12T17:09:01","Sleep":50,"CfgHolder":4617,"BootCount":12,"BCResetTime":"2020-05-12T16:30:02","SaveCount":17,"SaveAddress":"FB000"}}
18:26:00 MQT: stat/garden_17F3FB/STATUS2 = {"StatusFWR":{"Version":"8.2.0.6(tasmota)","BuildDateTime":"2020-05-12T11:28:43","Boot":6,"Core":"2_7_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"366/699"}}
18:26:00 MQT: stat/garden_17F3FB/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["internetofshit",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
18:26:00 MQT: stat/garden_17F3FB/STATUS4 = {"StatusMEM":{"ProgramSize":514,"Free":488,"Heap":24,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164020","FlashMode":3,"Features":["00000809","8FC82597","040003A1","0000144D","210013C0","40000800","00000020"],"Drivers":"1,2,3,4,8,9,10,12,29","Sensors":"1,2,3,5,9,10,48"}}
18:26:00 MQT: stat/garden_17F3FB/STATUS5 = {"StatusNET":{"Hostname":"garden_17F3FB-5115","IPAddress":"10.11.12.171","Gateway":"10.11.12.1","Subnetmask":"255.255.255.0","DNSServer":"10.11.12.130","Mac":"DC:4F:22:17:F3:FB","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
18:26:00 MQT: stat/garden_17F3FB/STATUS6 = {"StatusMQT":{"MqttHost":"10.11.12.3","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_17F3FB","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
18:26:00 MQT: stat/garden_17F3FB/STATUS7 = {"StatusTIM":{"UTC":"2020-05-12T17:26:00","Local":"2020-05-12T18:26:00","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"11:13","Sunset":"01:44"}}
18:26:00 MQT: stat/garden_17F3FB/STATUS10 = {"StatusSNS":{"Time":"2020-05-12T18:26:00","BH1750":{"Illuminance":0}}}
18:26:00 MQT: stat/garden_17F3FB/STATUS11 = {"StatusSTS":{"Time":"2020-05-12T18:26:00","Uptime":"0T00:16:59","UptimeSec":1019,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"internetofshit","BSSId":"1A:E8:29:51:AB:8D","Channel":8,"RSSI":82,"Signal":-59,"LinkCount":1,"Downtime":"0T00:00:04"}}}
```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Attach a BH1750 sensor to a Tasmota device, and power it up.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
I expect to see lux readings displayed regularly in the web UI, as well as transmitted on MQTT.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

device111, 2020-05-12:
so here is working properly with the actual dev.

ascillato2, 2020-05-16:
@agmlego 

Hi, please can you re test with latest development version? Can you recheck your wirings? Thanks

agmlego, 2020-05-18:
@ascillato2 
I just pulled and rebuilt the project from the development branch HEAD (9d15381f), and the problem persists.

My wiring is correct, as I can issue the `Sensor10 1` command and immediately begin receiving data from the sensor.

This applied to 10 sensors that I have locally.

arendst, 2020-05-18:
Do you always have to use command ``Sensor10 1`` or is it only once?

What happens if you restart your device?

agmlego, 2020-05-18:
I have to run that command every time the bus inits. This includes:

- Restarting the ESP
- Hotswapping sensors
- Power cycling the board

arendst, 2020-05-18:
Are you using more I2C devices on the BH1750 bus?

agmlego, 2020-05-18:
Yes, there is a BMP280 and a CHIRP sensor. The problem persists even with those removed however.

arendst, 2020-05-18:
There you have it! 

The chirp is NOT a real I2C device and it breaks the I2C bus protocol. It is not included in the tasmota-sensors.bin version for the same reason.

I cannot duplicate your issue on a single BH1750 device connected to a wemos.

So you will need to use the workaround of ``Sensor10 1`` to get some reading. Although I guess ``Sensor10 0`` is better as it doesn't change the read value.

agmlego, 2020-05-18:
OK, but the system does the same thing without the CHIRP sensor connected at all.

arendst, 2020-05-18:
And what happens with a single BH1750?

agmlego, 2020-05-18:
With a single BH1750 connected and absolutely nothing else on the bus, the problem persists.

arendst, 2020-05-18:
Well in that case I cannot reproduce your issue and cannot help. Sorry.

agmlego, 2020-05-18:
Looks like disabling CHIRP support from the build entirely resolves the problem.

agmlego, 2020-05-18:
So perhaps there is a conflict between the CHIRP module and the BH1750 module.

I guess I can use a rule to init the BH1750 on boot. That seems inelegant though.

arendst, 2020-05-18:
See this 

> The chirp is NOT a real I2C device and it breaks the I2C bus protocol. It is not included in the tasmota-sensors.bin version for the same reason.

So yes, it breaks the bus even when not connected as it changes I2C parameters while searching for devices.

A rule sounds like a good workaround hoping the real I2C devices stay sound.

closed 2020-05-18 by agmlego