arendst/Tasmota #8433: Integration of Victron Energy components

mandeltuete, 2020-05-13:
Hello everybody,i have been using tasmota for a while now and would like to thank you for this great system.
I would like to connect my Victron Energy components to the home automation system via tasmota and need a suitable driver. Victron Energy is a great manufacturer of solar components and offers a serial interface for reading out operating data on many devices. The interface uses the VE.Direct protocol. Unfortunately, we lack the necessary programming knowledge to develop a driver ourselves.
There are already two projects that are working on reading data from Victron devices via Arduino / ESP8266. It may be possible to reuse parts of this code.
I am quite sure that many other users would be happy about a suitable driver.

Links to the projects:

https://github.com/physee/Victron.Arduino-ESP8266
https://github.com/winginitau/VictronVEDirectArduino

VE.Direct Protocol
https://www.victronenergy.com/live/vedirect_protocol:faq

**Have you looked for this feature in other issues and in the docs?**  

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

rg2k, 2020-05-22:
Did you look at the MQTT option? You will need a Victron Color Controller (or a Raspberry Pi), but this works really well for integration to most home automation systems.

mandeltuete, 2020-05-22:
Yeah, I've been testing it for over a year. But the power consumption of the Raspberry Pi is too high and the Victron firmware is releatively unstable. During this time I had to reflash the SD card several times and bought different cards. I am also annoyed about the purchase of the expensive VE.Direct to USB cable (30€ / pcs.). Since I am using an ESP8266 with Tasmota on my solar system to record weather data anyway, connecting the Victron components would be very practical.

ascillato2, 2020-06-12:
Closing this issue as nobody is working on it. Sorry.

If anyone is interested on working on this, please just ask to reopen. Thanks.

closed 2020-06-12 by ascillato2

Aikawa24, 2023-04-13:
hi, Fabian Lauer made a Script to use the dbus.
ive got a esp32 with tasiota to read my smart meter data and get this into the Cerbo GX

https://github.com/fabian-lauer/dbus-shelly-3em-smartmeter

you can modify the script to get JSON data from Tasmota and integrate it to the Victron system

primemaster, 2023-04-20:
Hello all,

I was working the last couple of days on Victron VE.Direct serial RX support in Tastmota. The implementation is based on: https://www.victronenergy.com/upload/documents/VE.Direct-Protocol-3.32.pdf 

### HTML
The Tasmota main page currently looks like:
[image]

### JSON
`curl -s 'http://<IP>/cm?cmnd=status%2010'`

`{"StatusSNS":{"Time":"2023-04-20T21:55:23","VictronVE.Direct":[{"field": "PID","unit": "","value": "0xA053","description": "Product ID"},{"field": "FW","unit": "","value": "161","description": "Firmware version (16 bit)"},{"field": "SER#","unit": "","value": "<REDACTED>","description": "Serial number"},{"field": "V","unit": "mV","value": "26460","description": "Main or channel 1 (battery) voltage"},{"field": "I","unit": "mA","value": "-220","description": "Main or channel 1 battery current"},{"field": "VPV","unit": "mV","value": "40800","description": "Panel voltage"},{"field": "PPV","unit": "W","value": "3","description": "Panel power"},{"field": "CS","unit": "","value": "3","description": "State of operation"},{"field": "MPPT","unit": "","value": "2","description": "Tracker operation mode"},{"field": "OR","unit": "","value": "0x00000000","description": "Off reason"},{"field": "ERR","unit": "","value": "0","description": "Error code"},{"field": "LOAD","unit": "","value": "ON","description": "Load output state (ON/OFF)"},{"field": "IL","unit": "mA","value": "300","description": "Load current"},{"field": "H19","unit": "0.01 kWh","value": "2530","description": "Yield total (user resettable counter)"},{"field": "H20","unit": "0.01 kWh","value": "54","description": "Yield today"},{"field": "H21","unit": "W","value": "258","description": "Maximum power today"},{"field": "H22","unit": "0.01 kWh","value": "30","description": "Yield yesterday"},{"field": "H23","unit": "W","value": "238","description": "Maximum power yesterday"},{"field": "HSDS","unit": "","value": "218","description": "Day sequence number (0..364)"},{"Checksum": "0x4e"}],"bytes_read": 622779,"MetaTS": {"Start_TS": 3083017, "End_TS": 3083033, "Delta": 16}}}`

Before I create a PR, I'd like to ask:
- Is the sensor folder actually the correct place for a Victron device? Current filename: `tasmota/tasmota_xsns_sensor/xsns_107_victron.ino`
- The HTML table has four columns `<tr>  <th>%s</th> <td></td> <td>%s</td> <td></td> <td>%s</td> <td></td> <td>%s</td>  </tr>`, is that acceptable? I do not see the possibility for complex tables using the {e}, {s} and {m} 

Any suggestions and thoughts?

arendst, 2023-04-21:
As you seem to write a book I see that the current column count is an issue.

Also considering the amount of data you feel needs to be provided I suggest to stay using a sensor driver so most people can easily ignore the driver and some victron users can enable it for their use.

BTW the gui was never meant for providing this amount of data. Your backend should handle this.

primemaster, 2023-04-21:
Thanks @arendst for you reply.

In the HTML view, the left side
[image]
is what is actually transferred on the serial line. The right side 
[image]
is then translated via a look-up table in the code:
```
typedef struct {
    const char *label;
    const char *unit;
    const char *description;
} Tripple;

static const Tripple field_metadata[] = { 
  {"V",       "mV",        "Main or channel 1 (battery) voltage"},
  {"V2",      "mV",        "Channel 2 (battery) voltage"},
  {"V3",       "mV",       "Channel 3 (battery) voltage"},
  {"VS",       "mV",       "Auxiliary (starter) voltage"},
  {"VM",       "mV",       "Mid-point voltage of the battery bank"},
  {"DM",       "%%",       "Mid-point deviation of the battery bank"},
  {"VPV",      "mV",       "Panel voltage"},
  {"PPV",      "W",        "Panel power"},
  {"I",        "mA",       "Main or channel 1 battery current"},
  {"I2",       "mA",       "Channel 2 battery current"},
  {"I3",       "mA",       "Channel 3 battery current"},
  {"IL",       "mA",       "Load current"},
  {"LOAD",     "",         "Load output state (ON/OFF)"},
  {"T",        "°C",       "Battery temperature"},
  {"P",        "W",        "Instantaneous power"},
  {"CE",       "mAh",      "Consumed Amp Hours"},
  {"SOC",      "%%",       "State-of-charge"},
  {"TTG",      "Minutes",  "Time-to-go"},
  {"Alarm",    "",         "Alarm condition active"},
  {"Relay",    "",         "Relay state"},
  {"AR",       "",         "Alarm reason"},
  {"OR",       "",         "Off reason"},
  {"H1",       "mAh",      "Depth of the deepest discharge"},
  {"H2",       "mAh",      "Depth of the last discharge"},
  {"H3",       "mAh",      "Depth of the average discharge"},
  {"H4",       "",         "Number of charge cycles"},
  {"H5",       "",         "Number of full discharges"},
  {"H6",       "mAh",      "Cumulative Amp Hours drawn"},
  {"H7",       "mV",       "Minimum main (battery) voltage"},
  {"H8",       "mV",       "Maximum main (battery) voltage"},
  {"H9",       "Seconds",  "Number of seconds since last full charge"},
  {"H10",      "",         "Number of automatic synchronizations"},
  {"H11",      "",         "Number of low main voltage alarms"},
  {"H12",      "",         "Number of high main voltage alarms"},
  {"H13",      "",         "Number of low auxiliary voltage alarms"},
  {"H14",      "",         "Number of high auxiliary voltage alarms"},
  {"H15",      "mV",       "Minimum auxiliary (battery) voltage"},
  {"H16",      "mV",       "Maximum auxiliary (battery) voltage"},
  {"H17",      "0.01 kWh", "Amount of discharged energy (BMV) / Amount of produced energy (DC monitor)"},
  {"H18",      "0.01 kWh", "Amount of charged energy (BMV) / Amount of consumed energy (DC monitor)"},
  {"H19",      "0.01 kWh", "Yield total (user resettable counter)"},
  {"H20",      "0.01 kWh", "Yield today"},
  {"H21",      "W",        "Maximum power today"},
  {"H22",      "0.01 kWh", "Yield yesterday"},
  {"H23",      "W",        "Maximum power yesterday"},
  {"ERR",      "",         "Error code"},
  {"CS",       "",         "State of operation"},
  {"BMV",      "",         "Model description (deprecated)"},
  {"FW",       "",         "Firmware version (16 bit)"},
  {"FWE",      "",         "Firmware version (24 bit)"},
  {"PID",      "",         "Product ID"},
  {"SER#",     "",         "Serial number"},
  {"HSDS",     "",         "Day sequence number (0..364)"},
  {"MODE",     "",         "Device mode"},
  {"AC_OUT_V", "0.01 V",   "AC output voltage"},
  {"AC_OUT_I", "0.1 A",    "AC output current"},
  {"AC_OUT_S", "VA",       "AC output apparent power"},
  {"WARN",     "",         "Warning reason"},
  {"MPPT",     "",         "Tracker operation mode"},
  {"MON",      "",         "DC monitor mode"}
};
```

This table is from the Victron documentation:
[image]

> BTW the gui was never meant for providing this amount of data. Your backend should handle this.

Does that mean, that the HTML view should only show the two left most columns (which are actually coming from the serial line)?

arendst, 2023-04-21:
Add your intelligence to it like supplying the correct kWh value (multiplied by 0.01) and add the correct unit in the unit column. Let users use command energyres to show their desired resolution. Give those three letter words user recognisable short names. Who cares what the serial output looks like.

primemaster, 2023-04-23:
Adding some functionality to compute the final result for the HTML view makes sens. Also to add meaningful names instead of the Victron labels.
Regarding the desired resolution I do not fully understand what there should be done. I found
` EnergyFmt()` and `Settings->flag2.energy_resolution` in some other places of the Tasmota code. You mean to use such kind of formatting functions? If so, where/how can the user dynamically set the value of that flag?

arendst, 2023-04-23:
> If so, where/how can the user dynamically set the value of that flag?

Look for commands ``VoltRes``, ``AmpRes``, ``WattRes``, ``EnergyRes`` all available if an energy driver is active. As you'll be writing a dedicated sensor driver I suggest you take a look at driver ``xdrv_03_energy.ino`` how to write your own user support.

As an alternative you might want to NOT write a sensor driver but instead write an energy driver (``xnrg_25_victron.ino``) which is serviced by the energy driver. It will probably need some default values for voltage, current and power but it can easily be extended with your abundance of register values. In that case have a look at the ``xnrg_08_sdm120.ino`` driver which adds some registers too.

Bersaker, 2023-07-14:
Sounds interesting... Have you made any progress on this topic?

primemaster, 2023-10-25:
Hello @Bersaker,
I did not really continue on the implementation. 
As I have now a second Victron solar controller, I was thinking about adding support for multiple (2-4) controller to the same ESP8266. GND of the UART is the same as BATT-. Which means it is electrically possible if the controllers share the same BATT-. Of course that would mean up to 4 soft serials at 19200 baud. Where I don't know if that would be feasible. Especially as the each controller outputs its status every second taking ~100ms per controller. Which mean that the intervals might collide.
Additionally the HTML UI would be a mess.

Any thoughts on supporting multiple controllers?

primemaster, 2023-11-01:
> Hello @Bersaker, I did not really continue on the implementation. As I have now a second Victron solar controller, I was thinking about adding support for multiple (2-4) controller to the same [...]

@arendst : Any thoughts on supporting multiple (up to e.g. 4) Victron controllers with one Tasmota instance?

pbg42, 2023-11-15:
Hello @primemaster,
I have two Victron devices
- Blue Solar MPPT 75 I 15 solar charger
- Phoenix 12/250 converter

for an off-grid PV power supply and I want to log their data into my iobroker using MQTT. I already had a python script for a RASPI, but that is total overkill of hardware, so I want to use a cheap 5€ ESP01 board for each device to transfer the data with MQTT. I am very interested in your implementation, so that I can either complete it / adopt it for the Phoenix converter. I don´t think a multiple channel solution is required ( for me), I prefer 1 device for one task. Are you willing to share your current solution?

primemaster, 2023-11-20:
Hello @pbg42,

as I haven't worked on it in some month, I don't know in what shape the code is :-).
As there was no suggestion yet by @arendst if support for multiple VE.Direct input would be appreciated, I was planning to rewrite the code to support multiple input UARTs however limit to one connection by default.

pbg42, 2023-11-20:
Hi Micha,
I don't care about the shape of the code; as it is already an almost working version,  it shows at least which part of the project you made the extensions / modifications. So much much better than starting from scratch alone. 
So, I would be very happy I you could share your code, I won#t bother you with questions if you are busy. Or, do you think you will be done with rewriting within the next few weeks?

primemaster, 2023-12-03:
Hi @pbg42 

> Hi Micha, I don't care about the shape of the code; as it is already an almost working version, it shows at least which part of the project you made the extensions / modifications. So much much [...]

I pushed my current state to https://github.com/micha-m/Tasmota_VE_PoC/commits/master . Please note:
- Code is far away from a PR here
- The Tasmota code base was forked in April 2023 (needs rebase!)
- HTML view is still not in a nice table (not sure if that is actually possible)
- JSON / MQTT is not tested
- Not tested on ESP32
- Seems to keep up with (at least) two software serial inputs => Two controllers would be supported
- Prepared for up to 4 controllers
- Not using any of those fancy "energy" formatting functions (code size increase significant in my opinion)
- Still a lot of debugging prints included
- More cleanup (e.g. removing initial single serial support code) needed

I am happy to receive constructive feedback!

pbg42, 2023-12-10:
Thank you @primemaster , 
I will try to understand, learn and see how I can read and control my two devices based on that. I will give feedback, but don't expect before christmas - too busy with other things.
Thanks a lot for sharing! :+1:

pin32, 2024-03-16:
Hi, has this job made any progress?

pbg42, 2024-03-19:
Kind of :-) 
in this fork https://github.com/pbg42/Tasmota_VE_PoC I added/modified the fork of micha-m:
* support for up to 4 victron devices (tested with my 2 owned + 2 simulated by doubling the inputs)
* formatted output in SI units in web GUI and MQTT
* transfered the table data to PROGMEM to reduce RAM size
* prepared reading of HEX protocol and command interface, but this is not functional yet

what is still missing: 
* rebase to current development branch (it is still based on April 2023 fork from Micha-m)
* HEX protocol command trigger and control
* documentation of interface in the tasmota docs

so, currently far away from a pull request to tasmota development.

nulleinspeisung, 2024-04-15:
hi i would like to test it. I have a victron 100/20 mppt and would like to read data via tasmota. Furthermore i would like to send data via mqtt by global variable. I am not sure how to start. Do i have to change the firmware of tasmota?

3ricj, 2024-09-24:
HI there folks.  

I wanted to let you know I started over with this concept using berry script.  This is my first time writing berry (gosh, the documentation is terrible), so I welcome feedback.  

It operates on esp32.   It will parse the full VE-direct protocol for display reasons, and shows the following (it's currently dark here, but you get the idea):

[image]

It will transmit the untampered with values over MQTT to remain compatible with the VE-direct protocol to make it clear for consumers of the feed.  Almost all values are sent as INTs without quotes, but there are a few fields which are currently quoted -- not sure what the best practice is here for sending ascii hex values such as below: 

`
    "IL": 0,
    "OR": "0x00000001",
    "V": 12520,
    "PID": "0xA075",
    "CS": 0
`

You can download the code from here: 
https://github.com/3ricj/VictronTasmota

Please let me know if it works for you! 

If I can figure out how to, I will add support for the checksum for data validation. I can't find any repo for berry scripts which isn't dead, but if I can find it I'll try to get this code added to it.

sfromis, 2024-09-26:
If your question about "ascii hex values" is about what you can put into JSON, there is not much to do there, as JSON **only** support values as decimal numbers, strings, or boolean. Hence, numbers you want to express in hex format can only be a quoted string; alternatively, you can include it as decimal.

Checksum handling would usually be at byte level, where the `bytes` class in Berry is useful for treating data as a series of bytes, which you can sum (or however the algorithm) for checksum operations. Of course, not knowing how the data looks, I have no specific suggestion.

cundc1989, 2024-10-05:
Hi 3ricj,
thanks for your great work. After some tries, i got it running.
You are my Hero, today😉

[image]

Still another question. Ist there any posibility to get the rigth description into Tasmotrol ?

[image]

Thanks Chris

corbolais, 2025-02-03:
> HI there folks.
>
> I wanted to let you know I started over with this concept using berry script. This is my first time writing berry (gosh, the documentation is terrible), so I welcome feedback.
>
> It [...]

Thank you for the berry script! I think I'll go with that as opposed to the drivers. Easy to modify on the fly, too. :)