arendst/Tasmota #8437: IRHVAC bug

3ricj, 2020-05-14:
### PROBLEM DESCRIPTION
When I send a command with Celsius=Off, it sends Celsius=On anyway. 

For example:   

02:25:05 CMD: IRhvac {"Vendor":"Mitsubishi_Heavy_152", "Power":"On","Mode":"Hot","FanSpeed":3,"Celsius":"Off","Temp":68}
02:25:05 MQT: stat/tasmota_6A9E68/RESULT = {"IRHVAC":{"Vendor":"MITSUBISHI_HEAVY_152","Model":1,"Power":"On","Mode":"Auto","Celsius":"Off","Temp":68,"FanSpeed":"Medium","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}
02:25:05 MQT: tele/tasmota_6A9E68/RESULT = {"IrReceived":{"Protocol":"MITSUBISHI_HEAVY_152","Bits":152,"Data":"0x0xAD513CE51A08F703FC02FDC03F08F700FF807F","Repeat":0,"IRHVAC":{"Vendor":"MITSUBISHI_HEAVY_152","Model":-1,"Power":"On","Mode":"Auto","Celsius":"On","Temp":20,"FanSpeed":"Medium","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}}

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): nas-ir03w0, using the NAS-IR03W0 IR bridge template:   https://templates.blakadder.com/neo_coolcam_NAS-IR03W0.html
- [x] Tasmota binary firmware version number used: 8.2.0(tasmota)
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: _____
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

02:28:12 CMD: Backlog Template; Module; GPIO 255
02:28:12 MQT: stat/tasmota_6A9E68/RESULT = {"NAME":"Neo Coolcam IR","GPIO":[0,0,0,0,56,51,0,0,0,17,8,0,0],"FLAG":0,"BASE":62}
02:28:12 MQT: stat/tasmota_6A9E68/RESULT = {"Module":{"0":"Neo Coolcam IR"}}
02:28:12 MQT: stat/tasmota_6A9E68/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"56":"Led1i"},"GPIO5":{"51":"IRrecv"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"17":"Button1"},"GPIO14":{"8":"IRsend"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
02:28:45 MQT: stat/tasmota_6A9E68/STATUS = {"Status":{"Module":0,"FriendlyName":["IR1"],"Topic":"tasmota_6A9E68","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":1}}
02:28:45 MQT: stat/tasmota_6A9E68/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T01:41:50","StartupUTC":"2020-05-13T23:46:55","Sleep":50,"CfgHolder":4617,"BootCount":7,"BCResetTime":"2020-05-13T05:48:10","SaveCount":19,"SaveAddress":"F9000"}}
02:28:45 MQT: stat/tasmota_6A9E68/STATUS2 = {"StatusFWR":{"Version":"8.2.0(tasmota)","BuildDateTime":"2020-03-20T14:50:47","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"380/699"}}
02:28:45 MQT: stat/tasmota_6A9E68/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["publicn3rdarea",""],"TelePeriod":60,"Resolution":"558180C0","SetOption":["00008029","2805C8000100060000005A00000000000000","00000000","00000000"]}}
02:28:45 MQT: stat/tasmota_6A9E68/STATUS4 = {"StatusMEM":{"ProgramSize":674,"Free":328,"Heap":18,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"146085","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","000000CD","010013C0","C000F981","00000004"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
02:28:45 MQT: stat/tasmota_6A9E68/STATUS5 = {"StatusNET":{"Hostname":"tasmota_6A9E68-7784","IPAddress":"192.168.0.61","Gateway":"192.168.0.15","Subnetmask":"255.255.255.0","DNSServer":"8.8.8.8","Mac":"BC:DD:C2:6A:9E:68","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
02:28:45 MQT: stat/tasmota_6A9E68/STATUS6 = {"StatusMQT":{"MqttHost":"mqtt.cubesearch.com","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_6A9E68","MqttUser":"sonoff","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
02:28:45 MQT: stat/tasmota_6A9E68/STATUS7 = {"StatusTIM":{"UTC":"2020-05-14T01:28:45","Local":"2020-05-14T02:28:45","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"05:09","Sunset":"20:24"}}
02:28:45 MQT: stat/tasmota_6A9E68/STATUS10 = {"StatusSNS":{"Time":"2020-05-14T02:28:45"}}
02:28:45 MQT: stat/tasmota_6A9E68/STATUS11 = {"StatusSTS":{"Time":"2020-05-14T02:28:45","Uptime":"0T01:41:50","UptimeSec":6110,"Heap":18,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":23,"MqttCount":1,"Wifi":{"AP":1,"SSId":"publicn3rdarea","BSSId":"B0:B9:8A:5C:31:69","Channel":8,"RSSI":100,"Signal":-29,"LinkCount":1,"Downtime":"0T00:00:06"}}}

```
- [x] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
02:29:07 CFG: Saved to flash at F8, Count 20, Bytes 4096
02:29:08 MQT: tele/tasmota_6A9E68/STATE = {"Time":"2020-05-14T02:29:08","Uptime":"0T01:42:13","UptimeSec":6133,"Heap":21,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":23,"MqttCount":1,"Wifi":{"AP":1,"SSId":"publicn3rdarea","BSSId":"B0:B9:8A:5C:31:69","Channel":8,"RSSI":100,"Signal":-27,"LinkCount":1,"Downtime":"0T00:00:06"}}
02:29:20 CMD: IRhvac {"Vendor":"Mitsubishi_Heavy_152", "Power":"On","Mode":"Hot","FanSpeed":3,"Celsius":"Off","Temp":68}
02:29:20 SRC: WebConsole from 192.168.0.127
02:29:20 CMD: Group 0, Index 1, Command "IRHVAC", Data "{"Vendor":"Mitsubishi_Heavy_152", "Power":"On","Mode":"Hot","FanSpeed":3,"Celsius":"Off","Temp":68}"
02:29:20 MQT: stat/tasmota_6A9E68/RESULT = {"IRHVAC":{"Vendor":"MITSUBISHI_HEAVY_152","Model":1,"Power":"On","Mode":"Auto","Celsius":"Off","Temp":68,"FanSpeed":"Medium","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}
02:29:20 MQT: tele/tasmota_6A9E68/RESULT = {"IrReceived":{"Protocol":"MITSUBISHI_HEAVY_152","Bits":152,"Data":"0x0xAD513CE51A08F703FC02FDC03F08F700FF807F","Repeat":0,"IRHVAC":{"Vendor":"MITSUBISHI_HEAVY_152","Model":-1,"Power":"On","Mode":"Auto","Celsius":"On","Temp":20,"FanSpeed":"Medium","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}}
02:29:25 WIF: Checking connection...

Example 2: 

02:30:09 CMD: Group 0, Index 1, Command "IRHVAC", Data "{"Vendor":"GREE", "Power":"On","Mode":"Hot","FanSpeed":3,"Celsius":"Off","Temp":68}"
02:30:10 MQT: stat/tasmota_6A9E68/RESULT = {"IRHVAC":{"Vendor":"GREE","Model":1,"Power":"On","Mode":"Auto","Celsius":"Off","Temp":68,"FanSpeed":"Medium","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}
02:30:10 MQT: tele/tasmota_6A9E68/RESULT = {"IrReceived":{"Protocol":"GREE","Bits":64,"Data":"0x0x28094050002000D0","Repeat":0,"IRHVAC":{"Vendor":"GREE","Model":1,"Power":"On","Mode":"Auto","Celsius":"On","Temp":25,"FanSpeed":"Medium","SwingV":"Auto","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}}
02:30:25 WIF: Checking connection...

```

### TO REPRODUCE
Send any command with Celsius Off, and it actually sends the command with Celsius On.  This causes the display on the units to switch to Celsius.  

### EXPECTED BEHAVIOUR
I would expect it would send the command not in Celsius if told not to do so.  This is what the remotes are doing. 

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2020-05-14:
The IR encoding is managed by this lib: https://github.com/crankyoldgit/IRremoteESP8266

With `MITSUBISHI_HEAVY_152` the Celsius is indeed hardcoded. It means that either the IR lib does not support Celsius off, or the HVAC does not support it.

See https://github.com/crankyoldgit/IRremoteESP8266/blob/master/src/ir_MitsubishiHeavy.cpp

```
stdAc::state_t IRMitsubishiHeavy88Ac::toCommon(void) {
  stdAc::state_t result;
  result.protocol = decode_type_t::MITSUBISHI_HEAVY_88;
  result.model = -1;  // No models used.
  result.power = this->getPower();
  result.mode = IRMitsubishiHeavy152Ac::toCommonMode(this->getMode());
  result.celsius = true;
  result.degrees = this->getTemp();
  result.fanspeed = this->toCommonFanSpeed(this->getFan());
  result.swingv = this->toCommonSwingV(this->getSwingVertical());
  result.swingh = this->toCommonSwingH(this->getSwingHorizontal());
  result.turbo = this->getTurbo();
  result.econo = this->getEcono();
  result.clean = this->getClean();
  // Not supported.
  result.quiet = false;
  result.filter = false;
  result.light = false;
  result.beep = false;
  result.sleep = -1;
  result.clock = -1;
  return result;
}
```

3ricj, 2020-05-14:
It does seem that this is an upstream issue; should I file a report on IRemoteESP8266?

s-hadinger, 2020-05-14:
Yes

3ricj, 2020-05-14:
I actually use a GREE AC  -- I can tell you that with GREE, it does support Fahrenheit in the protocol, remote and AC unit.  I only included the MITSUBISHI_HEAVY_152 as that seems to be used quite a bit in the documentation as a reference unit, so I thought it might be helpful, but I don't have any first hand experience with the protocol.  

For my GREE unit, if I turn the unit on with my remote, it shows degrees in F correctly on the of the unit.  However, when I point it at the tasmota device, it decodes it as C.  That's a related (but maybe different) issue.  

If I have my remote pointed at the Tasmota device AND my AC unit, and power it on, the AC unit displays 68F, but the tasmota device shows 20c.  

18:01:56 MQT: tele/tasmota_6A9E68/RESULT = {"IrReceived":{"Protocol":"GREE","Bits":64,"Data":"0x0x0904205800200010","Repeat":0,"IRHVAC":{"Vendor":"GREE","Model":2,"Power":"On","Mode":"Cool","Celsius":"On","Temp":20,"FanSpeed":"Auto","SwingV":"Auto","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"On","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}}

The remote clearly indicates that I can change it from C to F using the remote control.  I've attached a picture of the instructions.  

[image]

s-hadinger, 2020-05-15:
Now duplicate of https://github.com/crankyoldgit/IRremoteESP8266/issues/1121
Please close this issue

effelle, 2020-05-15:
Closing this issue as it has been answered.

Support Information
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-05-15 by effelle

crankyoldgit, 2020-05-17:
Just a "For the record" note:
The IR library, when requested to do so, will try to use Fahrenheit where ever possible, **if** the underlying protocol has library support for it. Native Fahrenheit support is in the minority unfortunately.