arendst/Tasmota #8439: Connection lost after a Time

Arut42, 2020-05-14:
### PROBLEM DESCRIPTION
After one to two days the device is not able to get pinged. Then I have to kick it from Wlan and let it reconnect.
After that it is working normaly.

Most of the time this happens when the Device went in to Wlan-powersave-mode.
I'm using unfi Accespoint with a distance of around 10m.
Console output after reconect: 07:50:53 RSL: STATE = {"Time":"2020-05-14T07:50:53","Uptime":"2T15:30:10","UptimeSec":228610,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"xxx","BSSId":"xxx","Channel":11,"RSSI":50,"Signal":-75,"LinkCount":4,"Downtime":"0T00:00:13"}}

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [-] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Teckin SP22
- [x] Tasmota binary firmware version number used: 8.2.0
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: tuya
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
09:56:38 CMD: Backlog Template; Module; GPIO 255
09:56:38 RSL: RESULT = {"NAME":"Teckin","GPIO":[0,17,0,57,134,132,0,0,131,56,21,0,0],"FLAG":0,"BASE":52}
09:56:38 RSL: RESULT = {"Module":{"52":"Teckin"}}
09:56:38 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"17":"Button1"},"GPIO2":{"0":"None"},"GPIO3":{"56":"Led1i"},"GPIO4":{"134":"BL0937 CF"},"GPIO5":{"132":"HLWBL CF1"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"131":"HLWBL SELi"},"GPIO13":{"158":"LedLinki"},"GPIO14":{"21":"Relay1"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [-] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
09:59:51 CMD: Backlog Rule1; Rule2; Rule3;
09:59:51 RSL: RESULT = {"Rule1":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
09:59:52 RSL: RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
09:59:52 RSL: RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}

```
- [x] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
10:00:49 CMD: Status 0
10:00:49 RSL: STATUS = {"Status":{"Module":52,"FriendlyName":["xxx"],"Topic":"tasmota","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
10:00:49 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"2T17:40:06","StartupUTC":"2020-05-11T15:20:43","Sleep":50,"CfgHolder":4617,"BootCount":11,"BCResetTime":"2020-05-11T16:20:51","SaveCount":55,"SaveAddress":"FA000"}}
10:00:49 RSL: STATUS2 = {"StatusFWR":{"Version":"8.2.0(tasmota)","BuildDateTime":"2020-03-20T14:45:23","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8285","CR":"354/699"}}
10:00:49 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["xxx",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008001","2805C8000100060000005A00000000000000","00000001","00000000"]}}
10:00:49 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":577,"Free":424,"Heap":26,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","000000CD","010013C0","C000F981","00000004"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
10:00:49 RSL: STATUS5 = {"StatusNET":{"Hostname":"xxx","IPAddress":"xxx","Gateway":"xxx","Subnetmask":"255.255.255.0","DNSServer":"xxx","Mac":"xxx","Webserver":2,"WifiConfig":2,"WifiPower":17.0}}
10:00:49 RSL: STATUS7 = {"StatusTIM":{"UTC":"2020-05-14T09:00:49","Local":"2020-05-14T10:00:49","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"05:09","Sunset":"20:24"}}
10:00:49 RSL: STATUS9 = {"StatusPTH":{"PowerDelta":0,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
10:00:49 RSL: STATUS10 = {"StatusSNS":{"Time":"2020-05-14T10:00:49","ENERGY":{"TotalStartTime":"2020-05-11T14:59:36","Total":0.795,"Yesterday":0.000,"Today":0.157,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":229,"Current":0.000}}}
10:00:49 RSL: STATUS11 = {"StatusSTS":{"Time":"2020-05-14T10:00:49","Uptime":"2T17:40:06","UptimeSec":236406,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"ON","Wifi":{"AP":1,"SSId":"xxx","BSSId":"xxx","Channel":11,"RSSI":48,"Signal":-76,"LinkCount":4,"Downtime":"0T00:00:13"}}}
10:00:53 RSL: STATE = {"Time":"2020-05-14T10:00:53","Uptime":"2T17:40:10","UptimeSec":236410,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"ON","Wifi":{"AP":1,"SSId":"xxx","BSSId":"xxx","Channel":11,"RSSI":48,"Signal":-76,"LinkCount":4,"Downtime":"0T00:00:13"}}
10:00:53 RSL: SENSOR = {"Time":"2020-05-14T10:00:53","ENERGY":{"TotalStartTime":"2020-05-11T14:59:36","Total":0.795,"Yesterday":0.000,"Today":0.157,"Period":1,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":229,"Current":0.000}}

```
- [x] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
weblog 4
10:05:22 RSL: RESULT = {"WebLog":4}
10:05:23 CFG: Saved to flash at F9, Count 56, Bytes 4096
10:05:34 WIF: Checking connection...
10:05:53 RSL: STATE = {"Time":"2020-05-14T10:05:53","Uptime":"2T17:45:10","UptimeSec":236710,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"ON","Wifi":{"AP":1,"SSId":"xxx","BSSId":"xxx","Channel":11,"RSSI":48,"Signal":-76,"LinkCount":4,"Downtime":"0T00:00:13"}}
10:05:53 RSL: SENSOR = {"Time":"2020-05-14T10:05:53","ENERGY":{"TotalStartTime":"2020-05-11T14:59:36","Total":0.801,"Yesterday":0.000,"Today":0.163,"Period":6,"Power":96,"ApparentPower":129,"ReactivePower":86,"Factor":0.74,"Voltage":228,"Current":0.565}}
10:05:54 WIF: Checking connection...

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Wait 2 days if device went do powersaving it dont wake up

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Device should work

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

kugelkopf123, 2020-05-14:
Perhaps Setoption41 could help you. https://tasmota.github.io/docs/Commands/

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

giacomotontini, 2020-05-14:
Noticed the same thing on recent release.

Jason2866, 2020-05-15:
Sorry, we cant help since it is a router / AP problem.  See https://github.com/arendst/Tasmota/issues/7098#issuecomment-565432020

closed 2020-05-15 by ascillato2