arendst/Tasmota #8440: Modbus module adds garbage to commands

andyross0805, 2020-05-14:
### PROBLEM DESCRIPTION
When sending commands to MODBUS device, after command and before CRC code there are two "00" and "02" garbage bytes. Source of problem - xsns_53_sml.ino lines 2275&2276

### REQUESTED INFORMATION
If it`s needed for some devices to get this additional 00 02 bytes, I suppose to add additional option to switch it on and off for more compatibility with MODBUS devices.

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): Wemos D3
- [ ] Tasmota binary firmware version number used: 8.2 + Scripting ON
  - [ ] Pre-compiled
  - [v ] Self-compiled
    - [ ] IDE / Compiler used: gitpod
- [ ] Flashing tools used: NodeMCU_PyFlasher
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
`12:49:41 RSL: stat/tasmota_E477E1/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
12:49:42 RSL: stat/tasmota_E477E1/RESULT = {"Module":{"18":"Generic"}}
12:49:42 RSL: stat/tasmota_E477E1/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}`
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
12:50:46 RSL: stat/tasmota_E477E1/STATUS = {"Status":{"Module":18,"FriendlyName":["SolarCtrl2"],"Topic":"tasmota_E477E1","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
12:50:46 RSL: stat/tasmota_E477E1/STATUS1 = {"StatusPRM":{"Baudrate":9600,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"External System","Uptime":"0T00:01:28","StartupUTC":"2020-05-14T09:49:18","Sleep":50,"CfgHolder":4617,"BootCount":32,"BCResetTime":"2020-05-13T19:57:55","SaveCount":86,"SaveAddress":"F6000"}}
12:50:46 RSL: stat/tasmota_E477E1/STATUS2 = {"StatusFWR":{"Version":"8.2.0(tasmota)","BuildDateTime":"2020-05-14T07:06:14","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"382/699"}}
12:50:46 RSL: stat/tasmota_E477E1/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["AAR","Dubna1"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
12:50:46 RSL: stat/tasmota_E477E1/STATUS4 = {"StatusMEM":{"ProgramSize":599,"Free":404,"Heap":25,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164020","FlashMode":3,"Features":["00000809","87DAE397","043E83A1","000000CD","010013C0","C000F989","00000004"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6,53"}}
12:50:46 RSL: stat/tasmota_E477E1/STATUS5 = {"StatusNET":{"Hostname":"SolarCtrl2","IPAddress":"192.168.2.126","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.1","Mac":"84:F3:EB:E4:77:E1","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
12:50:46 RSL: stat/tasmota_E477E1/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_E477E1","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
12:50:46 RSL: stat/tasmota_E477E1/STATUS7 = {"StatusTIM":{"UTC":"2020-05-14T09:50:46","Local":"2020-05-14T12:50:46","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+03:00","Sunrise":"07:09","Sunset":"22:24"}}
12:50:46 RSL: stat/tasmota_E477E1/STATUS10 = {"StatusSNS":{"Time":"2020-05-14T12:50:46","MODBUS":{"Bat_Capacity":0}}}
12:50:46 RSL: stat/tasmota_E477E1/STATUS11 = {"StatusSTS":{"Time":"2020-05-14T12:50:46","Uptime":"0T00:01:28","UptimeSec":88,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"AAR","BSSId":"64:D1:54:9E:E0:40","Channel":1,"RSSI":98,"Signal":-51,"LinkCount":1,"Downtime":"0T00:00:06"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
Commands sent to MODBUS have only command code+CRC

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

kugelkopf123, 2020-05-14:
Can you additionally post the code from your scriptEditor? And what kind of modbus device you use. 

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

gemu2015, 2020-05-14:
no garbage
this is the number of MODBUS registers requested (intentionally always 2 ones)

http://www.simplymodbus.ca/FC03.htm
e.g.

11 03 006B 0003 7687

11: The Slave Address (11 hex = address17 )
03: The Function Code 3 (read Analog Output Holding Registers)
006B: The Data Address of the first register requested.
             ( 006B hex = 107 , + 40001 offset = input #40108 )
0003: The total number of registers requested. (read 3 registers 40108 to 40110)  
7687: The CRC (cyclic redundancy check) for error checking.

andyross0805, 2020-05-14:
Here is my script:
D

>B
=>sensor53 r

>M 1
+1,3,m,0,9600,MODBUS,1,1,010301000001,010301010001,010301030001,010301070003,010301080003,010301090003,010301110002

1, 010302xxuuxxxx@i0:1,Capacity,%,Bat_Capacity,0
1, 010302uuuuxxxx@i1:10,Battery Voltage,V,Bat_Volt,2
1, 010302uuxxxxxx@i2:1,Battery Temp,C,Bat_temp,0
1, 010302xxuuxxxx@i2:1,Contr Temp,C,Contr_temp,0
1, 010302UUuuxxxx@i3:10,Solar Volt,V,Sol_volt,2
1, 010302UUuuxxxx@i4:10,Charge curr,A,Charge_I,2
1, 010302UUuuxxxx@i5:10,Charge Power,W,Charge_pwr,2
1, 010304UUuuxxxxxxxx@i6:10,Today AH,AH,Today_ah,0
1, 010304xxxxUUuuxxxx@i6:10,Today disch AH,AH,Today_dis_ah,0
#

Device is Solar Charging controller and in documentation for MODBUS implementation there is no "data count" - just CRC. MODBUS telegram for it is: 1 byte address, 1 byte function code, N bytes data, CRC

andyross0805, 2020-05-14:
> 0003: The total number of registers requested. (read 3 registers 40108 to 40110)

So it`s not fixed 00 02 - it`s should vary depending on number of registers you want to read.
But in my case this parameter is not required in telegram.

gemu2015, 2020-05-14:
as you can see from your script you request mostly 1 register sometimes 3 registers,
this is not supported! 
if you omit the last 2 bytes of your request and adjust the values the decoder will work.

the 3 register values however are split into 2 values. you can combine them with a script command

the Data count is necessary on request! (as you can see from your script)
but not for the response.

andyross0805, 2020-05-14:
> the Data count is necessary on request! (as you can see from your script)
> but not for the response.
Data count is not required by my device. Moreover, now you will always ask your device for 2 registers, always! It`s not correct in my point of view. You can add number of registers to ask into your script (into command), not hardcode into communication part of Tasmota.

kugelkopf123, 2020-05-14:
@andyross0805 Can you please link to the user manual which also contains the data for the Modbus protocol and the Modbus registers?

gemu2015, 2020-05-14:
the driver was optimized to fit most meter devices and it works with almost any MODUS devices.
The 2 register demand is a compromise for script code and speed and almost any values fit with 2 registers. Until now a 3 register value was not implemented or needed.

i now modified the  code that you may give an r before the hex string 
e.g. r010301090003
which allows you to send other than 2 register requests.
however the 3 reg value still has to be implemented if you post the data sheet.

andyross0805, 2020-05-14:
If I correctly understand, some MODBUS devices require parameter "how many registers to read", but some has fixed amount of registers they return, so placing count parameter into script is more flexible.
But anyway, I have modified for myself xsns_53_sml.ino and debugging hardware (I suspect some bugs in return channel), but output from Tasmota now correct for me.
I have link to documentation from supplier [https://cloud.mail.ru/public/Zk8n/pYRCkJAmn/](url) - I dont know how long it will be active.

gemu2015, 2020-05-14:
link is dead

kugelkopf123, 2020-05-14:
@gemu2015 [https://cloud.mail.ru/public/Zk8n/pYRCkJAmn/](https://cloud.mail.ru/public/Zk8n/pYRCkJAmn/)

andyross0805, 2020-05-14:
Now I see response coming, but no luck - response 01030400510080AA42, but values are not parced. I simplified filters and use standard tasmota modbus implementation (which adds 0002 - number of registers to ask), but no luck - values are zero: 
>M 1
+1,3,m,0,9600,MODBUS,1,10,01030100

1, 010304xxuuxxxx@i0:1,Capacity,Perc,Bat_Capacity,0
1, 010304xxxxUUuu@i0:10,Battery Voltage,V,Bat_Volt,2

gemu2015, 2020-05-15:
this looks OK so far, should decode if checksum is ok
you may uncomment the log in the mbus decode
and may also put a log before the checksum check

uint16_t crc = MBUS_calculateCRC(&smltbuf[mindex][0],pos);
              if (lowByte(crc)!=smltbuf[mindex][7]) goto nextsect;
              if (highByte(crc)!=smltbuf[mindex][8]) goto nextsect;
              dval=mbus_dval;
              //AddLog_P2(LOG_LEVEL_INFO, PSTR(">> %s"),mp);

andyross0805, 2020-05-15:
> this looks OK so far, should decode if checksum is ok
> you may uncomment the log in the mbus decode
> and may also put a log before the checksum check

CRC is ok, logging uncommented, but where to see log?

gemu2015, 2020-05-15:
ok that means the decoder does not get there!

put a log here to see if compare was ok =>
if (found) {
        // matches, get value
       AddLog_P2(LOG_LEVEL_INFO, PSTR("matched %s"),mp);

gemu2015, 2020-05-15:
by the way the compare is NOT activ in dump mode you must set sensor53 d0

andyross0805, 2020-05-15:
Yes, I set sensor53 d0 and sensor53 r than. But nothing regarding parsing logged into syslog. Trying to add more logging into very begin of Parsing procedure.

gemu2015, 2020-05-15:
glad it worked, will make a remark in the docu regarding spaces.
in my last version of the driver i implemented the r (raw) MODBUS mode were you can enter the complete Request Sequence and request e.g. one or three registers.
it also contains other new features which allow putting values from registers together
and hiding the interim results.
i will give examples in the docu when it is merged into Tasmota

andyross0805, 2020-05-15:
Cool! Thanks a lot!

closed 2020-05-15 by andyross0805