arendst/Tasmota #8447: xiaomi philips color teperature control is swapped

kurumuru, 2020-05-14:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Color temperature slider from web interface 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/d  Configuration output here:
evelopment/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x] Device used (e.g., Sonoff Basic): xiaomi philips
- [x ] Tasmota binary firmware version number used: 8.3.0
  - [x ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
21:26:46 MQT: stat/xp_top_1/RESULT = {"NAME":"xp_my","GPIO":[0,0,0,0,0,0,0,0,47,0,0,37,0],"FLAG":0,"BASE":48}
21:26:47 MQT: stat/xp_top_1/RESULT = {"Module":{"48":"Xiaomi Philips"}}
21:26:47 MQT: stat/xp_top_1/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"38":"PWM2"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"37":"PWM1"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
21:23:47 CMD: Status 0`
21:23:47 MQT: stat/xp_top_1/STATUS = {"Status":{"Module":48,"FriendlyName":["Sonoff"],"Topic":"%hostname%","ButtonTopic":"0","Power":0,"PowerOnState":0,"LedState":1,"LedMask":"FFFF","SaveData":0,"SaveState":0,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
21:23:47 MQT: stat/xp_top_1/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://hbox:5556/data/firmwares/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:46","StartupUTC":"2020-05-14T20:23:01","Sleep":150,"CfgHolder":4617,"BootCount":96,"BCResetTime":"2020-03-21T15:40:54","SaveCount":320,"SaveAddress":"F4000"}}
21:23:47 MQT: stat/xp_top_1/STATUS2 = {"StatusFWR":{"Version":"8.3.0(tasmota)","BuildDateTime":"2020-05-14T16:17:08","Boot":31,"Core":"2_7_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"346/699"}}
21:23:47 MQT: stat/xp_top_1/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["sis",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008008","2805C8000100060000005AFF000000000000","00000200","00000000"]}}
21:23:47 MQT: stat/xp_top_1/STATUS4 = {"StatusMEM":{"ProgramSize":588,"Free":412,"Heap":24,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540C8","FlashMode":3,"Features":["00000809","8FDAE797","043683A0","000000CD","010013C0","C000F981","00000024"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
21:23:47 MQT: stat/xp_top_1/STATUS5 = {"StatusNET":{"Hostname":"xp_top_1","IPAddress":"192.168.1.103","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"5C:CF:7F:5B:15:4D","Webserver":2,"WifiConfig":2,"WifiPower":17.0}}
21:23:47 MQT: stat/xp_top_1/STATUS6 = {"StatusMQT":{"MqttHost":"hbox","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_5B154D","MqttUser":"device","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
21:23:48 MQT: stat/xp_top_1/STATUS7 = {"StatusTIM":{"UTC":"2020-05-14T20:23:47","Local":"2020-05-14T21:23:47","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"05:09","Sunset":"20:23"}}
21:23:48 MQT: stat/xp_top_1/STATUS10 = {"StatusSNS":{"Time":"2020-05-14T21:23:48"}}
21:23:48 MQT: stat/xp_top_1/STATUS11 = {"StatusSTS":{"Time":"2020-05-14T21:23:48","Uptime":"0T00:00:47","UptimeSec":47,"Heap":24,"SleepMode":"Dynamic","Sleep":150,"LoadAvg":5,"MqttCount":1,"POWER":"OFF","Dimmer":71,"Color":"B500","White":71,"CT":153,"Channel":[71,0],"Fade":"ON","Speed":6,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"sis","BSSId":"58:6D:8F:27:3D:83","Channel":11,"RSSI":92,"Signal":-54,"LinkCount":1,"Downtime":"0T00:00:05"}}}
```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
move slider to warm - light became cold

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
move slider to warm - light became warm.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
Last known working version 8.2.0.

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2020-05-14:
What exact bulb is it?

There's an easy fix, set the second pwm as `PWM2i` it will invert controls.

kurumuru, 2020-05-15:
Bulb looks like [this](https://www.amazon.co.uk/Xiaomi-MUE4088RT-Philips-Wi-Fi-Bulb-White/dp/B07PY84ZMV).
It seems that `PWMi` do the job. So that bulb doesn't work out of the box?

s-hadinger, 2020-05-16:
I can invert it in the code but it might break existing configuration. It the bulb a genuine Xiaomi Philips?

kurumuru, 2020-05-18:
Not 100% sure, but something was changed with that update.
A have 4 bulb, and all of them need pwm2 to be inverted.

blakadder, 2020-05-18:
That is a new model of a Xiaomi Phillips Bulb, the one in the template is the Philips Zhirui model

ascillato2, 2020-05-18:
Closing this issue as it has been answered.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-05-18 by ascillato2

s-hadinger, 2020-05-18:
@kurumuru I must have inverted mistakenly the way CT is managed in the past, but it must have been a long time ago. I'm reluctant to change the way CT is controlled, because a lot of new bulbs rely on this behavior. I will update the template in Tasmota and in the template repository to use PMW2i now.