arendst/Tasmota #8453: precompiled binary  with conditions

riker65, 2020-05-16:
I am not having a development IDE and do not know how to compile. So would it be possible to get a precompiled binary with condition and expression support?

**Have you looked for this feature in other issues and in the docs?**  

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._
o would it be possible to get a precompiled binary with condition and expression support?
**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._
o would it be possible to get a precompiled binary with condition and expression support?
**Additional context**  
_Add any other context or screenshots about the feature request here._
Not IDE available
**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-05-16:
See https://github.com/tasmota/binaries

riker65, 2020-05-16:
thanks perfect !!! really great

closed 2020-05-16 by ascillato2