arendst/Tasmota #8454: IRSend Raw - Invalid JSON

erhapp, 2020-05-16:
### PROBLEM DESCRIPTION
Sending raw data via IRSend results in error message.

Command send from console: IRSend 0,350,300,350,600,300,300,650,600,350,300,600,600,650,600,650,600,300,300,350,300,350,250,650,600,300

Error message received: {"IRSend":"Invalid JSON"}

### REQUESTED INFORMATION
- [ ] Device used: D1 Mini
- [ ] Tasmota binary firmware version number used: 8.3.0 - Pre-compiled
- [ ] Flashing tools used: Tazmotizer (8.2.0) + OTA (8.3.0)
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
08:25:11 MQT: Afstandsbediening/stat/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
08:25:11 MQT: Afstandsbediening/stat/RESULT = {"Module":{"0":"Generic"}}
08:25:11 MQT: Afstandsbediening/stat/RESULT = {"GPIO0":{"0":"Geen"},"GPIO1":{"0":"Geen"},"GPIO2":{"0":"Geen"},"GPIO3":{"0":"Geen"},"GPIO4":{"0":"Geen"},"GPIO5":{"0":"Geen"},"GPIO9":{"0":"Geen"},"GPIO10":{"0":"Geen"},"GPIO12":{"51":"IRrecv"},"GPIO13":{"0":"Geen"},"GPIO14":{"8":"IRsend"},"GPIO15":{"0":"Geen"},"GPIO16":{"0":"Geen"}}

- [ ] Provide the output of this command: `Status 0`:
08:25:46 MQT: Afstandsbediening/stat/STATUS = {"Status":{"Module":0,"FriendlyName":["Afstandsbediening"],"Topic":"Afstandsbediening","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
08:25:46 MQT: Afstandsbediening/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota-NL.bin","RestartReason":"Software/System restart","Uptime":"0T15:30:56","StartupUTC":"2020-05-15T15:54:50","Sleep":50,"CfgHolder":4617,"BootCount":26,"BCResetTime":"2020-04-28T17:15:23","SaveCount":161,"SaveAddress":"F9000"}}
08:25:46 MQT: Afstandsbediening/stat/STATUS2 = {"StatusFWR":{"Version":"8.3.0(tasmota)","BuildDateTime":"2020-05-14T16:26:07","Boot":31,"Core":"2_7_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"409/699"}}
08:25:46 MQT: Afstandsbediening/stat/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Happaerts-Valgaeren",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000100","00000000"]}}
08:25:46 MQT: Afstandsbediening/stat/STATUS4 = {"StatusMEM":{"ProgramSize":588,"Free":412,"Heap":22,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640EF","FlashMode":3,"Features":["00000413","8FDAE797","043683A0","000000CD","010013C0","C000F981","00000024"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
08:25:46 MQT: Afstandsbediening/stat/STATUS5 = {"StatusNET":{"Hostname":"Afstandsbediening","IPAddress":"192.168.0.226","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"195.130.131.4","Mac":"4C:11:AE:0D:E0:F3","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
08:25:46 MQT: Afstandsbediening/stat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.8","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_0DE0F3","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
08:25:46 MQT: Afstandsbediening/stat/STATUS7 = {"StatusTIM":{"UTC":"2020-05-16T07:25:46","Local":"2020-05-16T08:25:46","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"05:06","Sunset":"20:26"}}
08:25:46 MQT: Afstandsbediening/stat/STATUS10 = {"StatusSNS":{"Time":"2020-05-16T08:25:46"}}
08:25:46 MQT: Afstandsbediening/stat/STATUS11 = {"StatusSTS":{"Time":"2020-05-16T08:25:46","Uptime":"0T15:30:56","UptimeSec":55856,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"Happaerts-Valgaeren","BSSId":"38:43:7D:3F:BB:B1","Channel":11,"RSSI":100,"Signal":-23,"LinkCount":1,"Downtime":"0T00:00:06"}}}

### TO REPRODUCE

Enter command in console (or send via MQTT)

### EXPECTED BEHAVIOUR

{"IRSend":"Done"}

### ADDITIONAL CONTEXT

Command and raw string worked in 8.2.0

schrej-zz, 2020-05-16:
@erhapp,
Did you really search before you opened an issue?
You need tasmota-ir.bin for "send".
Joerg

erhapp, 2020-05-16:
@schrej-zz I know. The OTA upgraded to the standard version. Problem is that sendir works in the standard version. But only for some protocols. Raw sending is not avaiable. I switched back to the IR version.

closed 2020-05-16 by ascillato2