arendst/Tasmota #8457: Support for P9813 LED Controllers

rkkoszewski, 2020-05-16:
**Have you looked for this feature in other issues and in the docs?**  
I searched trough the Tasmota documentation, Github Issues and generally trough the internet without any results.

**Is your feature request related to a problem? Please describe.**  
Inability to make the P9813 work. 

**Describe the solution you'd like**  
I'm unable to get a [P9813](https://www.diymore.cc/products/full-color-rgb-led-strip-driver-module-shield-for-arduino-stm32-avr-v1-0) to work with Tasmota. Is there some other driver that is compatible with this chip? Or would it be possible to add compatibility to this chip?

**Describe alternatives you've considered**  
FastLED supports this chip. I tested it controlling the P9813 with an ESP01 by using FastLED and it works. Alternatively, would it be possible to add a shim to Tasmota to use FastLED to control the chip? (There are probably also directly drivers for the P9813 without using FastLED). 

**Additional context**  
I'm trying to get an [P9813](https://www.diymore.cc/products/full-color-rgb-led-strip-driver-module-shield-for-arduino-stm32-avr-v1-0) chip to work with an ESP01 with Tasmota. Tasmota is already flashed on the ESP01 and is working, and the P9813 is wired to the ESP01. This setup has been running successfully for a while now with my own [firmware ](P9813)which is using FastLED to control the P9813. I'm trying to migrate to Tasmota now. 

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2020-06-12:
> Is there some kind of requirements for this?

For a feature request? No, there isn't. You have requested a new feature. It has tagged as such, but I'm sorry that seems that so far there isn't any volunteer interested on this or that have enough free time to work for free in this support. 

> Is there any way I can help out with this?

Yes, forking Tasmota and developing this support. Then you can make a pull request to add it to Tasmota.

rkkoszewski, 2020-07-11:
Did not yet have time to look into it

rkkoszewski, 2020-08-08:
I'll reopen the issue once I find some time to work on it. Winter and rainy days are slowly comming back, so there should be some free time there :)

closed 2020-09-07 by stale[bot]