arendst/Tasmota #8461: Device not publish POWER=ON to MQTT after it powered up with PowerOnState=ON.

prakai, 2020-05-16:
### PROBLEM DESCRIPTION
I want to use the smart switch (Canwing CW-001) to control a lamp with the existing switch. If turn on existing switch (power is on), lamp is on, and turn off existing switch (power is off), lamp is off. Including the ability to turn off / turn on the lamp from Homebridge (Homebridge Mqttthing) via smart switch as well.

I have set PowerOnState=ON to turn the lamp on if existing switch is turned on. After turning on the lamp from the existing switch is perfect, but the lamp status on Homebrige is incorrect. 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Canwing CW-001
- [x] Tasmota binary firmware version number used: 8.3.0
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: tasmotizer-1.1a.exe
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
00:06:06 MQT: stat/CW_001_C58092/RESULT = {"Rule1":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
00:06:06 MQT: stat/CW_001_C58092/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
00:06:06 MQT: stat/CW_001_C58092/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
After turning on the lamp from the existing switch, and after smart switch connect to Wi-Fi, the lamp status on Homebrige should be ON. 

### SCREENSHOTS
#### Homebridge Logs
```
[5/16/2020, 11:37:02 PM] [Light-B] Received MQTT: tele/CW_001_C58092/LWT = Offline
[5/16/2020, 11:37:13 PM] [Light-B] Received MQTT: tele/CW_001_C58092/LWT = Online
[5/16/2020, 11:37:13 PM] [Light-B] Received MQTT: stat/CW_001_C58092/POWER = ON
[5/16/2020, 11:37:14 PM] [Light-B] Received MQTT: stat/CW_001_C58092/POWER = OFF
```

#### CW-001 Logs
```
00:01:05 MQT: tele/CW_001_C58092/LWT = Online (retained)
00:01:05 MQT: cmnd/CW_001_C58092/POWER = 
00:01:05 MQT: tele/CW_001_C58092/INFO1 = {"Module":"CW-001","Version":"8.3.0(tasmota)","FallbackTopic":"cmnd/DVES_C58092_fb/","GroupTopic":"cmnd/CW_001_C58092/"}
00:01:05 MQT: tele/CW_001_C58092/INFO2 = {"WebServerMode":"Admin","Hostname":"tasmota_C58092-0146","IPAddress":"192.168.1.26"}
00:01:05 MQT: tele/CW_001_C58092/INFO3 = {"RestartReason":"Power On"}
00:01:05 MQT: stat/CW_001_C58092/RESULT = {"POWER":"ON"}
00:01:05 MQT: stat/CW_001_C58092/POWER = ON
00:01:05 MQT: stat/CW_001_C58092/RESULT = {"POWER":"OFF"}
00:01:05 MQT: stat/CW_001_C58092/POWER = OFF
```

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-05-16:
Hi, sorry. Can you please explain further?Your logs say power = on, so where is the issue?

Also please, complete the template, specifically the part of status 0 output.

prakai, 2020-05-17:
I have found that the problem may be caused by homebridge-mqttthing on Homebridge.

Switch configurations:
```
PowerOnState = ON
PowerRetain = OFF
ButtonRetain = OFF
```
Homebridge configurations:
```
{
    "type": "lightbulb-OnOff",
    "name": "Light-B",
    "url": "xxxx",
    "username": "xxxx",
    "password": "xxxx",
    "mqttOptions": {
        "keepalive": 10
    },
    "logMqtt": true,
    "topics": {
        "getOn": "stat/CW_001_C58092/POWER",
        "setOn": "cmnd/CW_001_C58092/POWER"
    },
    "onlineValue": "Online",
    "onValue": "ON",
    "offValue": "OFF",
    "accessory": "mqttthing"
}
```

1. I have changed switch MQTT toptic differences from Homebridge accessory MQTT topic, then the switch is works well.

Switch logs:
```
00:00:32 MQT: tele/CW_001_C58092x/LWT = Online (retained)
00:00:32 MQT: cmnd/CW_001_C58092x/POWER = 
00:00:32 MQT: tele/CW_001_C58092x/INFO1 = {"Module":"CW-001","Version":"8.3.0(tasmota)","FallbackTopic":"cmnd/DVES_C58092x_fb/","GroupTopic":"cmnd/CW_001_C58092x/"}
00:00:32 MQT: tele/CW_001_C58092x/INFO2 = {"WebServerMode":"Admin","Hostname":"tasmota_C58092-0146","IPAddress":"192.168.1.26"}
00:00:32 MQT: tele/CW_001_C58092x/INFO3 = {"RestartReason":"Power On"}
00:00:32 MQT: stat/CW_001_C58092x/RESULT = {"POWER":"ON"}
00:00:32 MQT: stat/CW_001_C58092x/POWER = ON
00:00:36 MQT: tele/CW_001_C58092x/STATE = {"Time":"1970-01-01T00:00:36","Uptime":"0T00:00:39","UptimeSec":39,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"IoT","BSSId":"70:C7:F2:9C:93:C9","Channel":1,"RSSI":70,"Signal":-65,"LinkCount":1,"Downtime":"0T00:00:33"}}
```
2. I have changed switch MQTT toptic same as Homebridge accessory MQTT topic. The switch receives POWER = OFF from Homebrige after the message itself.

Switch logs: 
```
00:00:32 MQT: tele/CW_001_C58092/LWT = Online (retained)
00:00:32 MQT: cmnd/CW_001_C58092/POWER = 
00:00:32 MQT: tele/CW_001_C58092/INFO1 = {"Module":"CW-001","Version":"8.3.0(tasmota)","FallbackTopic":"cmnd/DVES_C58092_fb/","GroupTopic":"cmnd/CW_001_C58092/"}
00:00:32 MQT: tele/CW_001_C58092/INFO2 = {"WebServerMode":"Admin","Hostname":"tasmota_C58092-0146","IPAddress":"192.168.1.26"}
00:00:32 MQT: tele/CW_001_C58092/INFO3 = {"RestartReason":"Power On"}
00:00:32 MQT: stat/CW_001_C58092/RESULT = {"POWER":"ON"}
00:00:32 MQT: stat/CW_001_C58092/POWER = ON
>> 00:00:32 MQT: stat/CW_001_C58092/RESULT = {"POWER":"OFF"} <<
>> 00:00:32 MQT: stat/CW_001_C58092/POWER = OFF <<
00:00:37 MQT: tele/CW_001_C58092/STATE = {"Time":"1970-01-01T00:00:37","Uptime":"0T00:00:39","UptimeSec":39,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":22,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"IoT","BSSId":"70:C7:F2:9C:93:C9","Channel":1,"RSSI":82,"Signal":-59,"LinkCount":1,"Downtime":"0T00:00:33"}}
```

Jason2866, 2020-05-17:
Please close since it is not a Tasmota issue

closed 2020-05-18 by prakai